linuxwacom/xf86-input-wacom #5: only 1 pen button is recognized on Thinkpad Yoga 260.

rainwoodman, 2017-03-08:
lsusb: Bus 001 Device 005: ID 056a:5044 Wacom Co., Ltd

This is also mentioned in https://wiki.archlinux.org/index.php/Lenovo_Thinkpad_Yoga_260#TouchScreen_and_Stylus

Is this the right place to post the bug?

jigpu, 2017-03-08:
Thanks for the report. Most bugs are reported on [our Sourceforge bugtracker](https://sourceforge.net/p/linuxwacom/bugs) but here is fine as well :)

The Thinkpad X1 Yoga is configured so that its second button causes the pen to act as an eraser (see [SF Bug 327](https://sourceforge.net/p/linuxwacom/bugs/327/)). Perhaps that is what is going on here as well? If you have Arch Linux installed, try installing the "xorg-xinput" package and then run the following command:

    $ xinput test "$(xinput list --name-only | grep -m1 eraser)"

This command will print out eraser events as they're fed into the system. If you bring the pen into proximity without any buttons pressed, it shouldn't do anything. If, however, you bring the pen into proximity with the "non-functional" button pressed (or start pressing it while the pen is in proximity), it may start producing output. If so, that would indicate the Yoga 260 acts similar to the X1.

rainwoodman, 2017-03-09:
Thanks. Indeed it is emitted as eraser.

A somewhat related issue. When I log in I do see a warning about "wacom device is not recognized". The device seem to work otherwise.

jigpu, 2017-03-09:
Good to hear that this is "just" the same issue. I'll close this as a duplicate of bug 327 linked above. I'm still not entirely sure if its possible for us to fix since I'm not aware of any way for us to tell how a particular tablet's firmware is configured, but perhaps there is a solution...

As for the "Wacom device is not recognized" warning that you see, that is produced by GNOME (not our driver) when it does not recognize a particular tablet. The only effect is that you can't configure the tablet though the control panel. Since GNOME depends on libwacom for device detection, please [file a bug with our libwacom project](https://github.com/linuxwacom/libwacom/issues) which indicates that the Yoga 260 is not recognized. Although not strictly necessary, I would request that you run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the tarball it generates to that bug. This will provide more information about how the Yoga 260 is configured.

closed 2017-03-09 by jigpu

rainwoodman, 2017-03-09:
Thanks for the explanation. Ticket filed.

jigpu, 2017-03-09:
Thanks :)

Also, I almost forgot -- just in case the submitter of the bug at SourceForge is slow to respond, would you mind doing the kind of capture I outlined in [this comment](https://sourceforge.net/p/linuxwacom/bugs/327/#d26c) (except using /tmp/056a_5044.hidrecord as your output file) and uploading the results here? It will give me some data to chew on regarding exactly which bits are being toggled when the buttons are pressed.

rainwoodman, 2017-03-09:
[056a_5044.hidrecord.txt](https://github.com/linuxwacom/xf86-input-wacom/files/831731/056a_5044.hidrecord.txt)

Here you go. I did each multiple times.

cstoitner, 2019-08-03:
Sorry for dragging up this old issue... but I think not supporting remapping of these eraser buttons is really limiting the stylus.

I am using a Lenovo X1 tablet 3rd gen with the included stylus, and on Lenovos Windows pen configuration tool I can remap both barrel buttons to pretty much any function (including mouse buttons or the eraser function). I would expect the Linux driver to allow for the same functionality.

jigpu, 2019-10-11:
FYI: Issue opened for configuring the eraser button https://gitlab.freedesktop.org/libinput/libinput/issues/370