linuxwacom/xf86-input-wacom #8: wcmWaitForTablet() in wcmISDV4.c waits for 250 seconds instead of 250 ms

cfeck, 2017-05-21:
Line 877 in wcmISDV4.c says:

    xf86WaitForInput(pInfo->fd, 1000000)

According to xf86WaitForInput documentation, the "timeout" value is in microseconds, 1000000 us = 1 second.

The wait is iterated up to MAXTRY (== 250) times, which means it delays up to 250 seconds. According to http://linuxwacom.sourceforge.net/wiki/index.php/ISDV4_Protocol the expected delay is 250 ms.

A suggested fix is to change the call to

    xf86WaitForInput(pInfo->fd, 1000)

resulting in a wait of 1000 us = 1 ms per iteration, for a total of expected 250 ms.

Without this change, the boot times into graphical X11 raises by 250 s on a system with ISDV4 Wacom display tablet present.

jigpu, 2017-05-24:
Thanks for the diagnosis of this issue. There's a report on the SourceForge bug tracker (https://sourceforge.net/p/linuxwacom/bugs/333/) of long delays with ISDv4 initialization which could very easily be explained by this. I'll be writing up a patch shortly and posting it in the SourceForge bug for testing.

cfeck, 2017-06-08:
I have bad news. I patched the xf86-input-wacom package with the proposed change, but the system still waits a multiple of 250 seconds. Do you happen to know if the code above is replicated in other code bases, maybe in the X server directly?

jigpu, 2017-06-08:
The driver contains several calls to this function which could potentially cause this issue. From what I can tell, the problem originates in the X server: https://cgit.freedesktop.org/xorg/xserver/commit/?id=2fbf62b2fb3dcb29551251d09aa695715bb754f4

If you have the ability to compile the server from source I would recommend giving it a try. You'll need to checkout the master branch since the fix isn't in a released version yet.

cfeck, 2018-01-04:
I can confirm it is fixed now. Using xorg-server 1.19.5

closed 2018-01-04 by cfeck