linuxwacom/xf86-input-wacom #9: Pressure issues for all Intuos wacoms

rareshornet, 2017-11-08:
Hi guys,

First of all thanks for all the good stuff you guys are making for us.

I do have an issue on past 0.29 versions on the Intuos wacoms we use (PTH-660 intuso 5 and 4). The pressure seems its not working properly for the pen. At some point while pressing a bit more on the tip the pressure value gets inverted or something like that and instead of the line to get thicker and thicker it stops outputting anything on the canvas. If I only press a bit the line would get a bit thicker but then as I press more its interrupted. I'm not sure how to give you guys technical details about the issue. If there's a way i can log something please let me know and I'l send whatever you guys need. For now we use 0.29 and seems its working great but not sure what limitations we would have with new kmod drivers.

Any help would be much appreciated and thanks again for all the work you guys did so far.

Rares

rareshornet, 2017-11-08:
Just in case you need this info:

OS CentOS 7.4 
x wacom driver version: latest, 0.35.0
x kmod wacom: latest, 0.37.1

rareshornet, 2017-11-08:
I managed to isolate the problem starting with version 0.34.0. I built all the previous versions between 0.29 and 0.33.0 and seem to work fine

jigpu, 2017-11-08:
Hi there. This sounds a lot like an application issue that was uncovered when our driver changed its pressure range in 0.34. Details can be found at http://who-t.blogspot.com/2017/06/xf86-input-wacom-034-workaround-for.html?m=1

Since you already have version 0.35 installed, all you need to do is create a an xorg.conf.d snippet which enables the "Pressure2K" option. To quote the linked post:

> To get this to work, the following xorg.conf.d snippet is recommended
> 
>     Section "InputClass"
>         Identifier "Wacom pressure compatibility"
>         MatchDriver "wacom"
>         Option "Pressure2K" "true"
>     EndSection
> 
> Put it in a file that sorts higher than the wacom driver itself (e.g. /etc/X11/xorg.conf.d/99-wacom-pressure2k.conf) and restart X. Check the Xorg.log/journal for a "Using 2K pressure levels" message.

Please give it a try and let us know if it fixes your problem.

jigpu, 2018-01-09:
Closing this issue. Please re-open if the suggestion above did not resolve it.

closed 2018-01-09 by jigpu