linuxwacom/xf86-input-wacom #11: Pen tilt y is being replaced with tilt x

daniel-ferguson, 2018-03-03:
`xf86-input-wacom`: 0.36
OS: Arch linux
Tablet: Intuos 5 PTH-650/K

I noticed issues with tilt direction being broken in Krita and did some digging...

### Is my tablet broken?

Probably not, I plugged the tablet into my laptop running an older version of `xf86-input-wacom` (0.35.x) and tilt works fine.

### Where is the issue occurring

I took a dig into events being emitted by the kernel driver for my wacom pen with:

```
evtest | grep TILT
```

`ABS_TILT_X` and `ABS_TILT_Y` events are being emitted separately, with values corresponding to the angle I'm holding the pen.

On the other hand, looking at xorg events with:

```
xinput test $pen_id
```

Gave me rows like:
```
motion ... a[3]=39 a[4]=39 ...
```

Where `a[3]` and `a[4]` are always equal, and respond to changes in x tilt. Looking at the source for this project indicates they represent x and y tilt, respectively.

From `evtest`'s manpage it looks like `evtest` reports events emitted by the wacom kernel driver, where `xinput` will, I think, be reporting the output of `xf86-input-wacom`. That gave me enough confidence that this isn't an issue with the kernel driver.

---

I took a dig through source changes which have occurred since `xf86-input-wacom` 0.35 and spotted this line, which looks like it'd explain the issue I'm seeing: https://github.com/linuxwacom/xf86-input-wacom/blob/master/src/wcmFilter.c#L317

```
ds->tilty = wcmFilterAverage(state->tiltx, common->wcmRawSample);
```

---

Let me know if there's anything I can do to help with this, and thank you for your efforts maintaining this great project :+1:

daniel-ferguson, 2018-03-03:
Just a quick follow-up - I've rebuilt 0.36 with the change illustrated above and, after a reboot, seem to be getting correct `tilty`s showing up when inspecting events with `xinput test $pen_id`.

Brushes in krita are behaving well too.

jigpu, 2018-03-05:
Ouch, good find. Looks like I authored the offending commit (527fa95c29f6aee2c2d655e834ff3f45a9160c06) and was bit by the classic last-line effect...

Would you mind providing a copy of your fix? You can either send a git patch directly to the [linuxwacom-devel mailinglist](https://sourceforge.net/projects/linuxwacom/lists/linuxwacom-devel), attach a git patch to this bug, or make a pull request with the fix. In any case, please be sure to include the following two lines at the end of your commit message for tracking purposes:

```
Fixes: 527fa95c29 ("Refactor coordinate averaging to seperate function")

```

daniel-ferguson, 2018-03-05:
^ Sorry for the noise, github's commit message link formatting surprised me a few times.

jigpu, 2018-03-26:
Fix has been included with the 0.36.1 release.

closed 2018-03-26 by jigpu