linuxwacom/xf86-input-wacom #14: Disable cursor stop when lifting the pen.

thonkang, 2018-03-26:
When touch the pen to the tablet and then lift the pen a short distance, the cursor stops moving for a brief moment, even though the pen is still in range of the digitizer. Is there a way I can disable this "feature"? I have a CTL-480 tablet if it matters.

jigpu, 2018-03-26:
There is no "feature" that should be causing the kind of behavior that you describe. The pen should continue smoothly moving as you pick up the pen. The most likely cause is with the hardware or kernel driver rather than xf86-input-wacom.

We have a script which you can run to simultaneously capture events hardware, kernel, and xf86-input-wacom. You'll first need to install the following utilities:

 * **hid-recorder:** see https://bentiss.github.io/hid-replay-docs/
 * **evemu-record:** run `apt-get install evemu-tools` or `yum install evemu`
 * **xinput:** run `apt-get install xinput` or `yum install xorg-x11-server-utils`

Next, download and run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/5ae43a4a1a9b770822479cbabdf1ffd24e3685bb/capture.sh). It will ask you to choose a device to record: select the "stylus" device. Next, make a few strokes on the tablet with the stylus so that the issue is recorded. When you're done, press CTRL+C to stop the recording. Archive the three generated files into a tarball or ZIP file and upload the result here.

A few questions as well:
 * How often does this happen when you lift the pen a short distance off of the tablet surface? Always?
 * About how long does the cursor stop moving for?
 * What distribution (with version number) are you using?
 * What kernel are you using (run `uname -r` if you are unsure)?

thonkang, 2018-03-26:
> How often does this happen when you lift the pen a short distance off of the tablet surface? Always?
Always, every time I lift it. I can be hovering it within the digitizer range (not touching) and it still happens.
> About how long does the cursor stop moving for?
Half a second
> What distribution are you using?
Arch Linux
>What kernel are you using?
4.15.12-1-ARCH

[Here's the capture.](https://github.com/linuxwacom/xf86-input-wacom/files/1848798/capture.tar.gz)

jigpu, 2018-03-27:
Thanks for the details. Unfortunately I don't see any evidence of stutter in the hardware or kernel logs like I would have expected. I've also replayed the kernel log on an Arch Linux box and don't see any obvious stuttering. The X11 log doesn't contain timestamps, so its not much use -- I'll have to edit the capture script to add its own timestamps.

I'll let you know once I have an updated capture script that can gather timestamps from X11. Hopefully it reveals some better information.

I highly doubt it makes any difference, but what window manager / desktop environment are you using? Are you able to see if the problem persists within a minimalist WM like fluxbox?

thonkang, 2018-03-27:
I'm using gnome. Even though wayland is default now I just start it with Xorg when I use the tablet. Tried fluxbox, same results.

I just made a circling motion and lifted it up and you can see it noticably hang, here's a gif of it happening:
[image]

jigpu, 2018-04-06:
Sorry for the delay. I've updated the "capture.sh" script. Please download an updated copy from [here](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh) and gather a new set of logs. This will ensure we get timestamps from the xinput logs as well, letting us compare them to the device and kernel event timestamps.

Additionally, could you check bug #17? Looking through your last capture again, I do see a few occasions where the pen goes out of proximity for about as long as you're experiencing. It could be that you're accidentally leaving proximity even though you don't think you are. That bug links to a patch which *might* help...

999gary, 2018-04-09:
Just wanted to say I had the same issue with my friends ctl-480 and the modded driver fixed it right up, thank you

jigpu, 2018-04-17:
Glad to hear that @999gary! Any idea if it also works for you @thonkang?

By any chance, do either of you have a pen with an eraser? I'd like to make sure that it still functions properly with the patch applied. From the thread mentioned in bug #17:

> If your pen has an eraser, it would be particularly useful to know if that is still functioning properly, 
> especially if you bring the eraser into prox slowly. I have a feeling the modified code may sometimes 
> (always?) end up treating the eraser like the stylus when that happens (I assume the tablet will starts 
> sending position data before it knows it knows wether the tip or eraser end is being used -- one of the 
> hazards of "in range").
> 
> Probably the simplest way to test would be to open two terminals and run xinput test <stylus_id> and 
> xinput test <eraser_id> in each. Then, use the eraser for a bit. If everything is working right, only the 
> window ran with the eraser ID should ever print anything out. If there's any output from the stylus ID 
> window, that means I'll have at least one more modification to make...

999gary, 2018-04-18:
just checked with him and unfortunately no, a few more things, first that i dont use wacom although im considering getting one and this is my friends ctl480 and second is that this is the most popular tablet for people who play the game osu due to windows driver things i dont know too much about, anyway i hope this gets fixed, thank you for the work you do on these drivers

thonkang, 2018-05-02:
I apologize for taking so long to get back with a capture, I've been travelling without access to my computer and due to my mistake I had to install manjaro again.

Anyway, [here's the capture.](https://github.com/linuxwacom/xf86-input-wacom/files/1965789/capture2.zip) I did go out of range a few times but I was still experiencing jumps when it was in range. I'm judging this from when the light goes off on the tablet, which is when the pen stops tracking in windows, but it cuts out on linux even when the light is still on.

I don't have an eraser, and my friend that does have one apparently never has this problem in the first place so I'm not sure what that means.

I'm going to try building the modified driver and see if that fixes it.

jigpu, 2018-05-08:
My own apologies for the delay in responding. The tablet's light will turn off only when the pen goes completely out of prox, but will stay lit if the pen is in the (closer) "out of range" state. Your observations match up with my suspicion: Windows properly handles the "out of range" state and keeps the pen moving, while our driver mis-interprets "out of range" as fully "out of prox".

Please do build and test the modified driver. I strongly suspect that it will fix the jumping issue you're experiencing.

It would be a huge help if you can borrow your friend's pen for a while to do the eraser tests I requested on the modified driver. At this point, the lack of any testing with the eraser is what is stopping me from actually making this fix part of the official driver; I'm afraid that it will fix the jumps but break the eraser...

thonkang, 2018-05-08:
I've uninstalled xf86-input-wacom and built and installed the driver from https://github.com/jigpu/input-wacom/tree/bamboo-inrange and the hover issue appears to be fixed! The xsetwacom command no longer exists so not sure how I can change my area though...

I've talked to my friend and I'll have his pen tomorrow so I can test with it. He said the pressure sensitivity on the tip is broken but the eraser still works fine so it should be fine for testing. What exactly do I need to test with it, just that it's detected or do I need to run a capture with it?

jigpu, 2018-05-09:
You only needed to install the "input-wacom" driver from source. Removing xf86-input-wacom is the reason the "xsetwacom" command no longer exists. Go ahead and re-install xf86-input-wacom before continuing to be sure that everything works as expected :)

First, get an updated copy of the [capture.sh script](https://gist.github.com/jigpu/b7b47e4bede584cec8562f666fd84af4). I've made a few modifications to it that give me a little more information. Next, run the updated script and select the "stylus" device when prompted. Now use the device for a few minutes, switching back and forth between the stylus and eraser (you can just navigate around your desktop, doodle in GIMP, play a round of Osu -- anything really as long as you occasionally switch between the pen and eraser so we have data for both tools). Try to *slowly* bring the eraser into proximity at least a few times to be sure that we have a recording of what happens in that case. Once you've captured a few minutes worth of data, press CTRL+C on the capture script. It should automatically create a tarball of all the logs for you; just attach that to this bug for review.

I'll check the logs to see if there's any indication of the input-wacom driver becoming confused about whether the stylus or eraser is being used. If everything is fine, I'll go ahead and package the patch up for inclusion in the next driver release :)

thonkang, 2018-05-09:
[Here's the capture.](https://github.com/linuxwacom/xf86-input-wacom/files/1989090/record_1525891243.tar.gz) Let me know if there's anything else you need.

jigpu, 2018-05-18:
Thanks for the capture! It does indeed appear that my branch was causing problems with the eraser. I've fixed the issue and am sending the final version to the Linux kernel mailinglist. Once it has been accepted I'll port the code back to all of the kernel trees in this input-wacom driver. Thanks to everyone for their help tracking this down.

jigpu, 2018-05-30:
Issue should be resolved in the 4.18 kernel that should be released in mid- to late August. The fix has been backported to the input-wacom tree as well and should be available in its next release (0.41.0).

jigpu, 2018-07-06:
Issue has been fixed in the input-wacom 0.41.0 release. Still pending the Linux 4.18 release.

jigpu, 2018-08-20:
Linux 4.18 released.

closed 2018-08-20 by jigpu