linuxwacom/xf86-input-wacom #15: Unable to use wacom tablet after install the driver in my machine

Carolhug, 2018-03-28:
Hi, Just encounter an odd situation, I deployed this driver in two machines, both of them are Centos 3.10.0-693.11.1.el7.x86_64. tablet model:pth-660, 
After combined the driver use the same steps which you provided in the main page, one machine is working, but another machine still can't use the tablet properly. Debugged for a while still can't find any solution to fix it, Could you please help to take a look? 
Use "dmesg"  command and print the output, it already recognized the device:  
`[ 2368.035977] usb 1-4: USB disconnect, device number 7
[ 2374.132641] usb 1-4: new full-speed USB device number 8 using xhci_hcd
[ 2374.298847] usb 1-4: New USB device found, idVendor=056a, idProduct=0357
[ 2374.298852] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2374.298856] usb 1-4: Product: Wacom Intuos Pro M
[ 2374.298858] usb 1-4: Manufacturer: Wacom Co.,Ltd.
[ 2374.298860] usb 1-4: SerialNumber: 7JQ00A1012349
[ 2374.303690] input: Wacom Co.,Ltd. Wacom Intuos Pro M Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:056A:0357.000B/input/input23
[ 2374.303940] wacom 0003:056A:0357.000B: hidraw3: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom Intuos Pro M] on usb-0000:00:14.0-4/input0
[ 2374.306231] wacom 0003:056A:0357.000C: Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro M'. Ignoring.
[ 2374.306288] usb-storage 1-4:1.3: USB Mass Storage device detected
[ 2374.306620] scsi host9: usb-storage 1-4:1.3
[ 2375.307850] scsi 9:0:0:0: Direct-Access     Wacom    Intuos Pro M     1.37 PQ: 0 ANSI: 2
[ 2375.308427] sd 9:0:0:0: Attached scsi generic sg1 type 0
[ 2375.308852] sd 9:0:0:0: [sdb] 65536 512-byte logical blocks: (33.5 MB/32.0 MiB)
[ 2375.309093] sd 9:0:0:0: [sdb] Write Protect is on
[ 2375.309097] sd 9:0:0:0: [sdb] Mode Sense: 00 00 80 00
[ 2375.309379] sd 9:0:0:0: [sdb] Asking for cache data failed
[ 2375.309382] sd 9:0:0:0: [sdb] Assuming drive cache: write through
[ 2375.393425] sd 9:0:0:0: [sdb] Attached SCSI removable disk
`
But in the log, it keep saying this error: 
`journal: JS LOG: Unable to mount volume Wacom Start: Gio.IOErrorEnum: Error mounting /dev/sdb at /run/media/lcong/Wacom Start: Command-line `mount -t "exfat" -o "uhelper=udisks2,nodev,nosuid,uid=15001,gid=20003,iocharset=utf8,namecase=0,errors=remount-ro,umask=0077" "/dev/sdb" "/run/media/lcong/Wacom Start"' exited with non-zero exit status 32: mount: unknown filesystem type 'exfat`

"xsetwacom --list" can't find my tablet. "lsmod" can find the driver is loaded: 
`$lsmod  | grep wacom
wacom                  62424  0 
hid_wacom              91810  0 
wacom_w8001            12982  0 
`

Any suggestion will be appreciated 

Thanks.

jigpu, 2018-03-29:
Could you please provide the output of the following command from each machine? I suspect that the two machines might be running different versions of the driver, which could explain the difference in behavior.

~~~
$ grep -H "" /sys/module/*wacom/version
~~~

Carolhug, 2018-03-30:
Thanks Jason Gerecke, I just checked these two machines and got below
output, seems the driver is not installed properly in another machine, I'll
try to clean the driver and install it again!!!
Right machine:
--->$grep -H "" /sys/module/*wacom/version
/sys/module/hid_wacom/version:v2.00-0.39.0.1.g6cba8da
/sys/module/wacom/version:v1.53-0.39.0.1.g6cba8da

Problem in another machine:
$grep -H "" /sys/module/*wacom/version
grep: /sys/module/*wacom/version: No such file or directory

Is there a convenient way to install the driver? I have more than 20
machines, do I have to compile the driver in each of the machine?

Thanks!

Carolhug, 2018-04-03:
> First, just a quick note about the grep command output. The "No such file
> or directory" warning indicates that the module isn't loaded. Make sure
> that you have a tablet connected to the system when running the command.
>
> As for installing the driver on multiple machines, as long as all the
> machines share the same architecture (e.g. x86_64) and kernel, it should be
> possible to compile the driver on a single machine and install the result
> onto the rest. You could potentially try running the following to generate
> an archive of all the necessary files. It goes without saying that you
> should make sure the driver is working properly before creating the archive
> for other machines.
>
> $ tar -czPf wacom_$(uname -r).tar.gz /lib/modules/$(uname -r)/extra/*wacom* /etc/depmod.d/input-wacom.conf
>
> With the tarball in hand, you can use the following commands to extract
> and install it onto other machines running the same architecture/kernel (be
> sure to pass the 'P' option to tar!):
>
> $ sudo tar -xvzPf wacom_$(uname -r).tar.gz
> $ sudo depmod -a $(uname -r)
> $ sudo dracut -f /boot/initramfs-$(uname -r)
>
> After a reboot, the system should start using the updated driver.

Hi Jason,

Just realized the machine which didn't work is 3.10.0-693.21.1, the one is
working is kernel-3.10.0-693.11.1.el7.x86_64. Maybe that's the reason why
the tablet is not working. Ant now even the driver is installed property,
but it is still can't recognize the tablet, could you please help to
confirm if this driver can work for kernel 3.10.0-693.21.1?
# grep -H "" /sys/module/*wacom/version
/sys/module/hid_wacom/version:v2.00-0.39.0.1.g6cba8da
/sys/module/wacom/version:v1.53-input-wacom-master

Error log:
Apr  2 11:29:33 mtp-probe: checking bus 1, device 5:
"/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Apr  2 11:29:33 mtp-probe: bus: 1, device: 5 was not an MTP device

I'm going to try your suggestion to deploy the driver to other machines.

Thanks for your help!! Really appreciated

jigpu, 2018-04-06:
Even tiny differences in the kernel version can prevent the driver from working across different systems. You'll either need to compile the driver for each different kernel or update your systems to use the exact same kernel.

The reported version number is still not quite what I'd expect (the "wacom" version should have an actual commit ID rather than "master") but it might just be an artifact of how you built the driver... The current "master" branch (as well as 0.39.0.1.g6cba8da) is compatible with the PTH-660, which is why you see it working on your other system.

Could you please provide the last few lines of `dmesg` from the broken system after plugging in the tablet? I'm wondering if it is still logging an "Unknown device_type" message. That particular message indicates a particular failure that can occur when the driver tries to set up the tablet.

jigpu, 2018-04-17:
I don't see anything in the dmesg log that indicates a problem. The device is being recognized and input device nodes are being created. Could you run `xinput list` for me?

Carolhug, 2018-05-07:
Sorry for the later response email, i'm working on another project and just
got it finished.

Hmm, there is no wacom hardware be recognized in that machine:
--->$xinput  list
⎡ Virtual core pointer                        id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                  id=4    [slave  pointer
(2)]
⎜   ↳ Logitech USB Optical Mouse                  id=9    [slave  pointer
(2)]
⎣ Virtual core keyboard                       id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard                 id=5    [slave  keyboard
(3)]
    ↳ Power Button                                id=6    [slave  keyboard
(3)]
    ↳ Power Button                                id=7    [slave  keyboard
(3)]
    ↳ Sleep Button                                id=8    [slave  keyboard
(3)]
    ↳ AT Translated Set 2 keyboard                id=10    [slave  keyboard
(3)]
But from the "demsg" output, we can see the wacom is attached and
recognized  it is a  Wacom Intuos Pro M model.

Any advise will be appreciated!

Thanks.

jigpu, 2018-05-08:
Very interesting. That is indeed quite strange... I think it would be useful to gather some more in-depth logs from your system. Please run our [sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh) and attach the generated tarball to this bug. It will contain various system logs, as well as snapshots of key portions of sysfs that might give us a better idea of what's going on.

Carolhug, 2018-05-09:
Hi Jason,

I had run the sysinfo.sh script, got the generated tar file, please kindly
check the attached file.

Thanks.

jigpu, 2018-05-09:
I don't see any indication of the attachment... Maybe try attaching it directly through Github rather than email (just drag and drop the file onto the Github's reply box)?

Carolhug, 2018-05-14:
[sysinfo.aKUcwhOGri.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/1999301/sysinfo.aKUcwhOGri.tar.gz)

Uploaded again, could you please check? 
Let me know if you still can't see the attachment

jigpu, 2018-05-18:
Thanks. It looks like that log may have been captured from within a Wayland session, which might be some of the problem. GNOME has some rough support for tablets when using Wayland, but not any other compositors. Its best to use an Xorg session regardless of the desktop at this time (your prior `xinput list` indicates that you might have switched from Xorg to Wayland recently?)

I also see that the Xorg.0.log contains an error that explains why it the tablet wasn't showing up for you, even under Xorg. The error is "module ABI major version (24) doesn't match the server's version (21)". My guess is that this is because you copied the xf86-input-wacom driver to this machine from a different one. Similar to how you can only copy the binary kernel module to systems with the exact same kernel, you are limited in what systems you can copy the "wacom_drv.so" X11 driver to.

Please try reinstalling the xf86-input-wacom driver, either from the package respository (`sudo yum reinstall xorg-x11-drv-wacom`) or after building it from source on this machine.

Carolhug, 2018-05-21:
WOW, after I run (yum update xorg-x11-drv-wacom.x86_64) command upgraded
all the xorg packages, now the wacom is working!

For some reason this machine's xorg packages are still the old version, but
the kernel is the new one. Not sure why, so I just upgraded all of the xorg
packages.

Thanks very much!!!!!

Finally it's working.

jigpu, 2018-05-21:
Phew! Glad that did the trick :) It sounds like I can mark this as closed for now, but feel free to re-open it if necessary.

closed 2018-05-21 by jigpu

Carolhug, 2018-05-22:
Yes, please feel free to close that ticket. You saved my day! Really
appreciated!

Thanks.