linuxwacom/xf86-input-wacom #17: When the pen is moving over the tablet area very quickly, it skips over large distances

goeo-, 2018-04-03:
I'm having a problem very similar to this bug report:  https://sourceforge.net/p/linuxwacom/bugs/280

My tablet is a ctl-480s-n, and I'm running xf86-input-wacom 0.36.1-1, with the X.org X server 1.19.6, on linux 4.15.7-1-ARCH.

The issue is the same thing described in that bug report, possibly also a copy of #14: like demonstrated in the video sent by the person who reported that bug, when I move my pen quickly while it's not touching the tablet (drawing circles is a good way to reproduce this), it sometimes stops, and skips to another location.

Sorry if my explanation sounds weird, but the video posted in the original bug report explains it pretty well. The only difference I'm having is that my jumps happen in the top left, while the jumps seem to happen in the bottom right in the video.

Also, the RawSample setting seems to make the bug more consistent when set higher, while slowing down movement. This may make it easier to reproduce the bug, too.

goeo-, 2018-04-06:
Everyone noticing this problem seem to be osu! players, haha. You're absolutely right and I had misdiagnosed the issue, it was definitely about the maximum range being different than on windows; I installed the modified driver you mentioned, and now everything works great! Now I can pretty much remove windows. Thanks a lot!

jigpu, 2018-04-17:
Glad to hear! By any chance, does your pen have an eraser? I'd like to make sure that it still functions properly with the patch applied. From the mentioned thread:

> If your pen has an eraser, it would be particularly useful to know if that is still functioning properly, 
> especially if you bring the eraser into prox slowly. I have a feeling the modified code may sometimes 
> (always?) end up treating the eraser like the stylus when that happens (I assume the tablet will starts 
> sending position data before it knows it knows wether the tip or eraser end is being used -- one of the 
> hazards of "in range").
> 
> Probably the simplest way to test would be to open two terminals and run xinput test <stylus_id> and 
> xinput test <eraser_id> in each. Then, use the eraser for a bit. If everything is working right, only the 
> window ran with the eraser ID should ever print anything out. If there's any output from the stylus ID 
> window, that means I'll have at least one more modification to make...

closed 2018-04-17 by jigpu