linuxwacom/xf86-input-wacom #20: unable to use ctl-471

qcmiao1998, 2018-05-07:
I cannot use Wacom CTL-471 on Ubuntu 18.04 LTS.

skomra, 2018-05-07:
Hi qcmia1998, 

Just to confirm:

1. When you say you can't use CTL-471, do you mean the pen won't move the cursor? 
2. Also can you confirm that the VID:PID (`lsusb`) for this device is 056a:0300 ("Bamboo One")?
3. You are using "Ubuntu" not "Ubuntu on Wayland" from the sign in options.

jigpu, 2018-05-08:
Similar to my request in https://github.com/linuxwacom/libwacom/issues/22 would you mind running [our sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh) and attaching the generated tarball to this thread? It will contain several key logs that might give a clue about what is going on.

closed 2018-06-04 by qcmiao1998