linuxwacom/xf86-input-wacom #21: How do you fix aspect ratio?

Pshemas, 2018-05-28:
I'm trying to make old Intuos 3 tablet (PTZ430) work on Linux. It kinda works but is not really usable.
I've mapped it one of the monitors (out of 2 - one is 2k, other 1k) using the command:

xsetwacom set "Wacom Intuos3 4x5 Pen stylus" maptooutput 2560x1440+0+0

But the aspect ratio in wrong. This old tablet has 4:3 aspect ratio, while my monitors are 16:9. So when I'm trying to draw a circle I get ellipse instead.
Tried digging through manual :
https://github.com/linuxwacom/xf86-input-wacom/wiki/Calibration

- but I still have no clue what to enter to get correct aspect ratio.
Isn't there some sort of toggle/switch like on Windows?

jigpu, 2018-05-30:
Glad you managed to figure out the math -- it is indeed a little tricky, especially when trying to combine it with "MapToOutput". The GNOME / Unity / Cinnamon control panels provide the ability to fix the aspect ratio when mapping an output, but not xsetwacom.
Years ago I wrote a patch for xsetwacom (see https://github.com/jigpu/xf86-input-wacom/tree/match_aspect_ratio) which would have added a special flag for MapToOutput which would correct the aspect ratio, but it never made it into the project. Looking back through my emails, it doesn't look like there were any show-stopping problems with the patch, but it did need some work. It might be a good idea to update and resubmit it...

fosskers, 2018-06-21:
Yes please.

fosskers, 2018-06-25:
For people who wander in here from the future: the instructions for fixing aspect ratio have been updated [on the Arch Linux wiki](https://wiki.archlinux.org/index.php/Wacom_tablet#Adjusting_aspect_ratios).

In my case, this command correct things:
```text
xsetwacom --set 9 Area 0 0 15200 8550
```

tb2097, 2019-01-31:
Rather that doing things by command line, I have written a GUI that makes it a lot easier to save these types of configs and have them auto-load on login.  If you would like to give it a try, I'd definitely ben interested in how it works on Arch as I haven't done so yet.  Please try the latest 0.3.x release as it's the most feature complete version.
https://github.com/tb2097/wacom-gui/releases/

laino, 2019-07-29:
I hacked together a script that does just that. 

Usage: ```mapwacom "Wacom Intuos BT M Pen stylus" eDP1```

Don't forget to change the area values in the "Configuration" section to match your tablet:

```bash
#!/bin/sh
set -e

##### CONFIGURATION

AREALEFT=0
AREATOP=0
AREARIGHT=21600
AREABOTTOM=13500

######

WACOM="$1"
OUTPUT="$2"

AREAWIDTH=$(echo $AREARIGHT-$AREALEFT | bc -l)
AREAHEIGHT=$(echo $AREABOTTOM-$AREATOP | bc -l)

round(){
    echo $(printf %.$2f $(echo "scale=$2;(((10^$2)*$1)+0.5)/(10^$2)" | bc))
};

$(
xrandr | perl -n -e'/^'$OUTPUT' connected(?: primary) ([0-9]+)x([0-9]+)?/ &&'\
'print "export SCREENWIDTH=$1\nexport SCREENHEIGHT=$2"'
)

RESULTWIDTH=$(echo $AREAHEIGHT/$SCREENHEIGHT*$SCREENWIDTH | bc -l)
RESULTHEIGHT=$(echo $AREAWIDTH/$SCREENWIDTH*$SCREENHEIGHT | bc -l)

if (( $(echo $RESULTWIDTH \> $AREAWIDTH | bc -l) )); then
    RESULTHEIGHT=$(echo $AREAWIDTH/$RESULTWIDTH*$AREAHEIGHT | bc -l)
    RESULTWIDTH=$AREAWIDTH
else
    RESULTWIDTH=$(echo $AREAHEIGHT/$RESULTHEIGHT*$AREAWIDTH | bc -l)
    RESULTHEIGHT=$AREAHEIGHT
fi

RESULTRIGHT=$(round $(echo $AREALEFT + $RESULTWIDTH | bc) 0)
RESULTBOTTOM=$(round $(echo $AREATOP + $RESULTHEIGHT | bc) 0)

xsetwacom set "$WACOM" MapToOutput $OUTPUT
xsetwacom set "$WACOM" Area $AREALEFT $AREATOP $RESULTRIGHT $RESULTBOTTOM
```

Pinglinux, 2022-08-03:
Thank you all for providing solutions to adjust the tablet aspect ratio. Since most distributions are using Wayland as their default display server, all existing Xorg-based features will be in maintenance mode. No new features will be considered.

closed 2022-08-03 by Pinglinux