linuxwacom/xf86-input-wacom #22: Automatically disabling touchscreen when recognizing pen

AlessandroW, 2018-06-10:
Hello,
when using xournal my palm sometimes interrupts the writing process.
Resulting in partial letters / ignored lines.

Specs: Lenovo Yoga 720
`
xsetwacom --list  
`
>Wacom HID 50EE Pen stylus               id: 10  type: STYLUS      
>Wacom HID 50EE Pen eraser               id: 14  type: ERASER      
>Wacom HID 50EE Finger touch             id: 11  type: TOUCH    

Quick solution: `xinput disable "Wacom HID 50EE Finger touch"` . But in tablet-mode this isn't quite useful.
Alternative approach: map above command to a pen-button.
Desired solutoin:
 ` if pen_is_writing`
` touch = disabled`
 `  else `
`     touch = enabled`

Is something like that already implemented, or even possible?

jigpu, 2018-06-13:
The driver does implement basic palm rejection and should disable touch input whenever the pen is nearby. Touch input should also remain disabled even when the pen is completely removed from proximity, so long as the touchscreen reports that its tracking a touch that was present while the pen was in proximity (this is to prevent issues that occur when resting your hand on the screen and lifting the pen "too high" between some letters or words).

You can test the palm rejection by bringing the pen into proximity (close enough that it moves the cursor), and then touching the screen with your finger. The cursor shoudl not react to the finger's presence. Next, continue holding your finger to the screen and remove the pen from proximity. The cursor should still not react to your finger's presence, or even the addition of new fingers until you remove *all* fingers from the screen.

If the above test results don't match up, let us know so we can diagnose further.

AlessandroW, 2018-06-19:
You were right, everything worked liked described.
I guess it is a Xournal thing,  curiously Xournal itself has a "pen disables touch" property which was disabled when a opened the issue.
Anyway the driver does his job well and with the Xournal option enabled I didn't encounter any problem

Thanks for your help!

closed 2018-06-19 by jigpu