linuxwacom/xf86-input-wacom #23: gnome-settings-daemon missing in gnome 3.25

lee11lee, 2018-06-18:
Hi,

I'm trying to set AbsWheelUp and AbsWheelDown to scroll up and down. 
In the past on Gnome version 3.22 I could do that via: 

- disabling the wacom plugin (otherwise you can not map mouse actions to the pad) via: 
`gsettings set org.gnome.settings-daemon.plugins.gsdwacom active false`

- then hardcoding the two buttons via:
```
xsetwacom set "Wacom Intuos4 8x13 pad" AbsWheelDown 5
xsetwacom set "Wacom Intuos4 8x13 pad" AbsWheelUp 4
```

Now in the new gnome version 3.25.4 the setting to disable the wacom plugin is not there. The only settings I have in the plugins folder are:
```
$ gsettings list-children org.gnome.settings-daemon.plugins
housekeeping   org.gnome.settings-daemon.plugins.housekeeping
power          org.gnome.settings-daemon.plugins.power
color          org.gnome.settings-daemon.plugins.color
media-keys     org.gnome.settings-daemon.plugins.media-keys
xsettings      org.gnome.settings-daemon.plugins.xsettings
```

I also tried searching the dconf for Wacom settings, but with no results.

As I understand, there has been a major change to the wacom settings in gnome 3.24..

How can I still set the buttons to scrolling and bypass the gnome wacom plugin?

Thank you, any help would be greatly appreciated!

lee

jigpu, 2018-06-19:
I haven't had much luck finding an alternative to the `org.gnome.settings-daemon.plugins.gsdwacom active` property. Recent versions of GNOME have moved most of the Wacom logic out of gnome-settings-daemon and into Mutter. Along with this, most of the configuration has moved to be under the `org.gnome.desktop.peripherals` tree. There doesn't seem to be a way to switch GNOME's handling of tablet events on and off anymore.
Based on your xsetwacom commands, I assume your fundamental issue is that GNOME doesn't provide a way for the touch ring to send scroll events. I would file an issue against the [gnome-control-center tracker](https://gitlab.gnome.org/GNOME/gnome-control-center/issues) requesting that the software allow you to send scroll events (or whatever it is that you want the ring to do).

lee11lee, 2018-06-19:
Thank you for your help!

I've posted an issue as suggested:
(https://gitlab.gnome.org/GNOME/gnome-control-center/issues/118)

closed 2018-06-28 by jigpu