linuxwacom/xf86-input-wacom #26: Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro M' error debug on Centos 7.2.1511

Carolhug, 2018-08-06:
Hi guys, 
I'm trying to install the driver for my my Wacom Intuos Pro M to Centos 7(kernel: 3.10.0-693.el7.x86_64) machine, there is no error during the driver building, but I still can't get the device work, from the "dmesg" command, it is saying "Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro M'. Ignoring." But from "xinput list" or "xsetwacom --list" command, it seems finded the right driver and can recognize the tablet well, please kindly check below output
[root@greyelephant ~]# dmesg | grep -i wacom
[   10.902942] usb 3-13: Product: Wacom Intuos Pro M
[   10.902944] usb 3-13: Manufacturer: Wacom Co.,Ltd.
[   11.909045] usbcore: registered new interface driver wacom
[   11.918517] input: Wacom Co.,Ltd. Wacom Intuos Pro M Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-13/3-13:1.0/0003:056A:0357.0004/input/input5
[   11.918667] wacom 0003:056A:0357.0004: hidraw3: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom Intuos Pro M] on usb-0000:00:14.0-13/input0
[   11.918913] wacom 0003:056A:0357.0005: Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro M'. Ignoring.
[root@greyelephant ~]# xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech USB Optical Mouse              	id=9	[slave  pointer  (2)]
⎜   ↳ Wacom Intuos Pro M Pen cursor           	id=14	[slave  pointer  (2)]
⎜   ↳ Wacom Intuos Pro M Finger touch         	id=15	[slave  pointer  (2)]
⎜   ↳ Wacom Intuos Pro M Pen eraser           	id=13	[slave  pointer  (2)]
⎜   ↳ Wacom Intuos Pro M Pen stylus           	id=12	[slave  pointer  (2)]
⎜   ↳ Wacom Intuos Pro M Pad pad              	id=11	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=10	[slave  keyboard (3)]
[root@greyelephant ~]# grep -H "" /sys/module/*wacom/versions
grep: /sys/module/*wacom/versions: No such file or directory

There are no commid ID in the "wacom" version: 
[root@greyelephant ~]# grep -H "" /sys/module/*wacom/version     
/sys/module/hid_wacom/version:v2.00
/sys/module/wacom/version:v1.53-0.39.0

[root@greyelephant ~]# xsetwacom --list
Wacom Intuos Pro M Pen cursor   	id: 14	type: CURSOR    
Wacom Intuos Pro M Finger touch 	id: 15	type: TOUCH     
Wacom Intuos Pro M Pen eraser   	id: 13	type: ERASER    
Wacom Intuos Pro M Pen stylus   	id: 12	type: STYLUS    
Wacom Intuos Pro M Pad pad      	id: 11	type: PAD   

I'm not sure how to debug more about it, any advise will be appreciated. 

Thanks.

Carolhug, 2018-08-13:
Have no clue how to fix it after debugged for a while, would you please help? 

Thanks

jigpu, 2018-08-13:
My apologies. This slipped off my radar...

The "Unknown device type" error should be resolved by version 0.39 of input-wacom, which you appear to be using. I think, however, that you may have only partially installed the driver, as evidenced by your `grep -H "" /sys/module/*wacom/version` output. Specifically, it looks like only the "wacom" driver has been updated to version 0.39; the "hid_wacom" driver (which would be responsible for making your Intuos work) still appears to be the stock version.

Did you run `make install` from inside the input-wacom directory, or did you copy over the compiled ".ko" module manually? Definitely try running `make install` if you had been copying manually.

Carolhug, 2018-08-14:
Oh, thank you! I know why this driver hasn't installed success, my mistake.
I'm copied the install command from your instruction "if test -x
.autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make
install || echo "Build Failed". I'm using root, so I should remove the
sudo from the command. But unfortunately, I didn't.

Now, after manully run "make install", the driver is installed succeed.
I'll try again to see if my new Wacom can work properly

Thank you so much!
- Carol

jigpu, 2018-08-14:
I wouldn't have thought an unnecessary 'sudo' could cause a problem, but that's good to know for the future. Definitely let us know if resolves the problem with your Intuos :)

Carolhug, 2018-08-15:
hmm, still doesn't work after reboot.
Same error message from "dmesg" command: unknow device_type for 'Wacom
Co.,Ltd. Wacom Intuos Pro M'. Ignoring.
from command "grep -H “” /sys/module/*wacom/version" output, we can see the
driver is already intalled:
/sys/module/hid_wacom/version:v2.00
/sys/module/wacom/version:v1.53-0.39.0

What can I do about this problem in the next? 🤔

jigpu, 2018-08-17:
Hmm. I think this might be an issue with the initramfs not being updated. We've had reports that the stock driver loads instead of the updated one if the tablet was connected when the system started. The latest version of the input-wacom driver should fix this. Could you please try downloading version 0.41.0 of the driver and then following the same build/install steps?

Carolhug, 2018-08-23:
After upgrading to the latest version of input-wacom to 0.41.0, it still
can't work. Then I tried to upgrade xf86-input-wacom package to 0.31, also
doesn't work.

Anything I can do to fix this 😫

jigpu, 2018-08-23:
How strange... I tried installing input-wacom 0.41.0 onto a fresh CentOS 7.4 system and didn't have any issues.

Go ahead and run our [sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh) and attach the tarball it generates to this issue. It will include several important logs and information dumps. It should hopefully give me an idea of what's unique about your setup that may be causing trouble.

Next, assuming the devices are still listed in `xsetwacom`, lets try capturing the tablet events as they make their way through the system. This will give an idea of where the data flow is being cut off. For that, run my [capture.sh script](https://gist.github.com/jigpu/b7b47e4bede584cec8562f666fd84af4). It depends on a few programs to work properly, but it should let you know how to get them if they're missing. Once all the dependencies are in place, it should ask you to choose a device to record from the list. You'll want to choose the "stylus" device and press Enter. Next, drag the pen across your tablet a few times. Finally, press CTRL+C to stop the recording and attach the log file that it tells you about to this bug.

Carolhug, 2018-08-24:
Here is the sysinfo log file, I'm not able to capture the device
information, xinput command can't list my wacom tablet now. please kindly
check the attached file. Hope this file can tell root cause.

Thanks.

Carolhug, 2018-08-27:
Attached the file via github page, please kindly check. 

[sysinfo.3YcW2YI6Bj.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/2322808/sysinfo.3YcW2YI6Bj.tar.gz)

Thanks.

Windrill, 2020-06-13:
Came across the same error after installing a different Ubuntu kernel. `xsetwacom` won't detect my wacom device, but `lsusb` showed the wacom device `Bus 001 Device 006: ID 056a:037a Wacom Co., Ltd `.

The problem was after reinstalling ubuntu's kernels, initramfs did not register correctly to start wacom's drivers. This corresponds to the 'input-wacom' driver. From the [docs](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) note that the installation procedure included the kernel headers as parameters, and during installing the logs indeed showed that initramfs were being updated.

Following the docs to install input-wacom a second time fixed the issue for me.

jigpu, 2021-07-21:
This has likely been fixed by the patches in https://github.com/linuxwacom/input-wacom/pull/257 which correct several issues with the installation of input-wacom on RHEL. The fixes should be part of the next release (input-wacom 0.48.0)

closed 2021-07-21 by jigpu