linuxwacom/xf86-input-wacom #27: Lenovo Yoga 730-15ikb no pen input

organburner, 2018-08-21:
Hi,
I'm having issues with my Lenovo Yoga 730-ikb not registering any input from the included Lenovo Active Pen 2 that came with it. Otherwise the touchsreen functions when using fingers.
I'm not too good at troubleshooting input problems as this is the first 2-in-1 device I've had that I've tried running linux on. I'm currently running Linux Mint with cinnamon but also tested a live ubuntu stick and had fedora installed earlier and they both displayed the same issues.
xinput and xsetwacom finds the devices
`x@x-Lenovo-YOGA-730-15IKB:~$ xsetwacom --list devices
Wacom HID 516B Pen stylus       	id: 12	type: STYLUS    
Wacom HID 516B Finger touch     	id: 13	type: TOUCH     
Wacom HID 516B Pen eraser       	id: 16	type: ERASER    
x@x-Lenovo-YOGA-730-15IKB:~$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ MSFT0001:02 06CB:7F8F Touchpad          	id=11	[slave  pointer  (2)]
⎜   ↳ Wacom HID 516B Pen stylus               	id=12	[slave  pointer  (2)]
⎜   ↳ Wacom HID 516B Finger touch             	id=13	[slave  pointer  (2)]
⎜   ↳ Wacom HID 516B Pen eraser               	id=16	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=10	[slave  keyboard (3)]
    ↳ Ideapad extra buttons                   	id=14	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=15	[slave  keyboard (3)]
`
But touching the screen with the stylus and pressing the buttons on it does nothing.
I figure I might try to live boot a windows 10 install to check if it works in windows if I get the time, which is something I should have done before but right now this computer is in daily use and I want to avoid reinstalling OS on it and for some reason this thing runs windows usb sticks extremely slowly.
Any help is appreciated.

jigpu, 2018-08-21:
Our [capture.sh script](https://gist.github.com/jigpu/b7b47e4bede584cec8562f666fd84af4) can be used to gather event logs from the device, kernel, and Xorg to narrow down where the problem may lie. You'll need to install a few dependencies for the script to run -- the script should tell you what is missing and how to install them.

When you run the script, it should ask you to select a device to record. Enter the id of the stylus device, press enter, and then drag the pen across the screen a few times. Afterwards, press CTRL+C to stop recording. The script will package everything up into a tar.gz file which should then attach to this bug.

organburner, 2018-08-21:
[record_1534867809.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/2307099/record_1534867809.tar.gz)
[record_1534867867.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/2307100/record_1534867867.tar.gz)
[record_1534867900.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/2307101/record_1534867900.tar.gz)
I tested the pen stylus and pen eraser, also with the debugfs option but it says no events were recorded despite me dragging the stylus all over and pressing all the buttons on it.

jigpu, 2018-08-21:
Odd... The logs (both those in the recording, and those in your sysinfo) all look fine aside from the lack of events. Are you sure the AAAA battery in your pen is fresh and installed the right way around (or, if your pen is one of the rechargeable ones, that it has been docked recently)? Note that many pens use a second battery for the Bluetooth module, so it isn't sufficient to simply check that the pairing LED light blinks.

organburner, 2018-08-21:
Well, uh, I've been working in IT for close to 8 years and in all those years I apparently never learned to check that the battery is in the right way around.
I guess I didn't think about that being possible since the light came on and it paired in bluetooth but as you said that's a separate module.
I am sorry for wasting your time and I greatly appreciate your help!

closed 2018-08-21 by organburner

jigpu, 2018-08-21:
No worries! Simple problems are sometimes the most elusive :)