linuxwacom/xf86-input-wacom #28: Invers or natural 2-finger scrolling

sphh, 2018-08-31:
Can you please help me? How do I enable invers or natural vertical scrolling for two finger scrolling?

I found a couple of outdated sites (e.g. https://help.ubuntu.com/community/Wacom#Fixing_the_Inverted_Scrolling_Problem), but none works for me.

I use Linux Mint 19 and a "Wacom Bamboo Fun small (2FG)" tablet.

Thank you so much for your help!!

jigpu, 2018-09-12:
My apologies for the delayed reply... Unfortunately, I don't think that this is possible at the moment. The driver doesn't expose an option for changing the gesture-triggered button presses like it does for for wheels and strips. Even trying to negate the scroll distance (to "trick" the driver into scrolling backwards) doesn't work.

Probably your best option for working around the issue would be to let libinput take control of the touchpad. You'll want to make sure that the "xserver-xorg-input-libinput" package is installed (it should already be), create a file named `/etc/X11/xorg.conf.d/90-libinput-wacom.conf` with the following contents, and then reboot:

~~~
Section "InputClass"
        Identifier "libinput Wacom touchpad override class"
        MatchUSBID "056a:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTouchpad "true"
        Driver "libinput"
EndSection
~~~

After you reboot the touchpad should be under the control of libinput and you should be able to configure its scrolling direction using libinput's own tools. If you're lucky, the "Natural Scrolling" option in the Mouse & Touchpad area of the control panel might control the behavior, but if not you should be able to use other tools like xinput to change the setting for the touchpad.

I'll leave this open, but marked as a feature request since it could be useful to have.

sphh, 2018-09-12:
Thank you for your reply. Very much appreciated.

I'll try it and report back, but that might be only in three weeks time, because I cannot find the time before.

juligreen, 2021-01-05:
@jigpu  Your Answer worked!
Overall with libinput controlling the touchpad it feels way better to use.
Thanks!

Pinglinux, 2022-08-03:
libinput provided the solution.

closed 2022-08-03 by Pinglinux