linuxwacom/xf86-input-wacom #30: Unable to set 27qhd brightness/colour channels without windows?

Kirtai, 2018-09-26:
I'm trying to set up a Cintiq 27QHDT under Debian. Is there any way to set the displays brightness or r/g/b colour channels without having to use the Windows wacom drivers?

This model of Cintiq lacks hardware controls for its display, so would it be possible to add software controls to the driver? Or do I have to set it up under windows? The windows drivers have a tool to adjust them.

Note, I'm not talking about the graphics card settings, rather the cintiqs internal settings)

I need these for setting the colour calibration.

jigpu, 2018-09-26:
I have some really hastilly-written code for the Cintiq Pro 24 which also lacks hardware controls. At the moment it can only set the brightness and contrast, but in theory it could also control other properties that the hardware allows. I don't know if the code is compatible with the Cintiq 27QHDT, but it might be a start...

https://gist.github.com/jigpu/e05b17e24a60825c6f3bf59fe9638736

Could you please provide the output of `lsusb` for me? The code above requires an "FTDI F232H" device connected over USB, which is how the Cintiq Pro 24 exposes its hardware controls. If the 27QHDT uses a different type of device then my code won't work.

Kirtai, 2018-09-26:
Here's the lsusb output. It does appear to have a FT232H listed when the 27qhdt is attached.
[cintiq27qhdt.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2419203/cintiq27qhdt.txt)

Kirtai, 2018-10-02:
Tried out that script and got the attached errors.
[ddc-ftdi.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2438415/ddc-ftdi.txt)

Kirtai, 2018-10-02:
I just found a utility called [ddcutil](https://www.ddcutil.com/) that can talk directly to the Cintiq.
Attached is what it produces for the 27QHDT using interrogate & detect. (it also includes my other two monitors but I didn't know what was safe to remove)

Hope this is useful, will be experimenting with it to see what it's capable of

[ddc-27qhd-capabilities.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2438606/ddc-27qhd-capabilities.txt)
[ddc-27qhd-interrogate.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2438607/ddc-27qhd-interrogate.txt)

jigpu, 2018-10-05:
Interesting! I came across ddcutil but it wasn't able to detect the Cintiq Pro for some reason, which is why I threw together a proof-of-concept of my own. I still can't seem to convince ddcutil to work, but maybe that's just an issue with my setup...

It looks like ddcutil detects what I'd expect for your monitor, so you should be able to use it to adjust the color channels. The man page indicates that something like `ddcutil setvcp 10 30 --display 2` should adjust the display brightness (feature 10 listed in your ddc-27qhd-capabilites.txt file) to the value 30, for example. The "video gain" of the R, G, and B channels are apparently features 16, 17, and 1A -- you might see if they work as well.

Let me know if you have any luck setting the values :)

Kirtai, 2018-10-07:
On Debian, I had to install the I2C kernel drivers and either run it as root or add my user to the i2g group to get ddcutil to work. It's not very clear that you need those drivers.

Testing shows that brightness and contrast are changeable with commands like "ddcutil setvcp 10 30 --display 2" but the RGB gain channels seem to be read only. I haven't figured out if I'm giving the wrong values, they need to be unlocked somehow or maybe they are just read only and the gains are changed some other way. Maybe the 6 axis saturation will work though I haven't had the time to test it.

jigpu, 2018-10-08:
Wonderful! After running `sudo modprobe i2c-dev` I now see my own Cintiq listed in the output from `sudo ddcutil detect`. Thanks! Like you, I haven't had any luck setting the gain, and saturation also doesn't seem to work. I'll see if my script has any better success (in case its some kind of timing issue with ddcutil) on those properties...

Kirtai, 2018-10-16:
I actually have the Wacom Color Manager and (unfortunately windows only) software that's supposed to configure the internal LUTs and other settings. I'd try it but I don't know of any way to trace the DDC messages sent to the display so I can see what it's doing. Any suggestions?

jigpu, 2018-10-16:
You could potentially try using Wireshark to capture the USB traffic when running the Wacom Color Manager inside a virtual machine (see [here](https://wiki.wireshark.org/CaptureSetup/USB)). You'll need to run `sudo modprobe usbmon` before starting Wireshark in order to let it see the USB data. If everything is set up right, you should see some data being sent back and forth when you make a change in the Wacom Color Manager. You could start a recording before making a change, make the change, and then stop and save the recording to isolate just that single feature for analysis.

I wouldn't really recommend trying this if you aren't already familiar with Wireshark (or interested in learning the basics of packet capture), but it could be an option. I'm a little too swamped with other work to dive into it myself unfortunately :( The DDCutil guys might be able to help out though, or even know of a more appropriate tool for tracing the DDC messages than this particular sledgehammer :)

Pinglinux, 2022-08-04:
This is beyond the xf86-input-wacom driver scope.

closed 2022-08-04 by Pinglinux