linuxwacom/xf86-input-wacom #31: Crashes on Ubutu 14.04

weiqiyang, 2018-10-09:
I have all the drivers for Wacom installed on my Ubuntu system. I can see my Intuos5 under `xinput`, and I can control my cursor by the tablet, but several things goes wrong after the installation. 

- The cursor flashes, and then there is a chance that the system crashes and bring me back to the log-in menu.
- Several shortcuts are disabled (e.g. < Super >+< Space >, < Ctrl >+< Alt >+T).
- The "System Settings" window terminates when I click on "All Settings > Wacom Tablet".

All these issues happens only when I log in the system with the tablet plugged in, or when I plug in the tablet after logged in.

My Ubuntu is 4.4.0-137-generic, with nvidia-390, and gnome-shell version is 3.10.4.

jigpu, 2018-10-09:
The last two issues are almost certainly bugs in the Unity desktop. I'd suggest filing a bug at https://bugs.launchpad.net/ubuntu/+source/unity-control-center with details. I wouldn't expect much though -- the Ubuntu developers haven't been great about fixing bugs or updating Unity :( If it's possible to upgrade to Ubuntu 18.04, that might be best...

I'll see if I can replicate the crash you mention in your first bullet point, but I'm pretty sure that's a Unity bug as well.

Pinglinux, 2022-03-19:
14.04 is not supported any more...

closed 2022-03-19 by Pinglinux