linuxwacom/xf86-input-wacom #32: Ubuntu 16.04(wacom CTL-4100WL Bluetooth Model)

lmayorga1980, 2018-10-11:
Using the default drivers from Ubuntu 16.04 `xserver-xorg-input-wacom` I get the following error on the Xorg.log  

```
[   217.787] (II) Using input driver 'wacom' for 'Wacom Co.,Ltd. Intuos BT S Pen'
[   217.787] (**) Wacom Co.,Ltd. Intuos BT S Pen: always reports core events
[   217.787] (**) Option "Device" "/dev/input/event22"
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'stylus' for this device.
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'eraser' for this device.
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'cursor' for this device.
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'touch' for this device.
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'pad' for this device.
[   217.840] (EE) Wacom Co.,Ltd. Intuos BT S Pen: No type specified
[   217.864] (EE) PreInit returned 8 for "Wacom Co.,Ltd. Intuos BT S Pen"
[   217.864] (II) UnloadModule: "wacom"
[  1460.378] (--) synaptics: DLL07BF:01 06CB:7A13 Touchpad: touchpad found
[  2205.819] (II) config/udev: Adding input device Wacom Co.,Ltd. Intuos BT S Pen (/dev/input/event22)
[  2205.819] (**) Wacom Co.,Ltd. Intuos BT S Pen: Applying InputClass "Wacom USB device class"
[  2205.819] (**) Wacom Co.,Ltd. Intuos BT S Pen: Applying InputClass "Wacom class"
[  2205.819] (II) Using input driver 'wacom' for 'Wacom Co.,Ltd. Intuos BT S Pen'
[  2205.819] (**) Wacom Co.,Ltd. Intuos BT S Pen: always reports core events
[  2205.819] (**) Option "Device" "/dev/input/event22"
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'stylus' for this device.
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'eraser' for this device.
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'cursor' for this device.
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'touch' for this device.
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: Invalid type 'pad' for this device.
[  2205.872] (EE) Wacom Co.,Ltd. Intuos BT S Pen: No type specified
[  2205.900] (EE) PreInit returned 8 for "Wacom Co.,Ltd. Intuos BT S Pen"
[  2205.900] (II) UnloadModule: "wacom"
```

But the utility `libwacom-list-local-devices` is able to give me some output.This is when connected to Bluetooth and not through USB.

** libwacom-list-local-devices --database /usr/share/libwacom**
```
[16:15:24] $ libwacom-list-local-devices --database /usr/share/libwacom
[Device]
Name=Wacom Serial Tablet WACf004
DeviceMatch=serial:0000:0000;
Class=ISDV4
Width=0
Height=0
IntegratedIn=Display;System;
Styli=0xfffff;0xffffe;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
---------------------------------------------------------------

```

**xinput list**
```
[16:16:39] $ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech M325                           	id=11	[slave  pointer  (2)]
⎜   ↳ DLL07BF:01 06CB:7A13 Touchpad           	id=13	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=10	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD                    	id=12	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=14	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=15	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=16	[slave  keyboard (3)]
    ↳ FaceTime HD Camera (Display)            	id=17	[slave  keyboard (3)]
    ↳ Logitech K810                           	id=18	[slave  keyboard (3)]

```

**dmesg | grep -i wacom**

```
[16:17:57] $ dmesg | grep -i wacom
[  217.078838] usb 1-2: Manufacturer: Wacom Co.,Ltd.
[  217.653518] input: Wacom Co.,Ltd. Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:056A:0376.000E/input/input25
[  217.653864] wacom 0003:056A:0376.000E: hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-2/input0
[ 2205.167740] usb 1-2: Manufacturer: Wacom Co.,Ltd.
[ 2205.172255] input: Wacom Co.,Ltd. Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:056A:0376.0010/input/input28
[ 2205.172513] wacom 0003:056A:0376.0010: hidraw6: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-2/input0
```

jigpu, 2018-10-12:
Although the USB portion of the tablet will work with as old as a 4.10 kernel, the Bluetooth functionality I believe specifically requires Linux 4.17 or later. If you [install our "input-wacom" driver](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source), the device should be detected and work correctly.

closed 2018-10-12 by jigpu

lmayorga1980, 2018-10-12:
I have `4.4.0-137-generic` should I remove the default ubuntu packages?
```
xserver-xorg-input-wacom/xenial,now 1:0.32.0-0ubuntu3 amd64 [installed]
libwacom2/xenial-updates,now 0.22-1~ubuntu16.04.1 amd64 [installed]
libwacom-common/xenial-updates,xenial-updates,now 0.22-1~ubuntu16.04.1 all [installed]
libwacom-bin/xenial-updates,now 0.22-1~ubuntu16.04.1 amd64 [installed]
xserver-xorg-input-wacom-lts-xenial/xenial-updates,now 3:13 amd64 [installed]
```