linuxwacom/xf86-input-wacom #36: Pan mode does nothing

nonnymoose, 2018-12-25:
I'm using a Monoprice-rebranded Huion H610 graphics tablet, which should be compatible (and so far has been) with the wacom X driver. However, when I try to set any of the buttons to put the tablet in pan mode (such as `xsetwacom --set "10594 Pad pad" Button 3 "pan"`), it just causes the button to do nothing instead. Whether I'm holding it down or not, clicking or dragging does nothing.

System info:

    Arch Linux
    Kernel: 4.19.12-arch1-1-ARCH x86_64
    xf86-input-wacom: 0.36.1-1 (installed from official repository)
    xorg-server: 1.20.3-1
    value of PanScrollThreshold: 12

nonnymoose, 2018-12-25:
I was just updating my configuration and I noticed that `xsetwacom` isn't setting the pan option:
```
» xsetwacom --set "10594 Pad pad" Button 3 "pan"
» xsetwacom --get "10594 Pad pad" Button 3
[empty line]
````

jigpu, 2018-12-26:
Try setting one of the stylus buttons to pan (eg `xsetwacom set "10594 Pen stylus" button 2 pan` [not sure if that's the right device name...])

I've never tried making one of the pad buttons to be pan but the stylus buttons have worked for me in the past.

nonnymoose, 2018-12-26:
@jigpu Wow, you got the device name exactly right! It actually does work with the stylus. Thanks!

The reason I didn't realized that before was I had tried to map it to button 2, and for some strange reason, trying to pan with button 2 clicks button 3 instead. Pan on button 3 works fine.
In light of that, I'm not sure if I should close this issue or not.

jigpu, 2018-12-26:
I'll have to have a look at the panning code when I get back to my computer at the start of the new year. I'll see if I can reproduce the button click issue and figure out why setting the pad to pan doesn't work...

jigpu, 2022-04-15:
Was never able to reproduce the button click issue. Setting pad to pan does not work due to how panning works (it requires motion from the device, but pads do not send motion events; removing this limitation would require some thought...)

closed 2022-04-15 by jigpu