linuxwacom/xf86-input-wacom #40: Unsupported Wacom model ~#KT-0405-R00 V1.3-6 (ArtPad II, serial)

vladki77, 2019-01-06:
Hello,

I have an ancient serial Wacom tablet ArtPad II. I remember it used to work maybe 10 years ago. Now I happend to find it again in box of old stuff and decided to try it again. But it seems it is no more supported. X-org driver just ignores it, and kernel driver says unsupported/ I have tried the following:

`# inputattach --wacom_iv /dev/ttyS0 `
module wacom_serial4 was loaded and the message in subject was logged in syslog
However I do not remember that this was needed before.

I have this xorg.conf.d snippet (copied from old xorg.conf from old backups):
```
Section "Module"
        Load    "wacom"
EndSection

Section "InputDevice"
        Identifier   "Tablet Stylus"
        Driver       "wacom"
        Option       "Type" "stylus"
        Option       "Device" "/dev/ttyS0"
Endsection

Section "InputDevice"
        Identifier   "Tablet Eraser"
        Driver       "wacom"
        Option       "Type" "eraser"
        Option       "Device" "/dev/ttyS0"
Endsection

```
But in xorg.0.log is only loading the driver:
```
[  4430.767] (II) LoadModule: "wacom"
[  4430.768] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[  4430.768] (II) Module wacom: vendor="X.Org Foundation"
[  4430.768]    compiled for 1.19.1, module version = 0.34.0
[  4430.768]    Module class: X.Org XInput Driver
[  4430.768]    ABI class: X.Org XInput driver, version 24.1
[  4430.768] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
        Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
       ... etc, list of many models, not including ArtPad or KT-0405

```

I understand this piece of hardware is really old. I would like to know if it is still supported or not. If not, which version is the last that supports it? If yes, what has changed in the way to set it up?

Thanks
Vladki

jigpu, 2019-01-09:
The `inputattach` daemon is responsible for loading a kernel driver that can decode your serial tablet's data and create a standard `/dev/input/event*` device that the xf86-input-wacom device can use. I assume the "unsupported" message comes from the kernel driver, since it may not have had code for the ArtPad II added to it. Could you please provide the complete text of the message?

To use the xf86-input-wacom driver's internal serial tablet code, I'm pretty sure you'll need to add the "InputDevice" definitions to the /etc/X11/xorg.conf file -- I don't believe it works as a mere snippet. If it still doesn't work, post the Xorg.0.log contents again.

vladki77, 2019-01-17:
Running `inputattach --wacom_iv /dev/ttyS0` hangs, and the following is logged

```
serio: Serial port ttyS0
input input17: Unsupported Wacom model ~#KT-0405-R00 V1.3-6
```

Trying /etc/X11/xorg.conf.d/wacom.conf:
```
Section "Module"
        Load            "wacom"
EndSection

Section "InputDevice"
        Identifier      "Tablet Stylus"
        Driver          "wacom"
        Option          "Type" "stylus"
        Option          "Device" "/dev/ttyS0"
Endsection

Section "InputDevice"
        Identifier      "Tablet Eraser"
        Driver          "wacom"
        Option          "Type" "eraser"
        Option          "Device" "/dev/ttyS0"
Endsection

Section "ServerLayout"
        Identifier      "X.org Configured"
        InputDevice     "Tablet Stylus" "SendCoreEvents"
        InputDevice     "Tablet Eraser" "SendCoreEvents"
EndSection
```
xorg.log then gives:

```
[    36.485] (II) Using input driver 'wacom' for 'Tablet Stylus'
[    36.485] (**) Option "SendCoreEvents"
[    36.485] (**) Tablet Stylus: always reports core events
[    36.485] (**) Option "Device" "/dev/ttyS0"
[    36.485] (**) Option "Type" "stylus"
[    36.486] (**) Option "StopBits" "1"
[    36.486] (**) Option "DataBits" "8"
[    36.486] (**) Option "Parity" "None"
[    36.486] (**) Option "Vmin" "1"
[    36.486] (**) Option "Vtime" "10"
[    36.486] (**) Option "FlowControl" "Xoff"
```
and hangs.   This is on debian stretch:
linux-image-4.9.0-8-amd64                     4.9.130-2                             
xserver-xorg                                  1:7.7+19      
xserver-xorg-core                             2:1.19.2-1+deb9u5 
xserver-xorg-input-wacom                      0.34.0-1

When I try it on old debian squeeze, the log continues with:
```
[    38.259] (WW) Tablet Stylus: Waited too long for answer (failed after 3 tries).
[    38.259] (WW) Tablet Stylus: Query failed with 19200 baud. Trying 38400.
[    41.512] (WW) Tablet Stylus: Waited too long for answer (failed after 3 tries).
[    41.513] (II) Tablet Stylus: serial tablet id 0x90.
[    41.513] (EE) PreInit returned 8 for "Tablet Stylus"
[    41.513] (II) UnloadModule: "wacom"
```
and X starts normally
linux-image-3.2.0-6-686-pae             3.2.102-1 
xserver-xorg                            1:7.7+3~deb7u1
xserver-xorg-core                       2:1.12.4-6+deb7u8
xserver-xorg-input-wacom                0.15.0+20120515-2

vladki77, 2019-01-17:
version 0.15.0 did not work. It just did not hang.... The x-server worked fine, but the tablet had no effect. As you can see from the log, it says that the communication failed, and wacom module was unloaded.

The newer version 0.34.0 hanged completely, and the x-server did not start at all. I could not even switch to text console and had to log in to the computer remotely via SSH, to fix it.

jigpu, 2019-02-13:
It looks like this particular tablet starts up at 9600 baud, which would explain why the driver couldn't connect at either 38400 (the default) or 19200 (the fallback speed). The git history indicates that the driver used to support 9600 baud at one point, but the code was removed back in 2009.

It further looks like the remnants of the KD-0405-R code can still be found in the git history. In particular [this function](https://github.com/linuxwacom/xf86-input-wacom/blob/3b90a8c834ebd15b8d3bdabe46497080d820ce23/src/util/wacserial.c#L1099) seems to be responsible for decoding the packet and [this define](https://github.com/linuxwacom/xf86-input-wacom/blob/3b90a8c834ebd15b8d3bdabe46497080d820ce23/src/util/wacserial.c#L176) seems to specify the tablet features. This information could probably be used to update the kernel's wacom_serial4 driver to support this tablet. My guess with just a quick look though is that you might have good luck making just the following change and using the modified wacom_serial4 driver with inputattach set to use 9600 baud:
~~~
diff --git a/drivers/input/tablet/wacom_serial4.c b/drivers/input/tablet/wacom_serial4.c
index 38bfaca48eab..150f9eecaca7 100644
--- a/drivers/input/tablet/wacom_serial4.c
+++ b/drivers/input/tablet/wacom_serial4.c
@@ -187,6 +187,7 @@ enum {
 	MODEL_DIGITIZER_II	= 0x5544, /* UD */
 	MODEL_GRAPHIRE		= 0x4554, /* ET */
 	MODEL_PENPARTNER	= 0x4354, /* CT */
+	MODEL_ARTPAD_II		= 0x4B54, /* KT */
 };
 
 static void wacom_handle_model_response(struct wacom *wacom)
@@ -245,6 +246,7 @@ static void wacom_handle_model_response(struct wacom *wacom)
 		wacom->flags = F_HAS_STYLUS2 | F_HAS_SCROLLWHEEL;
 		break;
 
+	case MODEL_ARTPAD_II:
 	case MODEL_DIGITIZER_II:
 		wacom->dev->name = "Wacom Digitizer II";
 		wacom->dev->id.version = MODEL_DIGITIZER_II;
~~~

vladki77, 2019-03-03:
Hello,

thank you very much, this patch made it working. After that only:
`inputattach --daemon --baud 9600 --wacom_iv /dev/ttyS0 `
is needed, and then X detects the tablet and I can use it.

There are only two minor problems:
- at random occasions it generates middle button click, which switches gimp to moving canvas, and does not switch back, until I select some other tool
- when using the eraser, it spits out errors in xorg.log, but it works anyway:

```
[ 21656.404] (EE) BUG: triggered 'if (in_input_thread())'
[ 21656.404] (EE) BUG: ../../../../os/io.c:661 in WriteToClient()
[ 21656.404] (EE) ******** WriteToClient called from input thread *********
[ 21656.404] (EE) 
[ 21656.404] (EE) Backtrace:
[ 21656.404] (EE) 0: /usr/lib/xorg/Xorg (xorg_backtrace+0x4a) [0x56150b51a66a]
[ 21656.404] (EE) 1: /usr/lib/xorg/Xorg (WriteToClient+0x162) [0x56150b51dc82]
[ 21656.404] (EE) 2: /usr/lib/xorg/Xorg (WriteEventsToClient+0x252) [0x56150b3c0bb2]
[ 21656.405] (EE) 3: /usr/lib/xorg/Xorg (TryClientEvents+0x120) [0x56150b3c0d50]
[ 21656.405] (EE) 4: /usr/lib/xorg/Xorg (0x56150b361000+0x634ba) [0x56150b3c44ba]
[ 21656.405] (EE) 5: /usr/lib/xorg/Xorg (DeliverEventsToWindow+0x1d7) [0x56150b3c47a7]
[ 21656.405] (EE) 6: /usr/lib/xorg/Xorg (0x56150b361000+0x142b52) [0x56150b4a3b52]
[ 21656.405] (EE) 7: /usr/lib/xorg/Xorg (0x56150b361000+0x142b66) [0x56150b4a3b66]
[ 21656.405] (EE) 8: /usr/lib/xorg/Xorg (0x56150b361000+0x142b66) [0x56150b4a3b66]
[ 21656.405] (EE) 9: /usr/lib/xorg/Xorg (0x56150b361000+0x146aa0) [0x56150b4a7aa0]
[ 21656.405] (EE) 10: /usr/lib/xorg/Xorg (0x56150b361000+0x1524ff) [0x56150b4b34ff]
[ 21656.405] (EE) 11: /usr/lib/xorg/Xorg (XIChangeDeviceProperty+0x358) [0x56150b4b3eb8]
[ 21656.405] (EE) 12: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0x10df9) [0x7f9698f15df9]
[ 21656.405] (EE) 13: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0x7322) [0x7f9698f0c322]
[ 21656.405] (EE) 14: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0x8864) [0x7f9698f0d864]
[ 21656.405] (EE) 15: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0xe38f) [0x7f9698f1338f]
[ 21656.405] (EE) 16: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0x616b) [0x7f9698f0b16b]
[ 21656.405] (EE) 17: /usr/lib/xorg/modules/input/wacom_drv.so (0x7f9698f05000+0x63a6) [0x7f9698f0b3a6]
[ 21656.405] (EE) 18: /usr/lib/xorg/Xorg (0x56150b361000+0x1bb7dc) [0x56150b51c7dc]
[ 21656.405] (EE) 19: /usr/lib/xorg/Xorg (0x56150b361000+0x1bdec1) [0x56150b51eec1]
[ 21656.405] (EE) 20: /usr/lib/xorg/Xorg (0x56150b361000+0x1bb636) [0x56150b51c636]
[ 21656.405] (EE) 21: /lib/x86_64-linux-gnu/libpthread.so.0 (0x7f96a1e39000+0x74a4) [0x7f96a1e404a4]
[ 21656.405] (EE) 22: /lib/x86_64-linux-gnu/libc.so.6 (clone+0x3f) [0x7f96a1b82d0f]
[ 21656.405] (EE) 
```

vladki77, 2019-03-04:
UPDATE: after playing more with teblet and GIMP, I think it works as it should. The problem with moving layers/canvas is feature of GIMP - if I touch the scale on the window border, tool switches to "move layer", and I have to select new tool from the toolbox. Clicking the button on stylus, is indeed middle button, which moves the canvas (not layers), so it is ok. Pressure on stylus and eraser works too. So I think the above patch could be included to reeanble Artpad II support.

jigpu, 2019-03-04:
GIMP should move the canvas as long as the middle button is held down, but it should go back to working normally once the button is released. It shouldn't change the active tool at all... Is this what you're seeing?

I believe the "WriteToClient" bug may be fixed in version 0.34.2 or later of xf86-input-wacom. At least, that's the last time we had that sort of message appear.

vladki77, 2019-03-04:
As I said it was mostly error between keyboard and chair.
If I click the button on stylus, it generates middle click, and moves canvas, and when released it returns back to the previous brush tool. (That works only if the stylus is not drawing, i.e. touching the tablet only lightly, or held a few mm above). So I think that works as intended.
What I had a problem with was, that I unintentionally touched/clicked on the border of image (where the ruler is), and that switches to "move layer" tool, and never goes back. But that happens with mouse too. So again it works as intended. I just did not know about this function, and at first time I thought it is some random noise from tablet. But it is not.

I tested with version 0.34.0 (debian stable). If I have time, I'll try with testing/unstable.

vladki77, 2019-03-04:
And thank you once more. It is amazing to see that such old piece of HW is still working ;)

jigpu, 2019-03-05:
Ah, okay -- just wanted to make sure that the driver was properly releasing the middle click events :) I can send this patch off to the kernel mailinglist since all seems to be well. If you have any further issues, let us know.

closed 2019-03-05 by jigpu