linuxwacom/xf86-input-wacom #41: Cannot change Wacom Intuos S cursor proximity

dmitry-kulikov, 2019-01-14:
Cannot change Wacom Intuos S cursor proximity. Exact model name is Wacom Intuos Basic Pen S CTL-4100K-N.
Default value:
```sh
xsetwacom get 'Wacom Intuos S Pen cursor' CursorProximity
10
```
Physically it is about 1 cm. I would like to move cursor only when pen touches pad, so I do:
```sh
xsetwacom set 'Wacom Intuos S Pen cursor' CursorProximity 1
```
(or 0 instead of 1, there should not be much difference)
Command itself works, I can check result with get command and it shows 1 for CursorProximity.
But actual behavior was not changed, proximity is still about 1 cm.
```sh
xsetwacom --version
0.36.1
```
I was able to reach desired behavior on Windows using GUI settings tool, so it is not problem of device itself.

AlessandroMarinoni, 2019-09-19:
Hello,

I don't really know much about coding, but what should I do if I wanted to help with this issue (which I'm facing myself too)?

Is there any advice you could please give me?

And lastly, would CursorProximity be supported by the Wacom One S (CTL-472)?

jigpu, 2019-09-23:
The next release of xf86-input-wacom will include the ability to set CursorProximity for all pen devices (including the CTL-472). Once it has been released, you can install it by following [these instructions](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver) and then use `xsetwacom set <stylus_id> CursorProximity <n>` to adjust the height at which the pointer begins to respond to motion.

We're planning on making a release in a week or so, so stay tuned :)

Ref: https://github.com/linuxwacom/input-wacom/issues/54

closed 2019-12-12 by jigpu

cluelessperson, 2020-07-22:
Issue still occurring for me:

```
xsetwacom --version
0.39.0
```

`xsetwacom set 10 CursorProximity 30`

```
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  37 (X_ChangeDeviceProperty)
  Value in failed request:  0x143
  Serial number of failed request:  19
  Current serial number in output stream:  22
```

jigpu, 2020-07-23:
@cluelessperson can you run `xsetwacom list` and let us know which of the listed devices you were trying to set CursorProximity on (stylus, eraser, etc.)?

cluelessperson, 2020-07-24:
@jigpu Sure:  It's the STULUS (10)

```
xsetwacom list
Wacom Pen and multitouch sensor Finger touch	id: 9	type: TOUCH     
Wacom Pen and multitouch sensor Pen stylus	id: 10	type: STYLUS    
Wacom Pen and multitouch sensor Pen eraser	id: 19	type: ERASER
```

jigpu, 2020-07-24:
Thanks. Based on the name, it looks like this is a tablet PC sensor, probably based on AES tech. These devices don't report distance information, so it is impossible to change the CursorProximity value in any meaningful way. We should actually prevent it from even being read for these sensors...

BlueGuitarCL, 2022-06-19:
Owner of both CTL-472 and a CTL-4100WL here. Allthough at xsetwacom version 1.0.0.46.gbcd984b it works flawlessly for Relative mode, i wish i could use the CursorProximity for Absolute Mode too, as it can be configured on the driver GUI inside Windows.

whot, 2022-06-20:
@BlueGuitarCL - the proximity handling is completely separate from the device mode, please file a separate bug for this

idle-meerkat, 2023-05-01:
Any fix/workaround for this so far?