linuxwacom/xf86-input-wacom #42: Impossible to scroll text documents in Nextcloud under Gnome Shell Wayland session

RussianNeuroMancer, 2019-01-14:
Hello!

Original report: https://sourceforge.net/p/linuxwacom/bugs/346/

As I find today, it's still reproducible, under Gnome Shell Wayland too.

RussianNeuroMancer, 2019-01-18:
> The xf86-input-wacom driver isn't in use anywhere under a Wayland session. 

I wonder if difference described in [this comment](https://sourceforge.net/p/linuxwacom/bugs/346/#c9f0) by @jigpu could came from kernel driver instead of xf86-input-wacom?

> If the scrolling works with the same device in other applications, the issue is most likely in the toolkit or possibly the application itself. What toolkits does Nextcloud use?

Nextcloud text editor is web application. Issue is always reproducible in Blink-based web-browsers and I couldn't reproduce it in Firefox.

whot, 2019-01-18:
> I wonder if difference described in this comment by @jigpu could came from kernel driver instead of xf86-input-wacom?

do we have any evidence that blink uses the axis labels? I have never seen a client that actually cares about them.

RussianNeuroMancer, 2019-01-18:
@whot 

> do we have any evidence that blink uses the axis labels?

Is there any way I can help with checking if that the case, besides published s1ynaptics/wacom touchscreen recordings?

whot, 2019-01-21:
git grepping through the [chromium source](https://chromium.googlesource.com/chromium) code shows `ui/base/x/valuators.cc` using the axis labels (see `FindValuator`). So that' some indication that the axis labels matter, but I'm ETIME on figuring out exactly how that affects things in detail. If you can search through the code to get a better picture here, it's likely to contain the answer to this bug.

jigpu, 2019-01-21:
It could be interesting to try changing the axis labels of xf86-input-wacom to see if that makes a difference. Might be easier than spelunking through the Chromium sources.

RussianNeuroMancer, 2019-01-23:
> try changing the axis labels of xf86-input-wacom

As I understand I can change it [here](https://github.com/linuxwacom/xf86-input-wacom/blob/84400df38f0f9abe664de26a8d3747b10f3a05e5/src/wcmUSB.c#L1222), but I not quite sure how to represent ABS_X and ABS_Y events as ABS_MT_POSITION_X and ABS_MT_POSITION_Y events. How to do it?

jigpu, 2019-01-23:
The axes you need to change are actually inside the [wcmInitAxes function](https://github.com/linuxwacom/xf86-input-wacom/blob/master/src/xf86Wacom.c#L155). I think the only two changes you need to make are:
* Change `AXIS_LABEL_PROP_ABS_X` to `AXIS_LABEL_PROP_ABS_MT_POSITION_X`
* Change `AXIS_LABEL_PROP_ABS_Y` to `AXIS_LABEL_PROP_ABS_MT_POSITION_Y`

If that doesn't work and you want to experiment more, a complete list of axis values can be found in [xserver-properties.h](https://github.com/XQuartz/xorg-server/blob/master/include/xserver-properties.h#L65). The axes exposed in your Synaptics recording were "Abs MT Position X", "Abs MT Position Y", "Rel Horiz Scroll", and "Rel Vert Scroll".

jigpu, 2019-01-24:
Did you test NextCloud in a Wayland session or an Xorg session? The changes I proposed will only have an effect if you start an Xorg session since our driver isn't used in Wayland (which is why whot closed this particular bug).

Also, you'll need to run `xinput test-xi2 <touch id>` rather than `evemu-record` to see if the changes were applied. The former is used to see X11 device properties and events (which is what Chrome cares about) while the latter is used to see kernel device properties and events (which only really matters to xf86-input-wacom and libinput).

When you run the command, a bunch of text will be printed and a window will pop up; you can immediately close the window because we only care about what's contained in that first bit of text. What you're looking for is if the "Label" attributes were updated from "Abs X" and "Abs Y" to "Abs MT Position X" and "Abs MT Position Y". The following is a snippet of the output from the stock driver; yours should have the modified labels:

>		Class originated from: 8. Type: XIValuatorClass
>		Detail for Valuator 0:
>		  Label: Abs X
>		  Range: 0.000000 - 5184.000000
>		  Resolution: 10000 units/m
>		  Mode: absolute
>		  Current value: 2591.000000
>		Class originated from: 8. Type: XIValuatorClass
>		Detail for Valuator 1:
>		  Label: Abs Y
>		  Range: 0.000000 - 3240.000000
>		  Resolution: 10000 units/m
>		  Mode: absolute
>		  Current value: 1134.000000

RussianNeuroMancer, 2019-01-25:
> Did you test NextCloud in a Wayland session or an Xorg session? 

Xorg session.

> Also, you'll need to run xinput test-xi2 <touch id> rather than evemu-record to see if the changes were applied.

There is xinput test-xi2 record from session and it's seems like labels was indeed updated: https://paste.fedoraproject.org/paste/65mfysLzja1nZExRvtlBtQ

So probably this change is not enough for Chromium. Should I fill bugreport to Chromium project at this point, or there is other options to try?

There is easy way to get access to NextCloud editor btw: https://nextcloud.com/signup/

jigpu, 2019-01-25:
Thanks for the link! I've been toying around with the editor and think I may have found something. After reverting the changes to the wcmInitAxes function (they shouldn't be necessary), open up Chrome and navigate to chrome://flags/#touch-events. Try setting the option to "Enabled" and then re-launch Chrome. This seems to have gotten the NextCloud editor to work for me like it does in Firefox (I've only tested under Xorg, not Wayland).

The default "Automatic" setting is supposed to enable the touch events API when Chrome finds a touchscreen, but if changing it to "Enabled" fixes the issue for you then that means Chrome is having trouble recognizing the Cintiq as a touchscreen.

RussianNeuroMancer, 2019-01-25:
@jigpu it works indeed, thank you :)

Maybe put info about "Touch Events API" toggle somewhere, like https://github.com/linuxwacom/libwacom/wiki/Troubleshooting or https://linuxwacom.github.io to add more visibility?

jigpu, 2019-01-25:
Glad to hear it :) I'd file a bug against Chrome to be sure they're aware of the problem.

As for us, I'll see if I can find a good spot to document this problem for others.

RussianNeuroMancer, 2019-01-28:
@jigpu 

> I'd file a bug against Chrome to be sure they're aware of the problem.

I could do it, but I guess if you do this you for sure will describe issue better than me, and if you mention name of project that you maintain bugreport could get more attention from Chromium developers.

RussianNeuroMancer, 2019-08-25:
> The default "Automatic" setting is supposed to enable the touch events API when Chrome finds a touchscreen, but if changing it to "Enabled" fixes the issue for you then that means Chrome is having trouble recognizing the Cintiq as a touchscreen.

With default "Automatic" state of this flag I wasn't able to reproduce issue with Nextcloud editor anymore.