linuxwacom/xf86-input-wacom #43: Linux delay when 2 finger scroll or zoom

damarusama, 2019-01-14:
I'm experiencing a lag when I use a double finger touch scroll or zoom. 
I am using 0.36.1-1 on Arch Linux. The lag happens on scroll and on zoom. I am using Wacom Intuos PT S Finger touch. It's exactly as described in this previous bug:

https://sourceforge.net/p/linuxwacom/bugs/218/

There is an early delay then it 'kicks in' (scroll or zoom) so you can then continue to scroll or zoom while you still have your 2 fingers on the tablet, but the behavior is reset once you release your fingers.

jigpu, 2019-09-04:
I've been tinkering with this off-and-on and (finally) have some code that I think should solve the remaining issues. I've submitted a pull request to get this integrated, and you can get the code by cloning the appropriate branch in my repository with the following command.
~~~
$ git clone https://github.com/jigpu/xf86-input-wacom.git -b issue-43
~~~

jigpu, 2019-12-12:
Pull request merged and part of xf86-input-wacom 0.38.0

closed 2019-12-12 by jigpu