linuxwacom/xf86-input-wacom #44: Multi touch on wacom  sometimes ignores a finger

atrics, 2019-01-15:
Description:
=
 
We observed a bug while using a touchable wacom monitor on a X11 based linux
system. Using multifinger touch can lead to states in which a single finger in a
multi touch sceneario will be mostly ignored.
 
Prerequireries
==
 
Setup wacom device with 'xsetwacom --set [device] Touch on' and 'xsetwacom --set
[device] Gesture off'. After pressing multiple times with multiple fingers a
state can be reached where a single finger will be ignored, especially while
using two fingers. The first contact point of the second finger will be
recognized, but every movement afterwards disappears. This effect can be
visualized by compiling the fingerpaint example of qt (see https://doc.qt.io/qt-5/qtwidgets-touch-fingerpaint-example.html [^]). See the given result
[image] vs the expected result:
[image]

In the wrong state a single yellow point is drawn where a line is expected.
 
The state can 'repaired' by touching the monitor again with e.g 10 fingers
 
Further observations
==
 
We took a look into evdev event by monitoring the multitouch events out of the
evtest tools and an additional look into the output of 'xinput --test-xi2'.
There are 3 files representing the output of a two finger touch while in a false
state. The two finger where moved in parallel similar to 'resultExpected.jpg':
 
 - `evdev /dev/input[X] > evdev.txt` [evdev.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2760056/evdev.txt)
 - `xinput --test-xi2 [X] > test_xi2.txt` [test_xi2.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2760058/test_xi2.txt)
 - 'cat evdev.txt | python3 evdev_parser.py >  parsed_evdev_output_finger_positions.txt' 
[parsed_evdev_output_finger_positions.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2760068/parsed_evdev_output_finger_positions.txt)
[evdev_parser.py.zip](https://github.com/linuxwacom/xf86-input-wacom/files/2760067/evdev_parser.py.zip)

The evdev output seems to be plausible, as it updates each of the two fingers
with a reasonable position. The xinput output has a TouchEnd event of the second
finger (see line 122) that appears too early.
 
The bug occures only for wacom devices so we suspect the bug inside the
xorg-x11-drv-wacom package

System environments
==
 
The error occured for in two different environments with two different wacom
touch monitors. We couldn't reproduce the error with a different non-wacom touch
monitor on the same enviroments:
 
Centos 7:
===
 
cmd:
`  xsetwacom --list devices`
 
output:
```
  Wacom Express Key Remote Pad pad id: 13 type: PAD
  Wacom Cintiq Pro 32 Touch Finger touch id: 14 type: TOUCH
  Wacom Cintiq Pro 32 Pen stylus id: 15 type: STYLUS
  Wacom Cintiq Pro 32 Pen eraser id: 16 type: ERASER
```
 
 
cmd:
  yum list installed | sed '/\(wacom\|xorg-x11\)/!d':
 
output:
```
  libwacom.x86_64 0.30-1.el7 @base
  libwacom-data.noarch 0.30-1.el7 @base
  xorg-x11-drivers.x86_64 7.7-6.el7 @anaconda
  xorg-x11-drv-ati.x86_64 18.0.1-1.el7 @base
  xorg-x11-drv-dummy.x86_64 0.3.7-1.el7.1 @base
  xorg-x11-drv-evdev.x86_64 2.10.6-1.el7 @base
  xorg-x11-drv-fbdev.x86_64 0.5.0-1.el7 @base
  xorg-x11-drv-intel.x86_64 2.99.917-28.20180530.el7 @base
  xorg-x11-drv-nouveau.x86_64 1:1.0.15-1.el7 @base
  xorg-x11-drv-qxl.x86_64 0.1.5-4.el7.1 @base
  xorg-x11-drv-synaptics.x86_64 1.9.0-2.el7 @base
  xorg-x11-drv-v4l.x86_64 0.2.0-49.el7 @base
  xorg-x11-drv-vesa.x86_64 2.4.0-1.el7 @base
  xorg-x11-drv-vmmouse.x86_64 13.1.0-1.el7.1 @base
  xorg-x11-drv-vmware.x86_64 13.2.1-1.el7.1 @base
  xorg-x11-drv-void.x86_64 1.4.1-2.el7.1 @base
  xorg-x11-drv-wacom.x86_64 0.36.1-1.el7 @base
  xorg-x11-font-utils.x86_64 1:7.5-21.el7 @base
  xorg-x11-fonts-Type1.noarch 7.5-9.el7 @anaconda
  xorg-x11-proto-devel.noarch 2018.4-1.el7 @base
  xorg-x11-server-Xorg.x86_64 1.20.1-5.1.el7 @updates
  xorg-x11-server-common.x86_64 1.20.1-5.1.el7 @updates
  xorg-x11-server-utils.x86_64 7.7-20.el7 @anaconda
  xorg-x11-utils.x86_64 7.5-23.el7 @base
  xorg-x11-xauth.x86_64 1:1.0.9-1.el7 @anaconda
  xorg-x11-xinit.x86_64 1.3.4-2.el7 @anaconda
  xorg-x11-xkb-utils.x86_64 7.7-14.el7 @base
```
 
 
cmd:
`  lsusb | grep 056a`
 
output:
```
  Bus 002 Device 064: ID 056a:038b Wacom Co., Ltd
  Bus 002 Device 063: ID 056a:038c Wacom Co., Ltd
  Bus 002 Device 062: ID 056a:0389 Wacom Co., Ltd
  Bus 001 Device 026: ID 056a:0352 Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] tablet
  Bus 001 Device 024: ID 056a:0356 Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] touchscreen
  Bus 001 Device 023: ID 056a:038a Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] internal hub
  Bus 001 Device 021: ID 056a:038e Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] external hub
  Bus 001 Device 022: ID 056a:0331 Wacom Co., Ltd
  Bus 001 Device 020: ID 056a:038f Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] internal hub
  Bus 001 Device 019: ID 056a:038d Wacom Co., Ltd DTH-3220 [Cintiq Pro 32] internal hub
```
 
 
Fedora 29:
===
 
cmd:
`  xsetwacom --list devices`
 
output:
```
  Wacom Cintiq 24HD touch Finger touch id: 9 type: TOUCH
  Wacom Cintiq 24HD touch Pen stylus id: 10 type: STYLUS
  Wacom Cintiq 24HD touch Pad pad id: 11 type: PAD
  Wacom Cintiq 24HD touch Pen eraser id: 15 type: ERASER
```
 
 
cmd:
`  yum list installed | sed '/\(wacom\|xorg-x11\)/!d':`
 
output:
```
  libwacom.x86_64 0.32-2.fc29 @updates
  libwacom-data.noarch 0.32-2.fc29 @updates
  xorg-x11-apps.x86_64 7.7-22.fc29 @fedora
  xorg-x11-drv-ati.x86_64 18.1.0-1.fc29 @updates
  xorg-x11-drv-evdev.x86_64 2.10.6-3.fc29 @fedora
  xorg-x11-drv-fbdev.x86_64 0.5.0-2.fc29 @fedora
  xorg-x11-drv-intel.x86_64 2.99.917-39.20180618.fc29 @fedora
  xorg-x11-drv-libinput.x86_64 0.28.1-1.fc29 @updates
  xorg-x11-drv-nouveau.x86_64 1:1.0.15-6.fc29 @fedora
  xorg-x11-drv-nvidia.x86_64 3:415.25-1.fc29 @rpmfusion-nonfree-updates
  xorg-x11-drv-nvidia-cuda.x86_64 3:415.25-1.fc29 @rpmfusion-nonfree-updates
  xorg-x11-drv-nvidia-cuda-libs.x86_64 3:415.25-1.fc29 @rpmfusion-nonfree-updates
  xorg-x11-drv-nvidia-kmodsrc.x86_64 3:415.25-1.fc29 @rpmfusion-nonfree-updates
  xorg-x11-drv-nvidia-libs.x86_64 3:415.25-1.fc29 @rpmfusion-nonfree-updates
  xorg-x11-drv-openchrome.x86_64 0.6.0-6.fc29 @fedora
  xorg-x11-drv-qxl.x86_64 0.1.5-10.fc29 @fedora
  xorg-x11-drv-vesa.x86_64 2.4.0-4.fc29 @fedora
  xorg-x11-drv-vmware.x86_64 13.2.1-8.fc29 @fedora
  xorg-x11-drv-wacom.x86_64 0.36.1-5.fc29 @fedora
  xorg-x11-drv-wacom-serial-support.x86_64 0.36.1-5.fc29 @fedora
  xorg-x11-font-utils.x86_64 1:7.5-40.fc29 @fedora
  xorg-x11-fonts-misc.noarch 7.5-20.fc29 @fedora
  xorg-x11-proto-devel.noarch 2018.4-2.fc29 @fedora
  xorg-x11-server-Xorg.x86_64 1.20.3-2.fc29 @updates
  xorg-x11-server-Xwayland.x86_64 1.20.3-2.fc29 @updates
  xorg-x11-server-common.x86_64 1.20.3-2.fc29 @updates
  xorg-x11-server-utils.x86_64 7.7-26.fc29 @fedora
  xorg-x11-utils.x86_64 7.5-29.fc29 @fedora
  xorg-x11-xauth.x86_64 1:1.0.9-13.fc29 @fedora
  xorg-x11-xbitmaps.noarch 1.1.1-15.fc29 @fedora
  xorg-x11-xinit.x86_64 1.4.0-3.fc29 @fedora
  xorg-x11-xkb-utils.x86_64 7.7-27.fc29 @fedora
  xorg-x11-xkb-utils-devel.x86_64 7.7-27.fc29 @fedor
```
 
cmd:
`  lsusb | grep 056a`
 
output:
```
  Bus 003 Device 008: ID 056a:00f8 Wacom Co., Ltd DTH-2400 [Cintiq 24HD touch] tablet
  Bus 003 Device 007: ID 056a:00f6 Wacom Co., Ltd DTH-2400 [Cintiq 24HD touch] touchscreen
  Bus 003 Device 006: ID 056a:00f5 Wacom Co., Ltd
```

jigpu, 2019-01-15:
I've reformatted your "evdev.txt" file to be in the format expected by `evemu-play` so I can replay it locally: [evemu-replay.txt](https://github.com/linuxwacom/xf86-input-wacom/files/2760525/evemu-replay.txt). Replaying it on my (Arch Linux) development box doesn't show any issues, but I am admittedly not running the vanilla 0.36.1 version. I'll try replaying it on Fedora 29 and CentOS 7 to see if they react differently. I also haven't had any luck so far reproducing the issue with a Cintiq 24HDT on my development box either.

atrics, 2019-01-16:
Thanks for your fast response. Please let me know If I may provide help or more information about the screens or the system.

atrics, 2019-01-16:
The false state is normally reached if I put my complete palm onto the surface, rotate it and start drumming with my fingers

leinir, 2019-08-19:
It seems pertinent to make mention that this is a solution currently being adopted in at least one major distribution (in short, disable the wacom driver for the touchscreen part of devices which have a Wacom based digitizer and touch screen): https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1774242

Pinglinux, 2022-03-19:
The device only recognize figures

closed 2022-03-19 by Pinglinux