linuxwacom/xf86-input-wacom #45: Route KEY_PROG1 and friends through to X

whot, 2019-01-17:
The Cintiq 24HD sends `KEY_PROG1` through to `KEY_PROG3` for the special buttons, but the wacom driver discards them. Let's not do that, let's pass them on and let userspace figure out what to do with them. The default mapping is `XF86Launch1` through to `XF86Launch3` so they could be assigned.

More recent tablets send `KEY_CONTROLPANEL`, `KEY_ONSCREEN_DISPLAY`, and `KEY_BUTTONCONFIG` but those have keycodes outside the 255 X keycode range.

whot, 2019-01-17:
@jigpu, can you provide me with an evemu recording for the two series of tablets affected? I need to know if we only see `KEY_foo` on press or other data as well (`ABS_MISC`, etc.)

jigpu, 2019-01-18:
You'll be getting ABS_MISC as well as the KEY event; that's just part-and-parcel of our contract* with xf86-input-wacom. I don't have the muscles to dig out a 27QHD right now, but I've attached HID and evemu recordings of the 24HDT. Fun fact: they show off a bug I hadn't noticed in current kernels that prevents KEY_PROG3 from ever being sent :fearful: 

[24hdt-function-keys.evemu.log](https://github.com/linuxwacom/xf86-input-wacom/files/2774848/24hdt-function-keys.evemu.log)
[24hdt-function-keys.hid.log](https://github.com/linuxwacom/xf86-input-wacom/files/2774849/24hdt-function-keys.hid.log)

*One that we might consider revising at some point now that the pad is its own top-level device in upstream kernels...

whot, 2019-10-22:
Closing this one, its'... tricky.

The simple part is that with [kernel 5.4](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=670e90924bfe9cf80536eab66a31704e07b5fd00) all tablets send the new key codes and we don't need to care about `KEY_PROG*` anymore.

The difficult part is that the new key codes are outside the X server's 255 keycode range. They would need remapping down to something below 255 **and** a special XKB option to map them to something actually useful. Whether that is likely to happen is doubtful, I think this is a case where we just have to admit defeat due to ETIME and say that this will work under Wayland where we don't have those restrictions in place.

PR #46 (2021-02-02): Pass on the hw button keycodes
Fixes #45 

**This is still work in progress**

This patch adds the code to pass on `KEY_PROG1` and friends to the X server, to be passed on to the clients from there. Realistically, this doesn't do anything for the other three codes because they're above the 255 keycode limit and will get dropped by the server immediately.

The handling is hardcoded for the 6 buttons we care about, IMO there's little point to make this a fully generic.

I'm not sure if the events can occur on their own without any `EV_ABS` like we do for the normal buttons. Right now, the code a [... cut]
closed 2019-10-22 by whot