linuxwacom/xf86-input-wacom #48: PressureCurve keeps resetting

GregDMeyer, 2019-01-31:
I'm trying to play with PressureCurve for my tablet. Unfortunately, no matter how I set the value, it always resets to linear when I bring my stylus away from proximity, and then back in proximity again. I've tried:

 - setting it in xorg.conf (after boot, before I've brought the stylus in proximity, it is set to that value, but gets set to linear as soon as the stylus is detected)
 - turning wacom plugin off in gsettings (I tried setting it explicitly via gsettings also, but I get that "Schema “org.gnome.settings-daemon.peripherals.wacom.stylus” is relocatable (path must be specified)" and then cound't figure out the appropriate path)
 - setting it via xinput
 - setting it via xsetwacom

For both xinput and xsetwacom, if my stylus is in proximity, the values set correctly, but as soon as I leave proximity and come back again, they reset to linear.

Not sure if this is a bug or just a weird configuration issue, but I though opening an issue may help.

jigpu, 2019-02-08:
What distribution are you using, and what version of GNOME is installed? GNOME is well-known for resetting the pressure curve (and other options) to the values configured through the control panel whenever the pen enters proximity. Disabling the Wacom plugin in gsettings used to be a good workaround for GNOME misbehaviors but it has stopped doing much in recent releases. Its quite possible that you're using one of these recent releases that can't be worked around :(

If the pressure curve you want isn't settable through GNOME (or is being reset to linear despite a firmer/softer setting in the control panel), you will probably want to [file a bug here](https://gitlab.gnome.org/GNOME/gnome-control-center/issues) to let the GNOME developers know.

GregDMeyer, 2019-02-18:
Thanks so much for the detailed response!

I'm on Ubuntu 18.10 with GNOME Shell 3.30.1. Unfortunately due to [this bug](https://bugzilla.gnome.org/show_bug.cgi?id=793499) I can't configure anything through the Gnome Control Center anyway (though a fix for that bug has been committed now).

Anyway, I'll see if there remains a way to configure it in GNOME through the command line. Thanks again!

jigpu, 2019-07-17:
Closing since this is more of an incompatibility with GNOME that can't be resolved. I've updated the GNOME documentation (https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-3%3A-GNOME-3) to provide more information on where to find its configuration keys (which may be useful for manually setting a pressure curve) as well as providing additional detail on the current state of disabling the gsdwacom plugin.

closed 2019-07-17 by jigpu