linuxwacom/xf86-input-wacom #50: xorg-core issue after hwe update

thwaller, 2019-02-17:
While trying to install via:
`$ sudo apt-get install xserver-xorg-input-wacom`

I receive the following:
`The following packages have unmet dependencies:
 xserver-xorg-input-wacom : Depends: xserver-xorg-core (>= 2:1.18.99.901)`

What I have installed is:
xserver-xorg-core-hwe-18.04 version 2:1.20.1-3ubuntu2.1~18.04.1

which replaced:
xserver-xorg-core version 2:1.19.6-1ubuntu4.2

I understand that this is a different package, although the version does meet the requirements. Will there be compatibility for this, some sort of transitional package or workaround of sort?

jigpu, 2019-07-17:
This is an issue with the distribution package(s). We don't have any control over the packaging; issues should be filed with the distribution itself.

closed 2019-07-17 by jigpu

thwaller, 2019-07-18:
Could you explain further? Are you indicating that "xserver-xorg-input-wacom" is maintained elsewhere / by someone other than you?

jigpu, 2019-07-18:
That's correct. We write and maintain the code; Ubuntu (and Fedora, and SUSE, and etc.) customizes and compiles the code for their various releases and then packages the output for installation. If you go to https://packages.ubuntu.com/bionic/x11/xserver-xorg-input-wacom you'll see that the maintainer of the "xserver-xorg-input-wacom" package is the "Ubuntu X-SWAT" team. The "xserver-xorg-core-hwe-18.04" package is [also maintained](https://packages.ubuntu.com/bionic-updates/xserver-xorg-core-hwe-18.04) by the same team.

It looks like they recommend you file a bug on Launchpad rather than contacting the package maintainer directly. I think I would probably file a bug against the "xserver-xorg-core-hwe-18.04" package since I would expect it to still allow you to install packages that depend on "xserver-xorg-core".

thwaller, 2019-07-18:
Thanks for the clarification. It is very helpful.

kevin-brotcke, 2020-03-12:
Ran into the same error on Ubuntu 18.04. Got it working with `sudo apt install xserver-xorg-input-wacom-hwe-18.04`.