linuxwacom/xf86-input-wacom #51: Bamboo PAD CTH-301 pen recognized, but strokes are not read

gbraad, 2019-03-26:
When the Bamboo PAD is used, it can start a stroke, and when quickly stroked you can see a short diagonal line with some pressure information, but you can not draw an actual stroke. Hovering is correctly recignizedm, but as soon as the pen touches the surface the cursor doesn't move anymore.

gbraad, 2019-03-26:
`xinput --test [id]` shows the following:

```
motion a[0]=7063 a[1]=4611 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7041 a[1]=4621 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7027 a[1]=4623 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7025 a[1]=4622 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7033 a[1]=4615 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7051 a[1]=4603 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7076 a[1]=4589 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7104 a[1]=4577 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7136 a[1]=4563 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7168 a[1]=4550 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7199 a[1]=4542 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7228 a[1]=4537 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7255 a[1]=4533 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7278 a[1]=4533 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7296 a[1]=4535 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7311 a[1]=4540 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7317 a[1]=4565 a[2]=8977 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7277 a[1]=4565 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7281 a[1]=4563 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7276 a[1]=4562 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7265 a[1]=4559 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7254 a[1]=4553 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7244 a[1]=4548 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7232 a[1]=4543 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7217 a[1]=4541 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7201 a[1]=4540 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7185 a[1]=4540 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7170 a[1]=4540 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7154 a[1]=4541 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7140 a[1]=4543 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7126 a[1]=4546 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7113 a[1]=4549 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7099 a[1]=4552 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7084 a[1]=4555 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=7049 a[1]=4552 a[2]=12183 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6953 a[1]=4741 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6954 a[1]=4730 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6952 a[1]=4722 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6943 a[1]=4718 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6925 a[1]=4720 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
motion a[0]=6903 a[1]=4726 a[2]=0 a[3]=0 a[4]=0 a[5]=-900 
```

motion seems to stop when pressed on the surface.

gbraad, 2019-03-26:
Also used `evemu-record` and pressure and strokes are received:

```
E: 1.320012 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +11ms
E: 1.332003 0003 0000 7114	# EV_ABS / ABS_X                7114
E: 1.332003 0003 0001 3895	# EV_ABS / ABS_Y                3895
E: 1.332003 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +12ms
E: 1.341948 0003 0000 7124	# EV_ABS / ABS_X                7124
E: 1.341948 0003 0001 3901	# EV_ABS / ABS_Y                3901
E: 1.341948 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +9ms
E: 1.361995 0003 0000 7135	# EV_ABS / ABS_X                7135
E: 1.361995 0003 0001 3910	# EV_ABS / ABS_Y                3910
E: 1.361995 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +20ms
E: 1.371984 0003 0000 7143	# EV_ABS / ABS_X                7143
E: 1.371984 0003 0001 3918	# EV_ABS / ABS_Y                3918
E: 1.371984 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.381993 0003 0018 0109	# EV_ABS / ABS_PRESSURE         109
E: 1.381993 0001 014a 0001	# EV_KEY / BTN_TOUCH            1
E: 1.381993 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.393951 0003 0000 7151	# EV_ABS / ABS_X                7151
E: 1.393951 0003 0001 3928	# EV_ABS / ABS_Y                3928
E: 1.393951 0003 0018 0158	# EV_ABS / ABS_PRESSURE         158
E: 1.393951 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +12ms
E: 1.404022 0003 0018 0190	# EV_ABS / ABS_PRESSURE         190
E: 1.404022 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +11ms
E: 1.413942 0003 0000 7155	# EV_ABS / ABS_X                7155
E: 1.413942 0003 0001 3942	# EV_ABS / ABS_Y                3942
E: 1.413942 0003 0018 0210	# EV_ABS / ABS_PRESSURE         210
E: 1.413942 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +9ms
E: 1.423977 0003 0018 0223	# EV_ABS / ABS_PRESSURE         223
E: 1.423977 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.433980 0003 0000 7154	# EV_ABS / ABS_X                7154
E: 1.433980 0003 0001 3960	# EV_ABS / ABS_Y                3960
E: 1.433980 0003 0018 0226	# EV_ABS / ABS_PRESSURE         226
E: 1.433980 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.443966 0003 0000 7149	# EV_ABS / ABS_X                7149
E: 1.443966 0003 0001 3977	# EV_ABS / ABS_Y                3977
E: 1.443966 0003 0018 0228	# EV_ABS / ABS_PRESSURE         228
E: 1.443966 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.453965 0003 0000 7141	# EV_ABS / ABS_X                7141
E: 1.453965 0003 0001 3993	# EV_ABS / ABS_Y                3993
E: 1.453965 0003 0018 0230	# EV_ABS / ABS_PRESSURE         230
E: 1.453965 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +10ms
E: 1.466004 0003 0000 7131	# EV_ABS / ABS_X                7131
E: 1.466004 0003 0001 4007	# EV_ABS / ABS_Y                4007
E: 1.466004 0003 0018 0231	# EV_ABS / ABS_PRESSURE         231
E: 1.466004 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +13ms
E: 1.475996 0003 0000 7116	# EV_ABS / ABS_X                7116
E: 1.475996 0003 0001 4022	# EV_ABS / ABS_Y                4022
E: 1.475996 0003 0018 0234	# EV_ABS / ABS_PRESSURE         234
E: 1.475996 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +9ms
```

jigpu, 2019-03-26:
I've replayed the log that you linked to and don't see any issue on my end. One tap and a stroke are recorded in the log and that matches up with what I see. Would you mind running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) with your tablet connected and attaching the generated tarball to this bug (just drag and drop onto the comment box)? This will give us a better picture of the drivers that are installed and in use, as well as any errors that might be appearing in various relevant logs.

jigpu, 2019-03-26:
Also, perhaps https://github.com/linuxwacom/input-wacom/issues/40 could explain your issue?

gbraad, 2019-03-28:
Anyway, I ran the sysinfo and checked what got collected (you got some permission denied as it wasn;'t run as root), but I think you got the important bits anyway. It is on F29, and this might be related/same issue.
[sysinfo.Fal0X4KCUJ.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3016833/sysinfo.Fal0X4KCUJ.tar.gz)

jigpu, 2019-03-28:
Thanks. It does indeed sound like its probably the GNOME issue that you're experiencing. If so, you should be able to download and build a modified version of input-wacom which changes the device name.

The basic steps are:
1. Download and extract the [current input-wacom driver source](https://github.com/linuxwacom/input-wacom/releases/download/input-wacom-0.42.0/input-wacom-0.42.0.tar.bz2)
2. Download the patch I posted to [this comment](https://github.com/linuxwacom/input-wacom/issues/40#issuecomment-420740042)
3. Open a terminal and navigate to the extracted input-wacom directory
4. Apply the patch by running `patch -p1 < 0001-HACK-Rename-Bamboo-PAD-for-Ubuntu-18.04-compatibilit.patch.txt`
5. Follow the [input-wacom build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) to compile and install the driver
6. Reboot

gbraad, 2019-04-11:
Sorry, have been busy... I wanted to create a RPM to test this out, as
i do not want to create a one-of/snowflake. Especially when this is
likely fixed in a later version of mutter.

jigpu, 2019-04-11:
No worries. Just cleaning up some of the stale bugs. I'll keep this open for now :)

jigpu, 2019-06-20:
Closing. Feel free to re-open if the suggested workaround doesn't actually work for you.

closed 2019-06-20 by jigpu