linuxwacom/xf86-input-wacom #52: Surface GO: pen triggers eraser under X11 but not Wayland

ShapeShifter499, 2019-04-08:
After some back and forth on this https://github.com/linuxwacom/libwacom/issues/75  It was determined the issue lies somewhere in xf86-input-wacom.

When using the pen in X11 and Gnome, the pen is triggering the eraser (or whatever tool the 'eraser' end was set to). Applications like GIMP, Xournal, and Stylus Labs Write all showed the same issue. 

However under Wayland and Gnome these issues went away. The pen tip wrote fine with pressure sensitivity and the eraser end erased lines. All previously tested applications no longer had issue.

So this is where I'm at now, I'd like to see X11 support fixed as there are still some quirks with Wayland.

Baldurius, 2019-04-13:
I have the same problems as @ShapeShifter499 .

I've compiled the wacom driver following the [instructions](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver) looked into the log files by first running
```
xsetwacom list
```
which printed
```
ELAN9038:00 04F3:261A stylus    id: 16   type: STYLUS
ELAN9038:00 04F3:261A eraser    id: 17   type: ERASER
```
I then enable debug messages via
```
xsetwacom --set "ELAN9038:00 04F3:261A stylus" "ToolDebugLevel" 12
xsetwacom --set "ELAN9038:00 04F3:261A eraser" "ToolDebugLevel" 12
```
and looked into `/var/log/Xorg.0.log`.

Here is one moment where the stylus switches to the eraser:
```
[  1555.074] (II) ELAN9038:00 04F3:261A stylus (7:wcmSendEvents): [STYLUS] o_prox=true x=7210 y=3956 z=1567 b=true b=1 tx=10 ty=-26 wl=0 wl2=0 rot=0 th=0
[  1555.074] (II) ELAN9038:00 04F3:261A stylus (10:wcmRotateAndScaleCoordinates): rotate/scaled to 7210/3956
[  1555.074] (II) ELAN9038:00 04F3:261A stylus (6:wcmSendEvents): abs prox=1	x=7210	y=3956	z=1567	v3=10	v4=-26	v5=-900	v6=0	id=4500	serial=1	button=true	buttons=1
[  1555.074] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 0 numbers of data
[  1555.074] (II) ELAN9038:00 04F3:261A eraser (10:wcmDevReadInput): Read (1)
[  1555.079] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 1 numbers of data
[  1555.080] (II) ELAN9038:00 04F3:261A stylus (11:check_arbitrated_control): Event from active device; maintaining pointer control.
[  1555.080] (II) ELAN9038:00 04F3:261A stylus (7:wcmSendEvents): [STYLUS] o_prox=true x=7179 y=3955 z=1698 b=true b=1 tx=10 ty=-26 wl=0 wl2=0 rot=0 th=0
[  1555.080] (II) ELAN9038:00 04F3:261A stylus (10:wcmRotateAndScaleCoordinates): rotate/scaled to 7179/3955
[  1555.080] (II) ELAN9038:00 04F3:261A stylus (6:wcmSendEvents): abs prox=1	x=7179	y=3955	z=1698	v3=10	v4=-26	v5=-900	v6=0	id=4500	serial=1	button=true	buttons=1
[  1555.080] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 0 numbers of data
[  1555.080] (II) ELAN9038:00 04F3:261A eraser (10:wcmDevReadInput): Read (1)
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 1 numbers of data
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (6:check_arbitrated_control): Event from non-active non-touch device; yielding pointer control.
[  1555.082] (II) ELAN9038:00 04F3:261A stylus (7:wcmSendEvents): [STYLUS] o_prox=true x=7179 y=3955 z=0 b=false b=0 tx=0 ty=0 wl=2147483647 wl2=2147483647 rot=0 th=0
[  1555.082] (II) ELAN9038:00 04F3:261A stylus (6:wcmSendEvents): abs prox=0	x=7179	y=3955	z=0	v3=0	v4=0	v5=-901	v6=2147483647	id=2	serial=0	button=false	buttons=0
[  1555.082] (II) ELAN9038:00 04F3:261A stylus (6:wcmSendButtons): buttons=0
[  1555.082] (II) ELAN9038:00 04F3:261A stylus (4:sendAButton): TPCButton(off) button=0 state=0
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (7:wcmSendEvents): [ERASER] o_prox=false x=7187 y=3959 z=0 b=false b=0 tx=0 ty=-28 wl=0 wl2=0 rot=0 th=0
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (10:wcmRotateAndScaleCoordinates): rotate/scaled to 7187/3959
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (6:wcmSendEvents): abs prox=1	x=7187	y=3959	z=0	v3=0	v4=-28	v5=0	v6=0	id=9000	serial=1	button=false	buttons=0
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (6:wcmSendButtons): buttons=0
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 0 numbers of data
[  1555.082] (II) ELAN9038:00 04F3:261A eraser (10:wcmDevReadInput): Read (1)
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): Update to serial: 1
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[  1555.084] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[  1555.085] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.085] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[  1555.085] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): 
[  1555.085] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): Update to serial: 1
[... 96 lines cut ...]
[  1555.695] (II) ELAN9038:00 04F3:261A stylus (11:check_arbitrated_control): Event from active device; maintaining pointer control.
[... cut ...]
As @ShapeShifter499 mentioned in the other thread, the switch seems to have something to do with the tilt, as the pen tool seems to switch only when the tilt changes.

Although I don't understand some of the events happening there, I find it interesting that right before the change, it seems like a button event is detected while right after the change there are these weird serial update messages.

I also fiddled around with gdb a little but I did not find anything useful in the stack traces of the events.

I hope this helps a little. I would really love to see this fixed somehow.

Baldurius, 2019-04-13:
Another quick update:
I enabled further debug outputs but setting the "TabletDebugLevel" in addition to the "ToolDebugLevel" ([reference](https://github.com/linuxwacom/xf86-input-wacom/wiki/xsetwacom#table-of-new-parameter-names))

```
xsetwacom --set "ELAN9038:00 04F3:261A stylus" "TabletDebugLevel" 12
xsetwacom --set "ELAN9038:00 04F3:261A eraser" "TabletDebugLevel" 12
```

After that, I also got the debug output between device access (in my case /dev/input/event19) and the wacom event dispatcher that parses the events and forwards them to the tool handling.

Here is another log output of a sudden (and wrong) tool switch:
```
[   303.365] (II) /dev/input/event19 (10:wcmReadPacket): fd=29
[   303.365] (II) /dev/input/event19 (1:wcmReadPacket): pos=0 remaining=256
[   303.365] (II) /dev/input/event19 (10:wcmReadPacket): buffer has 96 bytes
[   303.365] (II) /dev/input/event19 (10:usbParseEvent): 
[   303.365] (II) /dev/input/event19 (10:usbParseEvent): 
[   303.365] (II) /dev/input/event19 (10:usbParseEvent): 
[   303.365] (II) /dev/input/event19 (10:usbParseEvent): 
[   303.365] (II) /dev/input/event19 (6:usbDispatchEvents): 4 events received
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[0]->type=3 code=26 value=4100
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[1]->type=3 code=27 value=2600
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[2]->type=3 code=40 value=31500
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[3]->type=0 code=0 value=0
[   303.365] (II) /dev/input/event19 (10:usbDispatchEvents): Dirty flag set on channel 1; sending event.
[   303.365] (II) /dev/input/event19 (10:wcmEvent): channel = 1
[   303.365] (II) /dev/input/event19 (10:wcmEvent): c=1 i=31500 t=8 s=1 x=4697 y=3613 b=0 p=0 rz=0 tx=41 ty=26 aw=0 aw2=0 rw=0 t=0 px=1 st=0 cs=0 
[   303.365] (II) /dev/input/event19 (11:wcmEvent): tool id=8 for ELAN9038:00 04F3:261A eraser
[   303.365] (II) ELAN9038:00 04F3:261A eraser (6:check_arbitrated_control): Event from non-active non-touch device; yielding pointer control.
[   303.365] (II) /dev/input/event19 (2:wcmSoftOutEvent): send a soft prox-out
[   303.365] (II) ELAN9038:00 04F3:261A stylus (7:wcmSendEvents): [STYLUS] o_prox=true x=4697 y=3613 z=0 b=false b=0 tx=0 ty=0 wl=2147483647 wl2=2147483647 rot=0 th=0
[   303.365] (II) ELAN9038:00 04F3:261A stylus (6:wcmSendEvents): abs prox=0	x=4697	y=3613	z=0	v3=0	v4=0	v5=-901	v6=2147483647	id=2	serial=0	button=false	buttons=0
[   303.365] (II) /dev/input/event19 (10:commonDispatchDevice): device type = 8
[   303.365] (II) /dev/input/event19 (10:wcmFilterCoord): common->wcmRawSample = 4 
[   303.365] (II) /dev/input/event19 (10:storeRawSample): initialize channel data.
[   303.365] (II) /dev/input/event19 (10:wcmCheckSuppress): level = 2 return value = 8
[   303.366] (II) ELAN9038:00 04F3:261A eraser (7:wcmSendEvents): [ERASER] o_prox=false x=4697 y=3613 z=0 b=false b=0 tx=41 ty=26 wl=0 wl2=0 rot=0 th=0
[   303.366] (II) ELAN9038:00 04F3:261A eraser (10:wcmRotateAndScaleCoordinates): rotate/scaled to 4697/3613
[   303.366] (II) ELAN9038:00 04F3:261A eraser (6:wcmSendEvents): abs prox=1	x=4697	y=3613	z=0	v3=41	v4=26	v5=0	v6=0	id=31500	serial=1	button=false	buttons=0
[   303.366] (II) ELAN9038:00 04F3:261A eraser (6:wcmSendButtons): buttons=0
[   303.366] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 0 numbers of data
[   303.366] (II) ELAN9038:00 04F3:261A eraser (10:wcmDevReadInput): Read (1)
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmGetProperty): Update to serial: 0
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.366] (II) ELAN9038:00 04F3:261A stylus (10:wcmSetProperty): 
[   303.367] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): 
[   303.367] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): Update to serial: 1
[   303.367] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.367] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): Update to serial: 1
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmSetProperty): 
[   303.368] (II) ELAN9038:00 04F3:261A eraser (10:wcmGetProperty): 
[   303.370] (II) ELAN9038:00 04F3:261A eraser (10:wcmReady): 1 numbers of data
```

I noticed that the switches from stylus to eraser and eraser to stylus correlate with the
```
[   303.365] (II) /dev/input/event19 (2:wcmSoftOutEvent): send a soft prox-out
```
lines. This call is located in line **1044** in `wcmCommon.c`.

ShapeShifter499, 2019-04-13:
Should this line be fixed or am I misunderstanding the code?  https://github.com/linuxwacom/xf86-input-wacom/blob/26cc50112cfe98d81ca4c38fd7bdff5ef08a8178/src/wcmISDV4.c#L724

Baldurius, 2019-04-14:
> Should this line be fixed or am I misunderstanding the code?
> 
> [xf86-input-wacom/src/wcmISDV4.c](https://github.com/linuxwacom/xf86-input-wacom/blob/26cc50112cfe98d81ca4c38fd7bdff5ef08a8178/src/wcmISDV4.c#L724)
> 
> Line 724 in [26cc501](/linuxwacom/xf86-input-wacom/commit/26cc50112cfe98d81ca4c38fd7bdff5ef08a8178)
> 
>  wcmEvent(common, 0, &out); 
> Changed to `wcmEvent(common, channel, &out); `?

Alhtough the line looks suspicious, I don't think that the ISDV4 backend is involved at all. In my logs, only the USB backend is used. Or am I missing something?

Furthermore, the channel is initialized at
https://github.com/linuxwacom/xf86-input-wacom/blob/26cc50112cfe98d81ca4c38fd7bdff5ef08a8178/src/wcmISDV4.c#L684
and never changes after that. So one would have to compute the channel from the device type or something first.

jigpu, 2019-04-15:
I haven't had a time to look into this yet (sorry!) but @Baldurius is correct that the ISDv4 backend isn't going to be used for this device. The issue will be in either `wcmUSB.c` (which reads input events from the kernel for processing) or `wcmCommon.c` (which does the actual processing and emission of X11 events).

I find the following lines from the log particularly suspect, however:
~~~
[   303.365] (II) /dev/input/event19 (6:usbDispatchEvents): 4 events received
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[0]->type=3 code=26 value=4100
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[1]->type=3 code=27 value=2600
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[2]->type=3 code=40 value=31500
[   303.365] (II) /dev/input/event19 (11:usbDispatchEvents): event[3]->type=0 code=0 value=0
~~~
These are the kernel events that are received and basically translate to:
~~~
EV_ABS ABS_TILT_X      4100
EV_ABS ABS_TILT_Y      2600
EV_ABS ABS_MISC        31500
EV_SYN SYN_REPORT      0
~~~
The presence of `ABS_MISC` in particular is concerning, since xf86-input-wacom assumes its presence means that the tablet doesn't use the "generic protocol". Instead, either "protocol 4" or "protocol 5" (which are similar Wacom-specific abuses of the evdev event semantics) is in use:

https://github.com/linuxwacom/xf86-input-wacom/blob/2062126997bfe2c014533873d73a95198b335305/src/wcmUSB.c#L799-L801

One effect of protocol 5 in particular is the driver "discovering" tool type information from the ABS_MISC value. In particular, the mask 0x008 is used to determine if a tool is an eraser. At least for the above log this is the case, and might be why the driver suddenly thinks an eraser is present: `31500 & 0x008 = 0x008`

https://github.com/linuxwacom/xf86-input-wacom/blob/2062126997bfe2c014533873d73a95198b335305/src/wcmUSB.c#L1157-L1183

I don't know what data the Surface Go pen is sending through ABS_MISC, but one way to see if it is a protocol issue would be to change the first hunk of code above to something more like the following so that the driver *forces* the use of generic protocol:
~~~
	if (1)
		common->wcmProtocolLevel = WCM_PROTOCOL_GENERIC;
~~~

Baldurius, 2019-04-15:
Yes! The pen is working now!

First I tried your suggestion and substituted `if (!ISBITSET(abs, ABS_MISC))` with `if (1)`. This did not change the behavior nor the log output.

Then I looked for `ABS_MISC` and changed
https://github.com/linuxwacom/xf86-input-wacom/blob/2062126997bfe2c014533873d73a95198b335305/src/wcmUSB.c#L1266-L1269
into
```
return STYLUS_ID;
```

While this resolve the eraser issue, the lines were sometimes cut off. I then continued looking for ABS_MISC and commented out
https://github.com/linuxwacom/xf86-input-wacom/blob/2062126997bfe2c014533873d73a95198b335305/src/wcmUSB.c#L1265-L1269

Now the stylus and even the eraser work as expected. This is of course far from a proper fix.

I printed some more of these ABS_MISC events:

```
[   865.079] (II) /dev/input/event9 (11:usbDispatchEvents): event[6]->type=3 code=40 value=31500
[   865.314] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=0
[   865.585] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=4500
[   865.765] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=9000
[   865.956] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=13500
[   866.688] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=4500
[   866.838] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=0
[   867.581] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=4500
[   867.814] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=13500
[   867.822] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=9000
[   867.971] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=13500
[   868.333] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=18000
[   868.369] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=0
[   868.407] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=31500
[   868.722] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=0
[   869.007] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=4500
[   869.330] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=9000
[   870.929] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=4500
[   871.034] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=0
[   871.462] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=31500
[   871.777] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=22500
[   871.837] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=13500
[   872.415] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=4500
[   872.423] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=9000
[   872.497] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=4500
[   872.648] (II) /dev/input/event9 (11:usbDispatchEvents): event[4]->type=3 code=40 value=0
[   873.114] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=4500
[   873.173] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=13500
[   873.219] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=9000
[   873.482] (II) /dev/input/event9 (11:usbDispatchEvents): event[3]->type=3 code=40 value=13500
[   873.753] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=4500
[   873.984] (II) /dev/input/event9 (11:usbDispatchEvents): event[5]->type=3 code=40 value=0
```

To me this looks like angle information (angle * 100) in 45 degree steps. This would also explain @ShapeShifter499 's observation that it has something to do with pen angle.

ShapeShifter499, 2019-04-20:
@jigpu any new updates with this?

jigpu, 2019-04-22:
Sorry, this hasn't been able to bubble to the top of my priority queue yet :slightly_frowning_face: Its good to hear that you were able to isolate something that changes the behavior though -- it definitely will save some time trying to understand what needs to be changed. I do somewhat wonder about the source of the ABS_MISC data, but it ultimately doesn't matter since we obviously can't assume non-Wacom hardware abuses the evdev protocol in the way we expect.

qzed, 2019-04-26:
@jigpu From the hid-recorder logs on https://github.com/linuxwacom/libwacom/issues/75 under `Usage (Pen)`:

```
# 0x35, 0x00,                    //   Physical Minimum (0)              679
# 0x47, 0xa0, 0x8c, 0x00, 0x00,  //   Physical Maximum (36000)          681
# 0x15, 0x00,                    //   Logical Minimum (0)               686
# 0x27, 0xa0, 0x8c, 0x00, 0x00,  //   Logical Maximum (36000)           688
# 0x09, 0x3f,                    //   Usage (Azimuth)                   693
# 0x81, 0x02,                    //   Input (Data,Var,Abs)              695
```

Seems to be the azimuth. This would fit @Baldurius' range of values and @ShapeShifter499's observed behavior.

ShapeShifter499, 2019-04-28:
@Baldurius I tried making a patch that I could apply to the PKGBUILD used by the official Arch Linux xf86-input-wacom package. After compiling and installing the pen side works but the eraser side does not. Are you using the official kernel? Are you making use of any xorg.conf files?

[PKGBUILD.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3124741/PKGBUILD.txt)

[wcmUSB.patch.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3124739/wcmUSB.patch.txt)

ShapeShifter499, 2019-05-03:
@Baldurius @jigpu any further updates yet?

ShapeShifter499, 2019-05-27:
Any updates on this?  It's been a while

jigpu, 2019-05-28:
I was able to spend some time on this Friday and have a set of patches that will hopefully work. They're on my other machine at the moment but I'll post them here as soon as I can.

ShapeShifter499, 2019-05-29:
@jigpu I hope they work.   I also suspect that it might fix https://github.com/linuxwacom/libwacom/issues/70 But I'm not sure.

jigpu, 2019-05-30:
I've pushed a branch with the fixes to my repository. You can get a copy by running the following `git clone https://github.com/jigpu/xf86-input-wacom -b issue-52` Afterwards, compile and install it using the standard instructions. The changes are generalizations of those @Baldurius mentioned and should hopefully do the trick. I'll send a proper pull request if they test out okay.

ShapeShifter499, 2019-06-01:
@jigpu Good news, it seems to have completely solve the issue for me. It also seems to have solved linuxwacom/libwacom#70 

Testing on both tablets. Works fine in, Xournal++, "Stylus labs Write", and GIMP.

For Xournal++ I needed to enable the "Enable GTK Touch/Scrolling workaround" under Edit > Preferences > Touchscreen

I do have a question about the code, is it disregarding tilt altogether?   Is there any plans to try to properly include tilt?    I don't know if I actually will need tilt but would be nice to know if there will be plans to properly handle tilting, might come in handy if I ever need it in the future for art or anything.

ShapeShifter499, 2019-06-01:
@jigpu oh one more thing. Keeps giving me a "battery low" message for my stylus when it nears the screen for the first time, both tablets.

ShapeShifter499, 2019-06-04:
@jigpu Do you need any logs from this before the code gets accepted upstream?

jigpu, 2019-06-04:
@ShapeShifter499 no, the "battery low" message won't be impacted by this particular patch and should be handled in a separate issue. The kernel driver would be responsible for that, so you'd either need to report it to https://bugzilla.kernel.org/ (or the developer of the kernel modifications if you're using them). Same for the ABS_MISC tilt information (if it provides anything of values over the ABS_TILT_X/Y that I thought it was already reporting.

ShapeShifter499, 2019-06-04:
@jigpu I'm a bit confused. I wouldn't know if tilt is being handled properly or not since I never had any stylus device before that supported it. My question about tilt is, *should* it be still handled with the code you posted or is it being ignored? 

The "battery low" might be because I haven't connected the bluetooth bit of my Pen as @qzed mentioned at https://github.com/linuxwacom/libwacom/issues/89

ShapeShifter499, 2019-06-06:
@jigpu ok with messing around in Krita, seems tilt works great! I think the code should be merged into mainline. 

But I have a question.   In 'xinput test' what are the different values exactly referring to?   what is ```a[5]=-900```?

jigpu, 2019-06-10:
@ShapeShifter499 glad to get confirmation that tilt is working correctly :) The sensor sends `ABS_TILT_X` and `ABS_TILT_Y` data already, which is why I was confused about your previous question. I'm not sure why the azimuth data is being sent in `ABS_MISC` as well, but ultimately we don't need it with the X/Y tilt data.

As for the `xinput test` output, the meanings of each index are defined in the [`wcmInitAxes` function](https://github.com/jigpu/xf86-input-wacom/blob/c091c9e295db9c271de60c87a1a9ce921d638159/src/xf86Wacom.c#L155). The `a[5]` value will contain either pen rotation or airbrush fingerwheel values depending on the stylus. The complete list of (stylus) values is: x, y, pressure, tilt-x, tilt-y, rotation OR abswheel.

ShapeShifter499, 2019-06-13:
@jigpu Unless I'm missing something, I'm going to close this now that #62 has been merged.  Thank you so much for programing a working fix!

ShapeShifter499, 2019-06-13:
@jigpu I'm going to reopen because a new release should also be made.

ShapeShifter499, 2019-06-17:
@jigpu Should this change go into the upstream 70-wacom.conf?

```
/etc/X11/xorg.conf.d/60-wacom.conf 
------
Section "InputClass"
        Identifier "SurfaceGo Touch"
        MatchProduct "ELAN9038:00 04F3:261A"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
```

jigpu, 2019-06-17:
That might be a good idea. Would you mind submitting a pull request with the modification? Are there other Surface devices with other IDs that would also benefit?

Also, as a matter of course, we normally mark issues as resolved once the code is merged since we'll obviously be making a release at some point anyway :)

ShapeShifter499, 2019-06-17:
/cc @qzed You might want to chime in here about the configuration bits.