linuxwacom/xf86-input-wacom #53: Pen works but finger touch does not

lukemcnaboe, 2019-04-10:
Recently bought a Lenovo Yoga 720, running Ubuntu 18.04. Finger touch/zoom worked great for about a month, then _all of a sudden_ stopped working. A friend's wacom pen does work as a mouse. 

The kernel does seem to acknowledge that the driver's installed; this is the relevant output of   `more /proc/input/devices`
[image]

and `xinput list` suggests that something is recognizing the device's finger-touch functionality... 
[image]

I did try cleaning the screen, and looking for similar issues, but to no avail. Any help would be appreciated!

jigpu, 2019-04-11:
Can you run [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather some more information? You can drag and drop the generated tarball onto the comment box here to attach it for review.

lukemcnaboe, 2019-04-11:
Thanks! Please let me know if you need anything else.

[sysinfo.sRUJCHTsYu.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3069270/sysinfo.sRUJCHTsYu.tar.gz)

jigpu, 2019-04-12:
Thanks! Unfortunately there isn't anything jumping out at me as obviously suspect in the logs. Is it just the two-finger "zoom" that is broken, or is it even impossible to perform single-finger "tap" interactions? Do you have the ability to test the touchscreen in another OS (e.g. a Windows partition, or with a live Ubuntu USB stick)?

jigpu, 2019-06-20:
Closing for now. Feel free to re-open with the requested information.

closed 2019-06-20 by jigpu