linuxwacom/xf86-input-wacom #55: Variation of Installation Instructions for Debian/Ubuntu

martinomburajr, 2019-04-13:
Installation for Ubuntu/Debian documentation can read

`sudo apt-get update && sudo apt-get install xserver-xorg-input-wacom'`

Otherwise one may get the following
`/bin/bash: apt-get update && apt-get install xserver-xorg-input-wacom: command not found
`

jigpu, 2019-04-15:
What version of Ubuntu or Debian are you using? I can't think of any reason that you'd get a `command not found` error with the `sudo -s` version of the command but not the proposed version.

martinomburajr, 2019-04-20:
Hello @jigpu , Sorry was away for a bit.

This is the error I get when running the  ` sudo -s 'apt-get update && apt-get install xserver-xorg-input-wacom'` command

[image]

jigpu, 2019-04-22:
Thanks. What version of Ubuntu or Debian are you using?

jigpu, 2019-06-20:
Closing. Feel free to re-open with the necessary information.

closed 2019-06-20 by jigpu