linuxwacom/xf86-input-wacom #56: cannot disable gsdwacom anymore in new ubuntu version

braindevices, 2019-04-14:
```
$ dconf read /org/gnome/settings-daemon/plugins/gsdwacom/active
false
```
However, it is still running:
```
$ ps x | grep gsd-wacom
 1630 tty2     Sl+    0:00 /usr/lib/gnome-settings-daemon/gsd-wacom
```
So the [document](https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-3:-gnome-settings-daemon) is probably wrong and need to be fixed.

jigpu, 2019-04-15:
Thanks for the heads-up. We definitely need to update the document. GNOME moved most of the functions performed by the gsdwacom plugin directly into Mutter in GNOME 3.24. Disabling the plugin on this version or newer (even if it worked) probably won't have the desired effect.

jigpu, 2019-06-10:
Page moved to https://github.com/linuxwacom/xf86-input-wacom/wiki/Tablet-Configuration-3%3A-GNOME-3, fleshed out, and 3.24 change documented.

closed 2019-06-10 by jigpu