linuxwacom/xf86-input-wacom #59: Pan scrolling doesn't account for area change

jj-kim, 2019-05-31:
The pan scrolling feature added in 21f498e works fine, but once the area is changed (e.g. `xsetwacom set <id> area <params>`), it doesn't work so great.  My guess is that the area change is not accounted for somewhere.

When I set the area to be an upper portion of my tablet, scrolling always goes down just from holding my pan button and pressing without dragging.  When I set the area to be a lower portion of my tablet, scrolling always goes up.  This suggests to me that the two components of the drag distance calculation has one side on the original area mapping and the other side on the new area mapping, making it think that there is constant drag even when the pen is not moving.  This in the `wcmPanscroll` function may be the culprit:
```
	*accumulated_x += (ds->x - priv->oldState.x);
	*accumulated_y += (ds->y - priv->oldState.y);
```

My system is a fresh install of Debian 10 on XFCE and I built the drivers from source.

jj-kim, 2019-06-15:
I tested on a fresh install of Arch on i3, and I was able to reproduce the issue.
```
$ uname -rvmpio
5.1.9-arch1-1-ARCH #1 SMP PREEMPT Tue Jun 11 16:18:09 UTC 2019 x86_64 unknown unknown GNU/Linux
$ pacman -Q xf86-input-wacom
xf86-input-wacom 0.36.1-1
```

jigpu, 2019-06-18:
Thanks, I've been able to replicate this as well and believe I've fixed it with the pull request referenced above. Feel free to test out the pull request and provide feedback.

jj-kim, 2019-06-18:
I just tested it out on Arch, and it seems to work.  Thank you very much.

closed 2019-12-12 by jigpu