linuxwacom/xf86-input-wacom #61: on screen help and swap monitors via xsetwacom

tylerecouture, 2019-06-08:
When using the gnome Wacom/Tablet Settings app in Ubuntu (16.04) there is an option to set tablet express keys to either the "on screen help" overlay/popup or to "swap monitors".

How can I do these to things directly with xsetwacom?

jigpu, 2019-06-10:
Neither of these is possible with xsetwacom alone. You can, however, configure one of the buttons to send a keyboard shortcut and then configure your desktop to launch a program in response to it. If your desktop does not allow you to launch programs in response to keyboard shortcuts, [Xbindkeys](https://wiki.archlinux.org/index.php/Xbindkeys) may help.

To swap monitors, running the following simple script should do the trick:
```bash
#!/bin/sh
for ID in $(xsetwacom list | cut -f2 | cut -d' ' -f2); do xsetwacom set "$ID" maptooutput next; done
```

The "on screen help" overlay/popup is a part of GNOME. I'm not sure if its a program that can be manually launched or a part of the shell itself. If it exists as a binary that can be executed though, you can use the same trick to open it.

closed 2019-06-10 by jigpu

tylerecouture, 2019-06-10:
Thanks!  Any tips on how I can track down where that on screen help comes from?

jigpu, 2019-06-10:
I've done some poking around and it looks like the OSD is a component of the shell itself. The control panel opens it by sending a dbus method call to it:
~~~
$ dbus-monitor
[...]
method call time=1560188597.511949 sender=:1.536 -> destination=:1.16 serial=51 path=/org/gnome/Shell/Wacom; interface=org.gnome.Shell.Wacom.PadOsd; member=Show
   object path "/dev/input/event18"
   boolean true
[...]
~~~
I've tried to trigger it manually by using the `dbus-send` command but it fails for some reason:
~~~
$ dbus-send --print-reply /org/gnome/Shell/Wacom org.gnome.Shell.Wacom.PadOsd.Show objpath:/dev/input/event18 boolean:true
Error org.freedesktop.DBus.Error.UnknownMethod: Method "Show" with signature "ob" on interface "org.gnome.Shell.Wacom.PadOsd" doesn't exist
~~~

Hopefully that's enough to help you with your search.

tylerecouture, 2019-06-10:
Awesome, Thank you!