linuxwacom/xf86-input-wacom #64: Cintiq 24 Pro, can the touch buttons in the top right be bound to an action?

Lovewyrm, 2019-06-20:
These things here:

https://i.imgur.com/t09Ix8T.jpg

The 'display mode' button works out of the box, or perhaps, in other words, internally?
As in, the tablet can switch between HDMI, USB-C etc.

But the "wrench", "virtual keyboard", etc buttons don't seem to do anything.
In Windows, with the Wacom drivers and the GUI mabobber thingum it would open the settings, a virtual keyboard, etc.

It would be really neat if these buttons could be mapped to something on Linux.
Is that possible?
```

 xinput list       
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech USB-PS/2 Optical Mouse         	id=8	[slave  pointer  (2)]
⎜   ↳ Wacom Express Key Remote Pad pad        	id=15	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 24 Pen stylus          	id=16	[slave  pointer  (2)]
⎜   ↳ Wacom Cintiq Pro 24 Pen eraser          	id=17	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ C-Media Electronics Inc. USB Audio Device	id=9	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=10	[slave  keyboard (3)]

```

It does not show the tablet itself, and debugging any of the devices at level 12 also didn't yield anything.
So, my guess is that, if this is possible, then either I have maybe outdated drivers, etc, or I have to use a different 'tool' to bind these things.
Maybe something that listens to the USB events of the Wacom tablet , catching the button signals and doing something with those, but I don't know what that could be.

Lovewyrm, 2019-06-20:
I ran "usbmon" on it, and this is the output I got from tapping the "wrench" button.
```
ffff8d7d7128c900 3990879020 S Bo:035:01 -115 31 = 55534243 00020000 00000000 00000600 00000000 00000000 00000000 000000
ffff8d7d7128c900 3990879038 C Bo:035:01 0 31 >
ffff8d7d7128c900 3990879041 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3990880056 C Bi:035:02 0 13 = 55534253 00020000 00000000 01
ffff8d7d7128c900 3990880061 S Bo:035:01 -115 31 = 55534243 01020000 60000000 80000603 00000060 00000000 00000000 000000
ffff8d7d7128c900 3990880087 C Bo:035:01 0 31 >
ffff8d80c84315c0 3990880091 S Bi:035:02 -115 96 <
ffff8d80c84315c0 3990881676 C Bi:035:02 -121 18 = 70000200 0000000a 00000000 3a000000 0000
ffff8d7d7128c900 3990881684 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3990881971 C Bi:035:02 0 13 = 55534253 01020000 4e000000 00
ffff8d7d7128c900 3992927022 S Bo:035:01 -115 31 = 55534243 02020000 00000000 00000600 00000000 00000000 00000000 000000
ffff8d7d7128c900 3992927036 C Bo:035:01 0 31 >
ffff8d7d7128c900 3992927039 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3992928000 C Bi:035:02 0 13 = 55534253 02020000 00000000 01
ffff8d7d7128c900 3992928008 S Bo:035:01 -115 31 = 55534243 03020000 60000000 80000603 00000060 00000000 00000000 000000
ffff8d7d7128c900 3992928030 C Bo:035:01 0 31 >
ffff8d80c84315c0 3992928035 S Bi:035:02 -115 96 <
ffff8d80c84315c0 3992929966 C Bi:035:02 -121 18 = 70000200 0000000a 00000000 3a000000 0000
ffff8d7d7128c900 3992929974 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3992930323 C Bi:035:02 0 13 = 55534253 03020000 4e000000 00
ffff8d7d7128c900 3994975030 S Bo:035:01 -115 31 = 55534243 04020000 00000000 00000600 00000000 00000000 00000000 000000
ffff8d7d7128c900 3994975050 C Bo:035:01 0 31 >
ffff8d7d7128c900 3994975054 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3994975937 C Bi:035:02 0 13 = 55534253 04020000 00000000 01
ffff8d7d7128c900 3994975944 S Bo:035:01 -115 31 = 55534243 05020000 60000000 80000603 00000060 00000000 00000000 000000
ffff8d7d7128c900 3994975959 C Bo:035:01 0 31 >
ffff8d80c84315c0 3994975962 S Bi:035:02 -115 96 <
ffff8d80c84315c0 3994977550 C Bi:035:02 -121 18 = 70000200 0000000a 00000000 3a000000 0000
ffff8d7d7128c900 3994977556 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3994977865 C Bi:035:02 0 13 = 55534253 05020000 4e000000 00
ffff8d7d7128c900 3997023021 S Bo:035:01 -115 31 = 55534243 06020000 00000000 00000600 00000000 00000000 00000000 000000
ffff8d7d7128c900 3997023034 C Bo:035:01 0 31 >
ffff8d7d7128c900 3997023038 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3997024047 C Bi:035:02 0 13 = 55534253 06020000 00000000 01
ffff8d7d7128c900 3997024052 S Bo:035:01 -115 31 = 55534243 07020000 60000000 80000603 00000060 00000000 00000000 000000
ffff8d7d7128c900 3997024078 C Bo:035:01 0 31 >
ffff8d80c84315c0 3997024083 S Bi:035:02 -115 96 <
ffff8d80c84315c0 3997026173 C Bi:035:02 -121 18 = 70000200 0000000a 00000000 3a000000 0000
ffff8d7d7128c900 3997026181 S Bi:035:02 -115 13 <
ffff8d7d7128c900 3997026526 C Bi:035:02 0 13 = 55534253 07020000 4e000000 00

```
This was given by : `Bus 006 Device 029: ID 056a:0386 Wacom Co., Ltd Wacom Cintiq Pro 24 Pen Hub
`
Maybe this could be of some help.

Lovewyrm, 2019-06-20:
Have a naive (since I don't know much about all of thise) dump of the button presses, via usbmon.

```
                                                 v--- Is the button, 2 is the one that looks like a list 4 the OnscreenKB, 8 the wrench
ffff8d8191209200 3884913099 C Ii:064:01 0 9 = 11020000 00000000 00

ffff8d8191209200 3884913099 C Ii:064:01 0 9 = 11020000 00000000 00 < pressed list
ffff8d8191209200 3884913106 S Ii:064:01 -115 192 <
ffff8d8191209200 3885123098 C Ii:064:01 0 9 = 11000000 00000000 00 < released list
ffff8d8191209200 3885123104 S Ii:064:01 -115 192 <
ffff8d8191209200 3885233095 C Ii:064:01 0 9 = 11020000 00000000 00
ffff8d8191209200 3885233101 S Ii:064:01 -115 192 <
ffff8d8191209200 3885327094 C Ii:064:01 0 9 = 11000000 00000000 00
ffff8d8191209200 3885327099 S Ii:064:01 -115 192 <  
ffff8d80f9cb6600 3885371096 C Ii:061:01 0 32 = 10000100 691e0100 00000000 00000000 00000000 00000000 00000000 00000000 < not related, removing in log now
ffff8d80f9cb6600 3885371101 S Ii:061:01 -115 32 <
ffff8d8191209200 3888393070 C Ii:064:01 0 9 = 11040000 00000000 00 < pressed OnscreenKB
ffff8d8191209200 3888393077 S Ii:064:01 -115 192 <
ffff8d8191209200 3888527069 C Ii:064:01 0 9 = 11000000 00000000 00 < released OnscreenKB
ffff8d8191209200 3888527077 S Ii:064:01 -115 192 <
ffff8d8191209200 3888687068 C Ii:064:01 0 9 = 11080000 00000000 00 < pressed Wrench
ffff8d8191209200 3888687075 S Ii:064:01 -115 192 <
ffff8d8191209200 3888773068 C Ii:064:01 0 9 = 11000000 00000000 00 < released Wrench
ffff8d8191209200 3888773075 S Ii:064:01 -115 192 <
ffff8d8191209200 3889027055 C Ii:064:01 0 9 = 11040000 00000000 00 < OKB
ffff8d8191209200 3889027063 S Ii:064:01 -115 192 <
ffff8d8191209200 3889127054 C Ii:064:01 0 9 = 11000000 00000000 00
ffff8d8191209200 3889127061 S Ii:064:01 -115 192 <
ffff8d8191209200 3889257051 C Ii:064:01 0 9 = 11020000 00000000 00 < List
ffff8d8191209200 3889257057 S Ii:064:01 -115 192 <
ffff8d8191209200 3889393050 C Ii:064:01 0 9 = 11000000 00000000 00
ffff8d8191209200 3889393055 S Ii:064:01 -115 192 <
ffff8d8191209200 3889523049 C Ii:064:01 0 9 = 11040000 00000000 00 < OKB
ffff8d8191209200 3889523056 S Ii:064:01 -115 192 <
ffff8d8191209200 3889627047 C Ii:064:01 0 9 = 11000000 00000000 00
ffff8d8191209200 3889627053 S Ii:064:01 -115 192 <
ffff8d8191209200 3889793045 C Ii:064:01 0 9 = 11080000 00000000 00 < Wrench
ffff8d8191209200 3889793052 S Ii:064:01 -115 192 <
ffff8d8191209200 3889867046 C Ii:064:01 0 9 = 11000000 00000000 00
ffff8d8191209200 3889867053 S Ii:064:01 -115 192 <
ffff8d8191209200 3889867046 C Ii:064:01 0 9 = 11000000 00000000 00
```

I posted a different dump earlier but, as it turned out, it was not related to the buttons at all.
I'll try to give as much info as I can, but I know nothing , so ...I'll need instructions, not in depth, just tell me what to run, if needed, and I'll try to deliver.

jigpu, 2019-06-20:
The wacom kernel driver does handle those buttons and should pass them through to userspace through the "pad" device*. The "wrench" button sends a `KEY_BUTTONCONFIG` event, the "keyboard" button a `KEY_ONSCREEN_KEYBOARD` event, and the "wrench" button a `KEY_CONTROLPANEL` event. See the log below.

There's an open pull request (https://github.com/linuxwacom/xf86-input-wacom/pull/46) that would have xf86-input-wacom pass these buttons through as normal keys. Unfortunately, the X server drops keys with codes > 255, and these buttons do indeed have such high-numbered keys. We could theoretically have the buttons configurable through xsetwacom, but since the buttons are really intended to be used for their silkscreened functions we're a bit hesitant to do so.

At least for the moment you should be able to replace the uinput and hidraw backend with something that reads the evdev device instead. Not much of an improvement, but it would compatible with other Cintiqs with these buttons** since it doesn't have to assume a particular HID report format.

~~~
$ sudo evemu-record
Available devices:
[...]
/dev/input/event17:	Wacom Cintiq Pro 32 Touch Finger
/dev/input/event18:	Wacom Cintiq Pro 32 Pen
/dev/input/event19:	Wacom Cintiq Pro 32 Pad
[...]
Select the device event number [0-25]: 19
[...]
################################
#      Waiting for events      #
################################
E: 0.000001 0001 0240 0001	# EV_KEY / KEY_BUTTONCONFIG     1
E: 0.000001 0003 0028 0015	# EV_ABS / ABS_MISC             15
E: 0.000001 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +0ms
E: 0.290100 0001 0240 0000	# EV_KEY / KEY_BUTTONCONFIG     0
E: 0.290100 0003 0028 0000	# EV_ABS / ABS_MISC             0
E: 0.290100 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +290ms
E: 1.586323 0001 0278 0001	# EV_KEY / KEY_ONSCREEN_KEYBOARD 1
E: 1.586323 0003 0028 0015	# EV_ABS / ABS_MISC             15
E: 1.586323 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +1296ms
E: 1.726315 0001 0278 0000	# EV_KEY / KEY_ONSCREEN_KEYBOARD 0
E: 1.726315 0003 0028 0000	# EV_ABS / ABS_MISC             0
E: 1.726315 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +140ms
E: 2.256334 0001 0243 0001	# EV_KEY / KEY_CONTROLPANEL     1
E: 2.256334 0003 0028 0015	# EV_ABS / ABS_MISC             15
E: 2.256334 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +530ms
E: 2.416335 0001 0243 0000	# EV_KEY / KEY_CONTROLPANEL     0
E: 2.416335 0003 0028 0000	# EV_ABS / ABS_MISC             0
E: 2.416335 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +160ms
~~~

*The "touch" button is an exception and actually sends its `SW_MUTE_DEVICE` event from the *touch* device rather than the pad.

**Earlier devices like the Cintiq 24HD use `KEY_PROG1`, `KEY_PROG2`, and `KEY_PROG3`, but as described in the pull request, that might change in the future...

Lovewyrm, 2019-06-20:
I see, thank's a bunch for clearing that up...and for making me aware of evemu, seems like quite the useful toolkit.

Don't hesitate to make those buttons bindable, though ;) 
Or if you do, put them behind a simple config option, it's just so useful to have those hooks.
Instead of OSK it could be a handwriting recognition program, instead of the 'control panel' it could pop up a program selector/runner, etc.

For the time being, while I use my hackneyed solution, I'll read into the driver/xsetwacom code and make these buttons emit mousebuttons instead of keys.

Here is a snippet from my universal key (or mouse lol) setup.

```
xsetwacom set "$PAD"  Button 2 31
xsetwacom set "$PAD"  Button 3 32
xsetwacom set "$PAD"  Button 8 33
xsetwacom set "$PAD"  Button 9 34
xsetwacom set "$PAD"  Button 10 35
xsetwacom set "$PAD"  Button 14 36
xsetwacom set "$PAD"  Button 15 37
```

I have actually customized xbindkeys, too, to automatically call that script that acts on the different focused windows, before I had something like this:

```
#"/home/wacom/scripts/wacom_application_hotkeys.sh 1"
#    b:30
#"/home/wacom/scripts/wacom_application_hotkeys.sh 2"
#    b:31
#"/home/wacom/scripts/wacom_application_hotkeys.sh 3"
#    b:32
#"/home/wacom/scripts/wacom_application_hotkeys.sh 4"
#    b:33
#"/home/wacom/scripts/wacom_application_hotkeys.sh 5"
#    b:34
#"/home/wacom/scripts/wacom_application_hotkeys.sh 6"

```

The b:n bit is listening for the 'button' that xsetwacom emits.
This kind of 'button' (that is aparrently different from keys and has its own space of values? ) I'm going to try and make the hardware buttons emit for the time being, then.

The functionality is just too juicy to pass up on, unless maybe one day the linux drivers are equivalent to the windows ones (well, except less bloated and awful, hehe).
But yeah, I want those buttons, those shiny shiny buttons and their sweet sweet functionality.
More keys! More keys! More bindings! Rather mor than less! Tear that silk away! Hehe.

But yeah, again, thank you for clearing this all up. Makes me anticipate the future more happily.
I'm already incredibly glad that I'm not tied to windows thanks to this marvelous suite y'all have made.

jigpu, 2019-06-20:
Sounds like you're set for now, so I'll go ahead and close the issue.

If I may ask though... How are you automatically calling scripts for the focused window? I'm familiar with xbindkeys (and have often recommended using it to run a script that calls `xsetwacom set <id> MapToOutput next`), but I don't quite understand what you're saying... I was toying around with application settings a little bit ago and ended up writing a [hacky script](https://gist.github.com/jigpu/8e9f435fd69580440d54932872019215), but I'd be interested in knowing what you're doing.

closed 2019-06-20 by jigpu

Lovewyrm, 2019-06-21:
Yeah, I'm set, thanks again.

Also, my script is basically like yours where the meat is, only mine doesn't do any fancy looping or deamoning or anything but literally runs every time a button is pressed.

get_focused_window_name.sh
```
#!/bin/bash
title_window () { 
    xprop -id $(xprop -root 32x '\t$0' _NET_ACTIVE_WINDOW | cut -f 2) _NET_WM_NAME | cut -d '=' -f 2 | cut -b 1-2 --complement | rev | cut -c 2- | rev 2>&1
}

title_root () {
    title_window |& grep -q 0x0
}

if title_root;
  then
    echo "Desktop"
  else
   title_window
fi
```

It uses the NET_ACTIVE_WINDOW / WM_NAME hint Window managers (can) have.
It then feeds the window name back into the main script which then, like yours, has a big old "bash case" switch.
An abbreviated snippet (it goes back and forth in okular):

```

#!/bin/sh
WINNAME=$(get_focused_window_name.sh)

# Mapping - 18 buttons, 1 touch ring
#
#     3  4
#   2   1   5
#       6
#
#  shift   control
#      alt
#  7  8  9
#  10  11  12
#      13
#  14      15
# 16 touchring up
# 17 touchring down

case "$WINNAME" in
    *kular )
        case "$1" in
            14|16 )
               xdotool key k 
               ;;
            15|17 )
               xdotool key j 
               ;;
        esac
        ;;
esac

```

And that's basically it.
It's fast enough to work even if you spam buttons.
The important bit is the window manager hints, I use xmonad so I had to add that functionality in the config, I don't know how the other managers do it, the more mainstream ones like KDE / XFCE probably do it automatically, but the niche ones like xmonad, dwm and co probably need some encouragement first.

jigpu, 2019-06-21:
Ah, gotcha. That's a pretty neat way of handling things. Thanks for the insight :)