linuxwacom/xf86-input-wacom #65: Need help setting up  my stylus buttons.

ReductoAdAbsurdum, 2019-07-02:
Here is the output of `xsetwacom -s  get "Wacom HID 4806 Pen stylus" all`

https://justpaste.it/6dn97

My stylus has two buttons on the side + eraser button.

I want one of the side buttons to act as a right click, I enter the following command:

`xsetwacom set "Wacom HID 4806 Pen stylus" Button 2 3`

Then one of the side buttons is mapped to right click but when I touch the tablet with the tip of the stylus, there are continuous, random right clicks...can someone help me map a side button(doesn't matter which one) to right-click ?

Many Thanks

jigpu, 2019-07-17:
Apologies for the delay in replying. You're definitely on the right track as far as remapping one of the side switches to be right-click. I'm not sure what would be causing "continuous random right clicks", but perhaps try the following instead:

~~~
xsetwacom set "Wacom HID 4806 Pen stylus" Button 2 +3
~~~

The addition of a "+" before the 3 tells the driver that you want button 3 to be held down as long as you're physically pressing the button (an undecorated "3" triggers a full press-and-release cycle when the button is pressed).

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu