linuxwacom/xf86-input-wacom #71: Cintiq Companion 2 ExpressKey mapping issue in GNOME

miascugh, 2019-07-21:
Hi, I've run into the following problem using Pop_OS! 19.04 with GNOME 3.32.2 and a Wacom Cintiq Companion 2:
[image]

This is what happens in the gnome on-screen help when I press all the tablet buttons top-to-bottom, left-to-right in order.

The tablet is recognized fine, pen and features seem to work, with the exception of the bottommost ExpressKey, which seems to send the same event as the top rocker ring button.
The layout that is described in cintiq-companion-2.tablet looks correct:

I tried to research the subject, but I feel like I'm not tech-savvy enough to get to the bottom of the problem. Is it possibly to do with this: https://github.com/linuxwacom/xf86-input-wacom/wiki/Consumer-Tablet-ExpressKey-Mapping-Issue ? The description does mention, that cintiqs ought no be affected by it.

```
[sysinfo.XPUZJUS3Bt.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3414521/sysinfo.XPUZJUS3Bt.tar.gz)
$ xsetwacom list devices
Wacom ISDv5 325 Pad pad         	id: 18	type: PAD       
Wacom ISDv5 325 Pen stylus      	id: 19	type: STYLUS    
Wacom ISDv5 325 Pen eraser      	id: 20	type: ERASER    
Wacom ISDv5 326 Finger touch    	id: 21	type: TOUCH     
```
```
$ libwacom-list-local-devices
[Device]
Name=Wacom Cintiq Companion 2
ModelName=DTH-W1310
DeviceMatch=usb:056a:0325;
PairedID=usb:056a:0326;
Class=Cintiq
Width=12
Height=7
IntegratedIn=Display;System;
Layout=cintiq-companion-2.svg
Styli=0x802;0x804;0x80a;0x80c;0x902;0x90a;0x100802;0x100804;0x10080a;0x10080c;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=11
[Buttons]
Left=A;B;C;D;E;F;G;H;I;J;K;
Right=
Top=
Bottom=
Touchstrip=
Touchstrip2=
OLEDs=
Ring=
Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107;0x108;0x109;0x130;
RingNumModes=0
Ring2NumModes=0
StripsNumModes=0

---------------------------------------------------------------
[Device]
Name=Wacom Cintiq Companion 2
ModelName=DTH-W1310
DeviceMatch=usb:056a:0325;
PairedID=usb:056a:0326;
Class=Cintiq
Width=12
Height=7
IntegratedIn=Display;System;
Layout=cintiq-companion-2.svg
Styli=0x802;0x804;0x80a;0x80c;0x902;0x90a;0x100802;0x100804;0x10080a;0x10080c;0x100902;0x10090a;0x120802;0x140802;0x14080a;0x160802;0x16080a;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=11
[Buttons]
Left=A;B;C;D;E;F;G;H;I;J;K;
Right=
Top=
Bottom=
Touchstrip=
Touchstrip2=
OLEDs=
Ring=
Ring2=
EvdevCodes=0x100;0x101;0x102;0x103;0x104;0x105;0x106;0x107;0x108;0x109;0x130;
RingNumModes=0
Ring2NumModes=0
StripsNumModes=0

---------------------------------------------------------------
```
Worth noting, though probably not related, I've got both a /usr/share/libwacom and /usr/local/share/libwacom directory and don't know how that came to be (I suppose that is the cause for the device being listed twice?)

Please let me know what further information to provide to help look into this issue!

miascugh, 2019-07-21:
It might be important to point out, that I'm not running Linux on the Cintiq Companion 2, but rather have it plugged into my laptop as an input device.

skomra, 2019-07-23:
Fix posted upstream

https://www.spinics.net/lists/stable/msg317838.html

miascugh, 2019-07-24:
@skomra Thanks for looking into it!

skomra, 2019-08-07:
Fix accepted in 5.3, and added to 4.9, 4.14, 4.19, and 5.2 stable trees. It will be in the 0.44 release of input-wacom.