linuxwacom/xf86-input-wacom #74: pen/eraser confusion with wacom bamboo ink plus

cstoitner, 2019-08-15:
On my Lenovo X1 Tablet 3. gen the Wacom bamboo ink plus stylus seems to confuse eraser and pen mode. Usually the eraser-tool is selected when the stylus is brought near the display, irrespective of pushing the eraser button. Sometimes the normal pen tool is selected instead. When the eraser button is pushed after they stylus is already near the display then the pen tool is usually selected (and it stays that way even after the button is released).

Kritas tablet tester windows usually prints the following when the stylus is brought near the display and than taken away again:
```
Pen tip taken away
Eraser brought near
...
Eraser taken away
Pen tip taken away
```
With the Lenovo pen it looks like this instead:
´´´
Pen tip brought near
...
Pen tip taken away
´´´

The Lenovo pen included with the tablet works without problems. Both pens work on windows without problems. I am running debian bullseye with xserver-xorg-input-wacom 0.34.99.1-1.

This bug sound pretty similar to https://github.com/linuxwacom/xf86-input-wacom/issues/39 and maybe https://github.com/linuxwacom/xf86-input-wacom/issues/52.

I have run the `capture.sh` script from the other issue on the stylus device, drawing a short line with either pen without any buttons pressed.
[lenovo_pen_stylus.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3506735/lenovo_pen_stylus.tar.gz)
[wacom_bamboo_ink_plus_stylus.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3506736/wacom_bamboo_ink_plus_stylus.tar.gz)

cstoitner, 2019-08-15:
The Wacom pen seems to work correctly with the libinput driver, though the Lenovo pen lost the eraser function.

jigpu, 2019-08-16:
I've replayed your logs locally and don't see the problem with xf86-input-wacom 0.37.0. This could mean that the problem is fixed with the latest version (e.g. by the fix for #39). Please try installing the driver from source and see if that helps. Instructions can be found [here](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver).

jigpu, 2019-08-16:
Oops, looks like I tested the wrong script :flushed: 

Indeed, the problem you describe looks like its a variant of #52 (and would probably explain #39 as well). Specifically, our driver is seeing that your hardware sends `ABS_MISC` events, switches to an incorrect protocol, and then uses the value of `ABS_MISC` to decide wether the event is coming from a pen or eraser. The decision is based on wether the fourth bit of `ABS_MISC` is set; your Lenovo pen sends 33 (fourth bit is not set) while your Wacom pen sends -32645 (fourth bit *is* set).

It looks like commit 365739656d didn't go quite far enough in preventing this issue from ocurring...

Noir-, 2019-09-17:
@cstoitner Sorry for bothering you and for the off-topic: I also have the Bamboo Ink Plus stylus and want to make it run unter Linux and a Thinkpad Yoga 14. However, for me it already fails at the Bluetooth connection stage. The pairing works but the connection is killed instantly after it's established. I don't want to abuse this issue so if you can provide some information about that please drop me a mail at: noir@mailbox.org

jigpu, 2019-09-18:
@Noir- I understand wanting to get in touch with somebody else that has the same stylus, but please open a new issue rather than replying off-topic. I've heard of a similar issue and may have a workaround. Note, however, that even with the workaround, the Bluetooth button does *not* work correctly under Linux at the moment.

jigpu, 2019-09-23:
I've just pushed a possible fix for this up to my personal repository if anyone is interested in testing it. You should run `git clone https://github.com/jigpu/xf86-input-wacom.git -b fix-74` to get a copy of the updated code and then follow the [these instructions](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver) to build the the driver. I don't have the hardware to test this myself, so if anyone can test this please let me know how it goes.

1ynxy, 2019-10-06:
I've tested this fix on a New Dell XPS 13 2-in-1 7390 with a FHD+ display, also applying the additions from this branch here <https://github.com/dotellie/libwacom/tree/dell-xps-2-in-1-7390> so that the Wacom HID 48ED panel is recognised, and it appears to fix the problems with the eraser automatically selecting when using a Dell PN579X.

jigpu, 2019-11-01:
After spending some time on this and other related pen ID issues, I've made some changes to the "fix-74" branch mentioned above and created a pull request for the updated version. Would anyone be willing to test the updated version to make sure it still works for them? You should delete the xf86-input-wacom source directory you may already have and then follow the instructions in my post just above.

1ynxy, 2019-11-01:
In order to test this update I re-installed the AUR version of xf86-input-wacom to confirm that the issue still persisted, then I cloned the mentioned repository to a clean folder and installed it. This solution still works.

closed 2019-11-04 by jigpu