linuxwacom/xf86-input-wacom #79: Anki Input

TowelSniffer, 2019-09-11:
the Anki flashcard program https://apps.ankiweb.net/ does not register Wacom pen touch properly. I was wondering if it is possible to make the Wacom pen touch be seen as a normal left-click.

jigpu, 2019-09-11:
I do not see any problems with the version of Anki currently provided on the linked site (2.1.15-linux-amd64). Can you provide additional information about the specific issue you see and how to reproduce it?

TowelSniffer, 2019-09-12:
The issue I am having is with https://chanind.github.io/hanzi-writer/ 

To replicate, set up a card template with the following:

<script>/*! Hanzi Writer v2.0.4 | https://chanind.github.io/hanzi-writer */
!function(t,i){"object"==typeof exports&&"object"==typeof module?module.exports=i():"function"==typeof define&&define.amd?define([],i):"object"==typeof exports?exports.HanziWriter=i():t.HanziWriter=i()}("undefined"!=typeof self?self:this,function(){return function(n){var r={};function o(t){if(r[t])return r[t].exports;var i=r[t]={i:t,l:!1,exports:{}};return n[t].call(i.exports,i,i.exports,o),i.l=!0,i.exports}return o.m=n,o.c=r,o.d=function(t,i,n){o.o(t,i)||Object.defineProperty(t,i,{configurable:!1,enumerable:!0,get:n})},o.n=function(t){var i=t&&t.t?function(){return t.default}:function(){return t};return o.d(i,"a",i),i},o.o=function(t,i){return Object.prototype.hasOwnProperty.call(t,i)},o.p="",o(o.s=9)}([function(e,t,i){"use strict";(function(t){var u="function"==typeof Symbol&&"symbol"==typeof Symbol.iterator?function(t){return typeof t}:function(t){return t&&"function"==typeof Symbol&&t.constructor===Symbol&&t!==Symbol.prototype?"symbol":typeof t};var i=t.performance&&function(){return t.performance.now()}||function(){return Date.now()},n=t.requestAnimationFrame||function(t){return setTimeout(function(){return t(i())},1e3/60)},r=t.cancelAnimationFrame||clearTimeout,o=function(t){for(var n=Object(t),i=arguments.length,r=Array(1<i?i-1:0),o=1;o<i;o++)r[o-1]=arguments[o];return r.forEach(function(t){if(null!=t)for(var i in t)Object.prototype.hasOwnProperty.call(t,i)&&(n[i]=t[i])}),n},h=Object.assign||o;var s=0;e.exports={e:o,arrLast:function(t){return t[t.length-1]},assign:h,average:function(t){return t.reduce(function(t,i){return i+t},0)/t.length},callIfExists:function(t,i){return t&&t(i),i},cancelAnimationFrame:r,colorStringToVals:function(t){var i=t.toUpperCase().trim();if(/^#([A-F0-9]{3}){1,2}$/.test(i)){var n=i.substring(1).split("");3===n.length&&(n=[n[0],n[0],n[1],n[1],n[2],n[2]]);var r=""+n.join("");return{r:parseInt(r.slice(0,2),16),g:parseInt(r.slice(2,4),16),b:parseInt(r.slice(4,6),16),a:1}}var o=i.match(/^RGBA?\((\d+)\s*,\s*(\d+)\s*,\s*(\d+)(?:\s*,\s*(\d*\.?\d+))?\)$/);if(o)return{r:parseInt(o[1],10),g:parseInt(o[2],10),b:parseInt(o[3],10),a:parseFloat(o[4]||1,10)};throw new Error("Invalid color: "+t)},copyAndMergeDeep:function t(i,n){var r=h({},i);for(var o in n){var s=i[o],e=n[o];s!==e&&(s&&e&&"object"===(void 0===s?"undefined":u(s))&&"object"===(void 0===e?"undefined":u(e))&&!Array.isArray(e)?r[o]=t(s,e):r[o]=e)}return r},counter:function(){return++s},emptyFunc:function(){},inflate:function(t,i){for(var n=t.split("."),r={},o=r,s=0;s<n.length;s++){var e=s===n.length-1?i:{};o[n[s]]=e,o=e}return r},objRepeat:function(t,i){for(var n={},r=0;r<i;r++)n[r]=t;return n},performanceNow:i,requestAnimationFrame:n,timeout:function(){var n=0<arguments.length&&void 0!==arguments[0]?arguments[0]:0;return new Promise(function(t,i){setTimeout(t,n)})},trim:function(t){return t.replace(/^\s+/,"").replace(/\s+$/,"")}}}).call(t,i(1))},function(Ga,Ha){var Ia;Ia=function(){return this}();try{Ia=Ia||Function("return this")()||eval("this")}catch(t){"object"==typeof window&&(Ia=window)}Ga.exports=Ia},function(t,i,n){"use strict";var r=n(0),e=r.average,v=r.arrLast,u=function(t,i){return{x:t.x-i.x,y:t.y-i.y}},s=function(t){return Math.sqrt(Math.pow(t.x,2)+Math.pow(t.y,2))},l=function(t,i){return s(u(t,i))},h=function(t){var i=10*(1<arguments.length&&void 0!==arguments[1]?arguments[1]:1);return{x:Math.round(i*t.x)/i,y:Math.round(i*t.y)/i}},d=function(t){var r=t[0];return t.slice(1).reduce(function(t,i){var n=l(i,r);return r=i,t+n},0)},p=function(t,i,n){var r=u(i,t),o=n/s(r);return{x:i.x+o*r.x,y:i.y+o*r.y}},a=function(t){var e=1<arguments.length&&void 0!==arguments[1]?arguments[1]:.05,u=t.slice(0,1);return t.slice(1).forEach(function(t){var i=u[u.length-1],n=l(t,i);if(e<n)for(var r=Math.ceil(n/e),o=n/r,s=0;s<r;s++)u.push(p(t,i,-1*o*(s+1)));else u.push(t)}),u},c=function(t){for(var i=1<arguments.length&&void 0!==arguments[1]?arguments[1]:30,n=d(t)/(i-1),r=[t[0]],o=v(t),s=t.slice(1),e=0;e<i-2;e++)for(var u=v(r),h=n,a=!1;!a;){var c=l(u,s[0]);if(c<h)h-=c,u=s.shift();else{var f=p(u,s[0],h-c);r.push(f),a=!0}}return r.push(o),r},f=function(t){if(t.length<3)return t;var s=[t[0],t[1]];return t.slice(2).forEach(function(t,i){var n=s.length,r=u(t,s[n-1]),o=u(s[n-1],s[n-2]);r.y*o.x-r.x*o.y==0&&s.pop(),s.push(t)}),s};t.exports={round:h,equals:function(t,i){return t.x===i.x&&t.y===i.y},distance:l,getPathString:function(t){var i=1<arguments.length&&void 0!==arguments[1]&&arguments[1],n=h(t[0]),r=t.slice(1),o="M "+n.x+" "+n.y;return r.forEach(function(t){var i=h(t);o+=" L "+i.x+" "+i.y}),i&&(o+="Z"),o},frechetDist:function(r,o){for(var s=[],t=0;t<r.length;t++){s.push([]);for(var i=0;i<o.length;i++)s[t].push(-1)}return function t(i,n){return-1<s[i][n]||(s[i][n]=0===i&&0===n?l(r[0],o[0]):0<i&&0===n?Math.max(t(i-1,0),l(r[i],o[0])):0===i&&0<n?Math.max(t(0,n-1),l(r[0],o[n])):0<i&&0<n?Math.max(Math.min(t(i-1,n),t(i-1,n-1),t(i,n-1)),l(r[i],o[n])):1/0),s[i][n]}(r.length-1,o.length-1)},length:d,rotate:function(t,i){return t.map(function(t){return{x:Math.cos(i)*t.x-Math.sin(i)*t.y,y:Math.sin(i)*t.x+Math.cos(i)*t.y}})},subtract:u,extendStart:function(t,i){var n=f(t);if(n.length<2)return n;var r=n[1],o=n[0],s=p(r,o,i),e=n.slice(1);return e.unshift(s),e},cosineSimilarity:function(t,i){return(t.x*i.x+t.y*i.y)/s(t)/s(i)},outlineCurve:c,u:p,h:f, [... cut ...]
  width: 150,
  height: 150,
  showCharacter: false,
  padding: 5
});
writer.quiz();

</script>

<div id="character-target-div"></div>

{{Front}}

This creates a character that you can sketch out with a mouse, hower it does not work with the Wacom tablet. by using Anki 2.1.5 alternate version I can get it to work, however, this is not ideal for me. I understand that this is most likely a problem with the Anki program but I was curious if I could somehow make the wacom touch register as a normal left-click.

jigpu, 2019-11-11:
I've recently taken another look into this issue and it seems like I can get the strokes to show up if I hold down one of the pen's side switches at the same time as I draw. I haven't tested on a tablet PC yet, but that does seem to hold for a Cintiq 27... If holding one of the buttons works for you that could help isolate the root cause.

TowelSniffer, 2019-11-11:
Hi yes, the same happens for me when holding down whatever button on my pen that has been mapped to right-click.

TowelSniffer, 2019-11-11:
one thing I might add is that when running a version of anki where this bug is present, when clicking on any kind of text on a card, it will immediately highlight the word you clicked on whereas on a version without this bug, it behaves more like a regular mouse click in that you have to rapidly double click in order to highlight the word.

TowelSniffer, 2021-03-26:
No longer an issue

closed 2021-03-26 by TowelSniffer