linuxwacom/xf86-input-wacom #80: Hovering is detected as an eraser on HP Elitebook

lytex, 2019-09-14:
I have a HP Elitebook x360 1030 with an active G3 pen using Ubuntu 19.04. While wacom driver works fine for desktop applications that are not designed to work with a stylus, on apps like xournalpp and Stylus Labs’ Write, it seems like hovering over the screen is detected as an eraser. Thus, handwriting apps change its tool from pen to eraser as soon as the computer stops detecting the stylus hovering.
I attach xinput and xsetwacom information:

[xinput.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3612519/xinput.txt)
[xsetwacom.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3612520/xsetwacom.txt)

jigpu, 2019-09-18:
Likely a duplicate of #74. Can you please run our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh), make a single stroke with your stylus, press CTRL+C to stop recording, and then attach the generated tarball to this issue? The log should contain enough info to show if its the same issue or not.

lytex, 2019-09-22:
Yes, of course. I have recorded inputs for both Wacom HID 4891 Pen stylus and Wacom HID 4891 Pen eraser xinput devices:
[record_stylus.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3640051/record_stylus.tar.gz)
[record_eraser.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3640052/record_eraser.tar.gz)

jigpu, 2019-09-23:
Thanks. That does confirm that your pen is being mis-detected as per #74. I'm going to close this particular issue since it is a duplicate, but you can subscribe to the other issue so that you are notified when updates happen. I've just posted a possible fix to that issue, so hopefully someone will be along soon who can test it out.

closed 2019-09-23 by jigpu

lytex, 2019-10-08:
`sudo apt install libwacom*` solved the issue. I also compiled and installed xf86-input-wacom as described in #74 and related issues, but it didn't seem to work.