linuxwacom/xf86-input-wacom #83: MAJOR ISSUE: My changes did not make it in to 0.37.0 even though they are listed.

ShapeShifter499, 2019-09-28:
I just now noticed that my changed did not make it into the release despite being listed. Why is it this?

It is listed under the 24 commits here: https://github.com/linuxwacom/xf86-input-wacom/compare/xf86-input-wacom-0.37.0...master

BUT the actual download for the release doesn't show the change required for Surface Go and the Nuvision tablet.

https://github.com/linuxwacom/xf86-input-wacom/releases/latest

I was using a specific Xorg config file containing the changes required up till now but I have removed that file thinking the package correctly contained the fixes. But it does not.

ShapeShifter499, 2019-09-28:
/cc @jigpu @bentiss

jigpu, 2019-09-30:
You're misinterpreting the list is all :) Your link lists the changes that were committed to master *after* the release of version 0.37.0. Your patch will be part of the 0.38.0 release that will be coming soon.

closed 2019-09-30 by jigpu

ShapeShifter499, 2019-10-01:
@jigpu yipes, I'm terribly sorry.   I thought they made it into 0.37