linuxwacom/xf86-input-wacom #85: Mouse scrollwheel disabled when pen is in proximity to tablet

Lepidopterist, 2019-10-08:
We've recently upgraded our workstations to CentOS 7 from CentOS 6.  (We're currently on CentOS 7.7)

Our artists are now complaining that the Mouse scrollwheel is disabled when the pen is in proximity to the tablet.

Historically, our artists (for good or bad) would use both tablet left handed and scrollwheel of the mouse righthanded whilst working.  This is now not possible.

I've attempted to find any information on what might be causing this behaviour, but my GoogleFu is clearly not what it used to be.

Any thoughts about where to start looking?

Many thanks

jigpu, 2019-10-11:
I have not heard of this issue before, and while I can replicate this in a CentOS 7.7 VM, I am not able to reproduce it on my Arch Linux workstation. I will continue to diagnose what may be causing this issue in that particular environment.

jigpu, 2019-10-22:
Sorry for the delay in replying, I haven't had much luck on my own, but maybe @whot will recognize these symptoms or at least be able to track down the cause.

It looks like pen proximity is interfering with the scroll distance values reported by the server. On my (working) Arch Linux system, I see the following sequence of events when scrolling down with my pen in prox. Each click of the mousewheel increases the scroll distance in `a[3]` by 5 units:
~~~
$ xinput test 16 | awk '/motion a\[3\]|button/'
motion a[3]=21165 
button press   5 
button release 5 
motion a[3]=21180 
button press   5 
button release 5 
motion a[3]=21195 
button press   5 
button release 5 
motion a[3]=21210 
button press   5 
button release 5 
motion a[3]=21225 
button press   5 
button release 5 
motion a[3]=21240 
button press   5 
button release 5 
motion a[3]=21255
~~~
On a CentOS 7.x system, on the other hand, the sequence keeps resetting itself. The sequence below is from a single spin of my wheel down:
~~~
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-2 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-2 
button press   5 
button release 5 
motion a[3]=-3 
button press   5 
button release 5 
motion a[3]=-4 
button press   5 
button release 5 
motion a[3]=-5 
button press   5 
button release 5 
motion a[3]=-6 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-1 
button press   5 
button release 5 
motion a[3]=-2 
button press   5 
button release 5 
motion a[3]=-3 
button press   5 
button release 5 
motion a[3]=-4 
button press   5 
button release 5 
motion a[3]=-5 
button press   5 
button release 5 
motion a[3]=-6 
button press   5 
button release 5
~~~
I've tried a few different versions of CentOS 7.x, and even RHEL 8.0 and everything that I've tested seems to exhibit basically the same behavior. It doesn't seem to be related to the input driver, having tried both the evdev and Synaptics drivers on 7.x and libinput on 8.0.

whot, 2019-10-22:
almost certainly an xserver bug but I don't recall the specific change that would trigger this one. My guess would be that [this commit](https://gitlab.freedesktop.org/xorg/xserver/commit/d7b1753d446ecde3ff58e3de39a634c3137473c8) may be the fix for it.

@Lepidopterist what's the last centos 6 version this worked on? This would narrow down as to what xserver version introduced the change.

Lepidopterist, 2019-10-23:
I think we were pretty up to date with CentOS 6.10, but let me confirm and come back to you.  

I'll have to reinstate our old PXE images and reimage a box so it might be a couple of days.

Thanks for investigating!

Lepidopterist, 2019-11-05:
Just following up on this - sorry for the delay.

It appears CentOS 6.10 works as expected.

I hope that helps!

Cheers

Pinglinux, 2022-03-17:
Not a X driver issue

closed 2022-03-17 by Pinglinux