linuxwacom/xf86-input-wacom #86: Wacom crashes after suspend on X200T

Akraboy, 2019-10-10:
I have a ThinkPad X200 Tablet running Arch Linux (5.3.5) and when I suspend the device the touch and pen inputs stop working.
I think my problem is different from https://github.com/linuxwacom/input-wacom/issues/110 so I'm making a new issue.
I also tried reloading wacom and wacom_w8001 but that doesn't fix it. Rebooting however does. It doesn't have to be a cold boot. My journal-log shows that isdv4-serial-inputattach crashed.

Journalctl:

[journal.log](https://github.com/linuxwacom/input-wacom/files/3702028/journal4.log)

Dmesg:

[dmesg.log](https://github.com/linuxwacom/input-wacom/files/3702029/dmesg4.log)

edit: I just messed around with suspend again and had touch working a few times after waking up the device so this doesn't 100% of the time.

jigpu, 2019-10-10:
It looks like the crash is caused by some function writing past the end of a buffer. Unfortunately the stack trace contained in your journalctl doesn't provide the name of the misbehaving function. If you are able to occasionally reproduce this, I would suggest building and installing a version of the driver with debug information included. Since you're using Arch, you should be able to do that by following these steps:

1. Create a temporary directory and save the PKGBUILD from https://git.archlinux.org/svntogit/packages.git/plain/trunk/PKGBUILD?h=packages/xf86-input-wacom into it
2. Open a terminal and navigate to the temporary directory
3. Run `sed -i -e '/conflicts=/a\' -e 'options=(debug !strip)' PKGBUILD` to ensure the package is built with the necessary debug flags
4. Run `sudo pacman -S base-devel` to install the necessary development tools
5. From inside the temporary directory run `makepkg -s -f` to install remaining build dependencies, compile the driver, and create a package
6. Run `sudo pacman -U xf86-input-wacom-0.37.0-1-x86_64.pkg.*` to install the package
7. Reboot

Afterwards if you are able to trigger the issue again, run `sudo coredumpctl info isdv4-serial-inputattach` to get a copy of the relevant debug info for us.

Akraboy, 2019-10-16:
`sudo coredumpctl info isdv4-serial-inputattach` returns: 
>No coredumps found.

`sudo coredumpctl info isdv4-serial-in` works though and it returned this:

[isdv4-serial-in.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3733682/isdv4-serial-in.txt)

Akraboy, 2019-10-17:
It doesn't crash now but during wake-up I have to keep the pen and my fingers away from the screen or a few things can happen:

Most of the time either touch stops working or the cursor moves around erratically on the screen when I touch the screen or move the pen near the screen.
Here is the journal from an instance where the cursor becomes erratic:

[journal_erratic.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3739952/journal_erratic.txt)

Here is the output from `sudo evtest` for both touch and pen events:
[evtest_pen.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3739959/evtest_pen.txt)
[evtest_touch.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3739961/evtest_touch.txt)
I noticed that touch also triggers pen events and the pen also triggers touch events. Its as if they are merged into a single input device.

Another problem is that when I draw on the screen with the pen during wake-up it can sometimes cause the system to crash. Here is the journal for that:

[journal_crash.txt](https://github.com/linuxwacom/xf86-input-wacom/files/3739982/journal_crash.txt)

jigpu, 2019-10-17:
In the erratic case, I notice the following interesting messages:
~~~
~~~

jigpu, 2019-10-17:
In the erratic case, I notice the following interesting messages (note that the device is detected as 2FG when the system starts up):
~~~
Oct 17 16:40:53 caramel isdv4-serial-inputattach[666]: parsing error code -1
Oct 17 16:40:53 caramel isdv4-serial-inputattach[666]: error during query.
Oct 17 16:40:54 caramel isdv4-serial-inputattach[666]: touch parsing error code -1
Oct 17 16:40:54 caramel kernel: serio: Serial port ttyS0
Oct 17 16:40:54 caramel kernel: input: Wacom Serial Penabled 1FG Touchscreen Pen as /devices/pnp0/00:05/tty/ttyS0/serio3/input/input14
Oct 17 16:40:54 caramel kernel: input: Wacom Serial Penabled 1FG Touchscreen Finger as /devices/pnp0/00:05/tty/ttyS0/serio3/input/input15
~~~
It looks like the error messages can occur if `isdv4-serial-inputattach` gets an unexpected response to its queries for sensor information. Since these responses come down the same channel as pen and touch data, interacting with the device before the program has finished initializing could very well cause this. I don't, however, understand how the program is able to trigger the creation of a (1FG?!) device after the failure... The error should result in it immediately exiting. Any kind of weird behavior is possible though with the device somehow being brought up incorrectly, so I'm not too surprised that you're seeing weird issues afterwards.

The kernel crash is a little harder to pin down, but it could potentially be some kind of similar problem, with the kernel getting unexpected data due to the early input.

Unfortunately, there doesn't appear to be a trivial fix for these issues. I don't have a good knowledge of the ancient serial protocol, and would need quite a bit of time to figure out how to make `isdv4-serial-inputattach` and the w8001 driver handle unexpected data more gracefully. Its not something that I can devote time to though, so I'd have to recommend the "don't do that" workaround unless/until somebody else can tackle it.

jigpu, 2019-10-18:
I've created a pull request for my fix to the original issue, so it at least should be fixed in the next release of xf86-input-wacom.

The other two issues aren't likely to be dealt with anytime soon, but I would still appreciate if you could open separate issues for each. One on this same [xf86-input-wacom bug tracker](https://github.com/linuxwacom/xf86-input-wacom/issues/) for the erratic case, and another on our [input-wacom bug tracker](https://github.com/linuxwacom/input-wacom/issues/) for the kernel crash case.

jigpu, 2019-10-23:
Pull request has been merged. The next release of xf86-input-wacom should include it.

closed 2019-10-23 by jigpu