linuxwacom/xf86-input-wacom #87: Build failed, I appreciate any help

changeling420, 2019-10-14:
So I tried installing the kernel and the driver, but on both the build failed. With message:
 bash: ./configure: No such file or directory
Build Failed
not sure where to go from here.

changeling420, 2019-10-14:
Also, I'm running Linux Mint 19.2

Thanks

changeling420, 2019-10-14:
Sorry, this is what happens when I try to build the kernel

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/4.15.0-65-generic/build
checking kernel version... 4.15.0-65-generic
checking RHEL6 minor release... 0
checking RHEL7 minor release... 0
checking power supply version... pre-v4.1
checking abs accessor version... pre-v2.6.36
checking CONFIG_MODULE_SIG... y
checking CONFIG_MODULE_SIG_FORCE... 
checking CONFIG_LOCK_DOWN_KERNEL... y
checking CONFIG_LOCK_DOWN_IN_EFI_SECURE_BOOT... 
checking secure boot state... off
checking if modules must be signed... recommended
checking CONFIG_MODULE_SIG_HASH... sha512
checking for module signing hash algorithm... sha512
checking for kernel autogenerated key at /lib/modules/4.15.0-65-generic/build/certs/signing_key.pem... no
checking for kernel autogenerated cert at /lib/modules/4.15.0-65-generic/build/certs/signing_key.x509... no
checking for shim MOK key at /var/lib/shim-signed/mok/MOK.priv... no
checking for shim MOK cert at /var/lib/shim-signed/mok/MOK.der... no
checking for rEFInd MOK key at /etc/refind.d/keys/refind_local.key... no
checking for rEFInd MOK cert at /etc/refind.d/keys/refind_local.cer... no
checking for module signing key... 
checking for module signing certificate... 
configure: WARNING: Kernel module signing is recommended but we are unable to find signing keys. Consider re-running configure with the options '--with-signing-key=<key>' and '--with-signing-cert=<cert>' set.

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/4.15.0-65-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0'
Making all in 4.5
make[2]: Entering directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.15.0-65-generic/build M=/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-4.15.0-65-generic'
make[3]: *** No rule to make target 'and'.  Stop.
make[3]: Leaving directory '/usr/src/linux-headers-4.15.0-65-generic'
Makefile:39: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0/4.5'
Makefile:377: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/gamesteeze/Downloads/Programs and Drivers/input-wacom-0.44.0'
Makefile:308: recipe for target 'all' failed
make: *** [all] Error 2
Build Failed

jigpu, 2019-10-14:
I've created a separate issue for your problems building the "input-wacom" driver. As for *this* issue with the xf86-input-wacom driver, please make sure that you are inside the xf86-input-wacom directory before running the build commands. Also, are you running `./configure` on its own, or are you running the pair of commands specified [in the wiki](https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver)?

changeling420, 2019-10-15:
I ran it again in the directory where I saved the zip files. But the build still failed. I will try it again in a different directory. Thanks

jigpu, 2019-11-11:
Closing this issue. Feel free to re-open if you still have problems building.

closed 2019-11-11 by jigpu