linuxwacom/xf86-input-wacom #88: Intuos pro recording pen strokes without touching (Ubuntu 18.14)

lofreqnet, 2019-10-15:
I have an issue with my intuos pro (PTH-651) being hypersensitive to the point where almost all pen movements within about an inch of the tablet are being recorded as if the pen were touching the tablet surface. This is with the tablet plugged in, using Krita. Pressure sensitivity is otherwise working fine.

jigpu, 2019-10-16:
Unfortunately I haven't been able to replicate this issue myself. Can you please try to reproduce this issue while our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) is running? You can press CTRL+C when done and attach the generated tarball directly to this issue.

lofreqnet, 2019-10-18:
Do you have instructions on running this? I have enough knowledge to copy and paste commands and that's about it, and currently it's asking me to install other stuff.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jigpu, 2019-10-18:
It sounds like you're running it properly. The program requires certain debug tools to be installed in order to work, however. The install message should look something like the following and suggest commands to run.
~~~
The following utilities are missing and must be installed to capture logs:
  * xinput [Try: `apt-get install xinput` or `yum install xorg-x11-server-utils`]
  * evemu-record [Try: `apt-get install evemu-tools` or `yum install evemu-record`]
  * hid-recorder [Try: `dnf install hid-replay` or https://bentiss.github.io/hid-replay-docs/ ]
~~~
Since you're using Ubuntu, you will want to run whatever commands it suggests that start with `apt-get` through the `sudo` command. For example, you might run `sudo apt-get install xinput` and `sudo apt-get install evemu-tools`. For hid-recorder, since there's no `apt-get` command listed, you'll need to go to [the website it links to](https://bentiss.github.io/hid-replay-docs/) and follow the "From the source" instructions on that page (because there's no pre-built Ubuntu package unfortunately).

Once you've installed all the programs that the script needs to work, it should ask you to choose a device to record and then let you know once it is capturing events. At that point you should reproduce the issue. Once the issue has been reproduced, press CTRL+C to quit and the program will tell you where the logs that it saved are located.

After the recording is complete you should log into Github, open this bug, and then drag+drop the file onto the comment box to upload the file for review. You should also write a description (and -- if you want bonus points :grin: -- a screenshot or screen recording) of what you were doing in the recording so that I can understand what parts of of the log to focus on. For example, if the bug took a few seconds to trigger, or if you used your pen to open Krita and weren't able to see the issue until later in the recording, let me know that so I can focus on the later part of the logs.

lofreqnet, 2019-10-18:
> After the recording is complete you should log into Github, open this bug, and then drag+drop the file onto the comment box to upload the file for review. You should also write a description (and -- if you want bonus points 😁 -- a screenshot or screen recording) of what you were doing in the recording so that I can understand what parts of of the log to focus on. For example, if the bug took a few seconds to trigger, or if you used your pen to open Krita and weren't able to see the issue until later in the recording, let me know that so I can focus on the later part of the logs.

Thanks a ton, Jason! I SEEM to have done it correctly, but it didn't tell me where the logs were located after I exited. Are they the "output" files autom4te.cache folder?

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jigpu, 2019-10-18:
Did you see the `Press CTRL+C to stop recording` message? After pressing CTRL+C you should see the following message:
~~~
Archiving capture data...
Files have been archived to [SOME FILE].tar.gz
Please upload this file for analysis.
~~~
Did you see that message?

The only "autom4ate.cache" folder should be from the hid-recorder build process and doesn't contain any data from the script.

lofreqnet, 2019-10-18:
OK yeah that message was missing. :T In fact there was no message at all, just a new line. Though it did seem to be recording stuff.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

jigpu, 2019-10-23:
Weird. I wonder if you were running hid-recorder program (the only thing which would have created an "autom4te.cache" folder) rather than the capture.sh script. Could you please re-run the capture.sh script and let me know what happens?

lofreqnet, 2019-10-31:
Oops, that's exactly what I was doing after all. I've attached the log file (responding to this through email so hopefully it doesn't get eaten).

Thanks again for your help.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

whot, 2019-11-01:
@lofreqnet  I don't think github allows attachments over email, this one certainly got dropped. Also, please trim the email replies to just your message to avoid noise in the bugreport.

lofreqnet, 2019-11-08:
@whot Thanks for the tip, wasn't sure if it was capturing other stuff in the email. I'll just login here to reply from now on.

@jigpu Attached is the log file!
[record_1572556653.tar.gz](https://github.com/linuxwacom/xf86-input-wacom/files/3826188/record_1572556653.tar.gz)

lofreqnet, 2019-11-25:
Just wondering if anyone's taken a look at this yet.
Coloring with a mouse sucks.

whot, 2020-02-05:
From the evemu recording in there: ```
# Input device name: "Wacom Intuos Pro M Pad"
```
so this is the wrong device, we'll need the stylus portion of the pad, not the pad, sorry.

Pinglinux, 2022-08-10:
@lofreqnet Do you still experience the issue? If yes, please record the device with "Wacom Intuos Pro M Pen" in its name and share the output here.

closed 2022-08-10 by lofreqnet