linuxwacom/xf86-input-wacom #89: Changing CursorProximity does nothing

AlessandroMarinoni, 2019-10-17:
Hello,

I've seen that in version 0.37.99 and later in 0.38 the possibility of changing CursorProximity to devices like mine (CTL-472) has been added.

However, if I use either `xsetwacom set "Wacom One by Wacom S Pen eraser" CursorProximity 55` or `xsetwacom set "Wacom One by Wacom S Pen eraser" CursorProximity 2` or any other value, nothing changes at all.

I'm using the tablet in absolute mode.
Am I doing something wrong, or is it a bug?

jigpu, 2019-10-17:
As documented in the man page, the updated CursorProx setting only applies in relative mode. Is there a reason you are wanting to use this in absolute mode?

AlessandroMarinoni, 2019-10-17:
I'm really sorry, I even read the manual, I must have missed it...

As I mainly use the tablet to play osu, it is very handy to set an active area that's comfortable to reach only with wrist movements, so that I can reach any part of the screen easily, while also having the possibility to jump from here to there just by pointing where I want the cursor to be.
I also would need to set CursorProxymity though because it sometime misses the signal, due to the fact that the closer i point to the edges the more I raise the pen.

Is there something that can be done (I would do it myself if only I was capable...) to let us set CursorProx while in absolute mode?

I'm already very grateful for the fact that this last update happened, if this means a whole lot of new work I could try to figure out a way to play with relative...

jigpu, 2019-10-17:
It doesn't sound like CursorProximity would be any help to you since the setting can only _decrease_ the usable hover height. Absolute mode acts as though CursorProximity is infinite, and is ultimately limited by hardware's proximity signal. If the tablet itself says the pen has left prox, we can't really do much better :D

I haven't actually tested it, but I suspect that the Intuos line of tablets may have a bit better hover height if you can find one to try out. Your CTL-472 is based on older technology and might not be able to detect the pen at as large of a distance. I know the Intuos Pro has a very large hover height compared to tablets I've used in the past, but that's probably a bit much for just Osu :)

AlessandroMarinoni, 2019-10-17:
Oh okay, this I didn't know: the fact that it was already set to the maximum 😅

Yeah the Intuos Pro is waay to much for osu, and in particular for me as I'm just a casual player.
Do you pheraps think that the Intuos S (CTL-4100K-S) will be better in this regard?

And lastly, thanks for your continued efforts on this driver!

jigpu, 2019-10-17:
I _assume_ the Intuos will be a little better, but as I said, I haven't actually tested it :)

closed 2019-10-17 by jigpu

AlessandroMarinoni, 2019-10-17:
Okay I understand, thanks once again :)