linuxwacom/xf86-input-wacom #91: Erratic cursor movements caused by touching or moving the pen near the screen during wake-up on X200T

Akraboy, 2019-10-22:
This issue is a follow-up to another issue (https://github.com/linuxwacom/xf86-input-wacom/issues/86) where this new problem was already mentioned. I will explain it again here:

If I wake my ThinkPad X200T from suspend and at the same time touch the screen or hover with the pen over it there is a chance that the touch and pen devices are not working correctly. Touch and pen inputs will result in both touch and pen events and the cursor moves all over the screen while randomly clicking.

To fix this I have to suspend the laptop again and wake it up while being careful that I'm not causing any touch or pen inputs during wake-up.

The following files are from the previous issue (#86):
[Journal log](https://github.com/linuxwacom/xf86-input-wacom/files/3739952/journal_erratic.txt)
[evtest for pen events](https://github.com/linuxwacom/xf86-input-wacom/files/3739959/evtest_pen.txt)
[evtest for touch events](https://github.com/linuxwacom/xf86-input-wacom/files/3739961/evtest_touch.txt)

whot, 2019-10-22:
ftr, both pen and touch events look normal on a quick glance. Are these the recordings from when the device misbehaves?

Akraboy, 2019-10-23:
Yes the events were recorded while I was experiencing the issue.

Here I graphed the first 84 X coordinates from the touch events file:

[image]

As you can see there are about 40 jumps between low and high values in less than 1 second.

Another thing is that the maximum values are incorrect. The maximum pen pressure for example should be 255 not 918. These incorrect maximum values are exceeded multiple times in the pen events file (pressure values 1007, 979 and 978).

whot, 2019-10-24:
ok, on closer look - yes, there's something wrong with the device. Just the first three events:

```
Event: time 1571315774.297552, type 3 (EV_ABS), code 0 (ABS_X), value 9343
Event: time 1571315774.297552, type 3 (EV_ABS), code 1 (ABS_Y), value 33347
Event: time 1571315774.297552, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 639
Event: time 1571315774.297552, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 0
Event: time 1571315774.297552, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 1
Event: time 1571315774.297552, -------------- SYN_REPORT ------------
Event: time 1571315774.305418, type 3 (EV_ABS), code 0 (ABS_X), value 8703
Event: time 1571315774.305418, type 3 (EV_ABS), code 1 (ABS_Y), value 63566
Event: time 1571315774.305418, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 847
Event: time 1571315774.305418, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1571315774.305418, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 0
Event: time 1571315774.305418, -------------- SYN_REPORT ------------
Event: time 1571315776.026895, type 3 (EV_ABS), code 0 (ABS_X), value 58470
Event: time 1571315776.026895, type 3 (EV_ABS), code 1 (ABS_Y), value 8131
Event: time 1571315776.026895, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 632
Event: time 1571315776.026895, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1571315776.026895, type 1 (EV_KEY), code 331 (BTN_STYLUS), value 1
Event: time 1571315776.026895, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 1
```
That's definitely an issue with the kernel or the device itself. Punting to @jigpu

jigpu, 2019-10-28:
Mentioned in the other bug (#86), but the root cause appears to be isdv4-serial-inputattach getting junk / unexpected data from the tablet and then incorrectly setting up the device. Device shows up as  "Wacom Serial Penabled 2FG" on boot, and then (after some parsing errors) as a "Wacom Serial Penabled 1FG" on resume. If the kernel is trying to decode 2FG reports with its function for 1FG sensors, things are going to be quite erratic indeed :)

Next steps:
* Understand why isdv4-serial-inputattach doesn't die after the parse failure
* Determine how to filter out the bad data so the device can be set up properly

Pinglinux, 2022-08-10:
The device is more than 10-years old. We recommend the workaround (don't touch it while the wakes up).

closed 2022-08-10 by Pinglinux