linuxwacom/xf86-input-wacom #93: Cursor click offset with Qt-based applications

jigpu, 2019-11-06:
There has been a steady stream of reports from a number of different users about programs not responding to clicks or seeing that they are offset. The common thread seems to be that all the affected programs are based on Qt. Not all Qt-based applications are affected, and in some cases the problem only occurs when interacting with certain UI elements or windows.

## Possible Workaround
In at least two instances (OBS and VLC), running the program from the command line with a special environment variable set appears to resolve the issue. Please try launching your program (e.g. "obs", "vlc", etc.) as follows:
~~~
QT_XCB_TABLET_LEGACY_COORDINATES=1 <program>
~~~

## Reports in the Wild
* [Weird OBS studio mapping](https://twitter.com/pigdev/status/1156191821401341952)
  * [Bug report](https://obsproject.com/mantis/view.php?id=1520)
* [VLC affected by bug](https://twitter.com/pigdev/status/1157043649634586626)
* [Some apps in KDE Neon](https://www.reddit.com/r/kde/comments/cr4nga/issue_with_wacom_in_kde_neon/) (OBS, Gwenviewer)
* [Pen clicks not working in Manjaro KDE](https://forum.manjaro.org/t/wacom-pen-clicks-not-working-in-some-applications/105389) (LibreOffice, VirtualBox)
* [Anki issues](https://github.com/linuxwacom/xf86-input-wacom/issues/79)

## Technical Description
Copied from the OBS bug report:

>When a pen tablet is used to interact with the main OBS UI, the position of the cursor does not match up with where OBS thinks it is. The amount of offset in X and Y changes depending on (and is equal to) how far the window is from the top-left corner of the screen. Most controls are affected, but it seems like the drag handles between the various window sections are not. Other windows within OBS (e.g. Settings or About) are not affected either. Only pen input seems to be affected; mouse navigation works fine.
>
> The issue has been reproduced under both X and Wayland for GNOME, as well as when running KDE or awesome as a window manager. The tablet works fine in all other applications I have tested, including other Qt-based applications without special tablet handling like Konsole and Kompare.
>
> Running OBS from the command-line with `QT_XCB_TABLET_LEGACY_COORDINATES=1 obs` seems to work around the issue. This environment variable was added in Qt 5.13.0 and 5.12.4 (https://github.com/qt/qtbase/blob/5.13.0/dist/changes-5.13.0#L60). I suspect that the issue has something to do with the custom widgets that OBS seems to use, since the issue is only present on the main UI and not in other Qt applications.

bogdancovaciu, 2019-12-05:
Hello,

I can confirm that the workaround is 100% successful with all mentioned applications in Manjaro KDE Plasma.
Much obliged for keeping the eye on this and providing such useful information. 
Cheers!

jigpu, 2019-12-20:
Upstream bug report: https://bugreports.qt.io/browse/QTBUG-77826

jigpu, 2020-01-13:
Marked as fixed in 5.14.1, 5.15.0 Alpha in the upstream report. Upstream bug fixes:
 * https://codereview.qt-project.org/c/qt/qtbase/+/284141
 * https://codereview.qt-project.org/c/qt/qtbase/+/285182

Not clear if being backported to 5.12 also (first commit above is on the 5.12 branch though)

braindevices, 2020-04-24:
Thanks a lot. The workaround is 100% functional!

RaphaelNanje, 2020-06-04:
Still getting the offset bug with 5.14.2 on Krita.
The workaround is still a viable solution.

mrgaturus, 2020-07-06:
Another misc bug: cursor isn't hovering correctly on some widget (mostly notable at menubar) even using workaround
[image]

poke1024, 2020-09-27:
Hello. I'm still seeing this with Krita 4.3.0 on a fresh install of Manjaro Linux 20.1 with Gnome 3.36.6. I'm not sure how to check the installed `xf86-input-wacom` version. With `QT_XCB_TABLET_LEGACY_COORDINATES`, things work perfectly.

themissingcow, 2020-11-04:
> Another misc bug: cursor isn't hovering correctly on some widgets (mostly notable at menubar) even using workaround

Fwiw, just tracked the menu bar issue down to [this commit](https://code.qt.io/cgit/qt/qtbase.git/commit/src/widgets/widgets?id=c5307203f5c0b0e588cc93e70764c090dd4c2ce0). Will file a report.

Pinglinux, 2022-08-11:
At this point, we can tell the root causes are in Qt. Please follow the progress at Qt site.

closed 2022-08-11 by Pinglinux

The-Maize, 2023-10-03:
is there any way to sett that command as a global option for those that use a tablet as a mouse?