linuxwacom/xf86-input-wacom #95: 70-wacom.conf incomplete?

arneko, 2019-11-25:
Hi,

I've been struggling to get pen input with my  HP Spectre x360 13-ac033ng to work properly with the HP Tilt Pen on ArchLinux. Multiple sources stated, that it should register as a wacom device, and it showed up when calling

`$ libwacom-list-local-devices 
# Device node: /dev/input/event7
[Device]
Name=ELAN 24DB
ModelName=
DeviceMatch=i2c:04f3:24db;
Class=ISDV4
Width=12
Height=7
IntegratedIn=Display;System;
Styli=0xffffe;0xfffff;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
# StatusLEDs=
NumStrips=0
Buttons=0
`

However, xsetwacom wouldn't show it.

After searching through my xorg log, I found, that above device was not using the wacom driver, and the [Archlinux Wiki](https://wiki.archlinux.org/index.php/Wacom_Tablet#Tablet_recognized_but_xsetwacom_and_similar_tools_do_not_display_it) hinted me towards the 70-wacom.conf. In the original 70-wacom.conf I did not find my device, so I added it:

`# HP Spectre
Section "InputClass"
        Identifier "HP Spectre Touch"
        MatchProduct "ELAN0732:00 04F3:24DB"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
`

With this section, the pressure and the eraser work properly + the device now shows up as wacom device.
Not sure about Tilt, and there are some more features that I cannot get to work yet (Presenter mode? Anyone? Not even sure where to look for that...)

Now, my question is: Is above Section really missing, or is something else wrong on my system?

whot, 2019-11-25:
it's a combination of multiple things.

xsetwacom [works only with the wacom xorg driver](https://who-t.blogspot.com/2016/12/the-future-of-xinput-xmodmap-setxkbmap.html), so if another driver is running it simply won't find the device.

That driver on modern distributions is the xf86-input-libinput driver which also has tablet support but is otherwise only partially compatible with the wacom driver (read: it works fine for most application, but the configuration options are completely different).

Which driver is picked for the device is up to the configuration snippets. We ship one that matches on all Wacom Inc devices and some other tablets (see [70-wacom.conf](https://github.com/linuxwacom/xf86-input-wacom/blob/master/conf/70-wacom.conf)) but those others have to be added one-by-one [1]. xf86-input-libinput ships [one that matches on all tablets](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/blob/master/conf/40-libinput.conf), and there may be distribution-specific snippets too. Fedora for example uses an extra snippet for Wacom-branded touchpads.

So, what this all comes down to: Yes, that section is missing because the Elan devices are enabled one-by-one. If you say it works, we can add that snippet to the default config file. There's a good chance the tablet worked anyway through the libinput driver (if not, that'd be a bug) and just the xsetwacom-based configuration didn't work.

[1] this was historically required because of bugs in the firmwares and driver peculiarities. It may not be true anymore but we simply can't test even a meaningful fraction of them.

arneko, 2019-11-25:
Thanks for the clarification. And yes, the libinput driver worked to some extend (read, it recognized both finger and stylus input, but could not distinguish the tip from the eraser and I could not get pressure to work)

Also, while the finger input properly changed with the orientation of the device, the stylus input did not.... This situation is now reversed, but I guess that can be remedied by some script (or may be caused by some script that I installed and forgot about...), but this is definitely something that I can handle otherwise.

So as far as I can tell so far, above section is an **improvement** to the previous situation and, personally, I'd welcome it to be part of the official release.

Apparently a number of HP Spectre devices have variants of this hardware, so it's quite possible that there will be more devices missing. So my section-title and identifier may not be 100% accurate. My hardware is a **HP Spectre x360 13-ac033ng** 

Aside from that:
Are you saying, I should also file a bug in libinput for the missing eraser+pressure detection?

Thanks for your time! I really appreciate it.

whot, 2019-11-26:
> but could not distinguish the tip from the eraser and I could not get pressure to work

all these would be good to file [in the libinput issue tracker](https://gitlab.freedesktop.org/libinput/libinput/issues) because libinput needs to be able to handle this.

> while the finger input properly changed with the orientation of the device, the stylus input did not....

libinput 1.14 does rotate tablet and finger correctly but only when it knows that the two are paired together. So this too is likely a bug in libinput that should be fixed.

So yes, please do file bugs for all these, thanks.

aleksfadini, 2019-11-29:
@arneko I'm having the some issues with the stock non-tilt pen with a HP Spectre x360 13-aw000dx, which is amazing for anything else under Arch Linux. I tried with and without the arch wacom package.
1 - Did you get to try the non-tilt pen with wacom.conf?
2 - Can you confirm that the tilt-pen is basically usable for artwork when using the config you suggested?

Thank you

arneko, 2019-11-29:
Sorry, I did not test the non-tilt pen.

WIth above config, the pen is quite usable, I'd say.

I tried gimp and xournal++.

Have you tried adding above xorg and does it work for you? If not, check if you indeed have the same touch panel device.

I also wrote a bug report for libinput - the intermediate result is, that future versions (or current master) of libinput work much better.

aleksfadini, 2019-11-30:
Turns out that both
`dmesg | grep -i wacom`
and 
`libwacom-list-local-devices `
come out with nothing, so I'm guessing the non-tilt pen has nothing to do with wacom. I'm a bit at a loss here since with libinput the pen "skips" some pixels when drawing. Maybe I should get the tilt pen and revert to Wacom drivers.
Any ideas/advice?

whot, 2020-03-09:
> can I force it to use wacom drivers instead of libinput?

under X, yes, you can assign `Driver "wacom"` in your xorg.conf snippet, see the first comment.

>  with libinput the non tilt pen is a bit laggy and requires a lot of pressure when drawing. 

lagginess is not usually something we can fix because it's not our fault - firmware. if the pressure is off that is something we can fix, please file a libinput bug for this, thanks.

jigpu, 2021-07-21:
Closing due to inactivity.

closed 2021-07-21 by jigpu