linuxwacom/xf86-input-wacom #97: S Pen Not Recognized

Aster-the-Med-Stu, 2019-12-26:
I am using a Samsung Notebook 9 Pro (NP940X5N-X01US), but the digitizer doesn't get recognized.

```xinput``` shows it:
```
~> xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Barcode Reader  Mouse                   	id=10	[slave  pointer  (2)]
⎜   ↳ Barcode Reader  Consumer Control        	id=12	[slave  pointer  (2)]
⎜   ↳ G2Touch Multi-Touch by G2TSP            	id=15	[slave  pointer  (2)]
⎜   ↳ ELAN0B00:00 04F3:306A Touchpad          	id=16	[slave  pointer  (2)]
⎜   ↳ WCOM0033:00 2D1F:001E Mouse             	id=17	[slave  pointer  (2)]
⎜   ↳ WCOM0033:00 2D1F:001E Pen (0)           	id=22	[slave  pointer  (2)]
⎜   ↳ WCOM0033:00 2D1F:001E Eraser (0)        	id=23	[slave  pointer  (2)]
```

But running ```libwacom-list-local-device``` or ```xsetwacom --list``` gives me nothing.

whot, 2020-01-07:
judging by the device name ("WCOM0033:00 2D1F:001E Pen (0)") I think you may be using the libinput xorg driver. what does `xinput list-props " WCOM0033:00 2D1F:001E Pen (0)"` say?

Aster-the-Med-Stu, 2020-01-10:
It gives me the following output:
```
~> xinput list-props "WCOM0033:00 2D1F:001E Pen (0)"
Device 'WCOM0033:00 2D1F:001E Pen (0)':
	Device Enabled (169):	1
	Coordinate Transformation Matrix (171):	0.500000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Node (292):	"/dev/input/event12"
	Device Product ID (293):	11551, 30
	libinput Tablet Tool Pressurecurve (653):	0.000000, 0.000000, 0.000000, 0.000000, 1.000000, 1.000000, 1.000000, 1.000000
```

And weird enough... After booting up this time ```WCOM0033:00 2D1F:001E Eraser (0)```  no longer shows up.

Aster-the-Med-Stu, 2020-01-10:
It shows as following:
```
~> xinput list-props "WCOM0033:00 2D1F:001E Pen (0)"
Device 'WCOM0033:00 2D1F:001E Pen (0)':
	Device Enabled (169):	1
	Coordinate Transformation Matrix (171):	0.500000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Node (292):	"/dev/input/event12"
	Device Product ID (293):	11551, 30
	libinput Tablet Tool Pressurecurve (643):	0.000000, 0.000000, 0.000000, 0.000000, 1.000000, 1.000000, 1.000000, 1.000000
```

For eraser... I couldn't find it after booting up this time.

whot, 2020-01-14:
the libinput driver works a bit differently to the wacom driver - specifically it only generates tools as they come into proximity. You'll find that as you move the eraser in, that tool will appear again.

what's missing here is that one of the xorg.conf snippets doesn't apply to this device and thus the wacom drive isn't assigned. the default snippets assign to strings like "Wacom" (missing in the device name) and the wacom vendor ID 0x5ac but you have a different one for some reason (11551 == 0x2d1f). So none of the default snippets applies and you fall back to the libinput driver instead.

find the 70-wacom.conf snippet on your host, copy it to /etc/X11/xorg.conf.d and add WCOM as a match product string. That should fix the immediate issue - long term I'd say we should add WCOM to the default matches - @jigpu?

Aster-the-Med-Stu, 2020-01-17:
Generic match should have already been done, according to ```/usr/share/X11/xorg.conf.d/70-wacom.conf```:

```
# Serial Wacom devices should always be one of tablet, touchscreen, or
# touchpad so we can safely get away with just one match section in
# these cases
Section "InputClass"
        Identifier "Wacom PnP device class"
        MatchPnPID "WACf*|WCOM*|WACM*|FUJ02e5|FUJ02e7|FUJ02e9"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection
```

But why does the above config not working...?

whot, 2020-01-17:
hm, indeed. attach your xorg.log please and that should explain it

Aster-the-Med-Stu, 2020-01-18:
Here it is:
```
[    39.897] (II) event12 - WCOM0033:00 2D1F:001E: is tagged by udev as: Tablet
[    39.980] (II) event12 - WCOM0033:00 2D1F:001E: tablet 'WCOM0033:00 2D1F:001E' unknown to libwacom
[    39.981] (II) event12 - WCOM0033:00 2D1F:001E: device is a tablet
[    39.981] (II) event12 - WCOM0033:00 2D1F:001E: device removed
[    40.026] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:15.2/i2c_designware.2/i2c-8/i2c-WCOM0033:00/0018:2D1F:001E.0008/input/input15/event12"
[    40.026] (II) XINPUT: Adding extended input device "WCOM0033:00 2D1F:001E" (type: TABLET, id 18)
[    40.031] (II) event12 - WCOM0033:00 2D1F:001E: is tagged by udev as: Tablet
[    40.032] (II) event12 - WCOM0033:00 2D1F:001E: tablet 'WCOM0033:00 2D1F:001E' unknown to libwacom
[    40.032] (II) event12 - WCOM0033:00 2D1F:001E: device is a tablet
[    40.032] (II) config/udev: Adding input device WCOM0033:00 2D1F:001E (/dev/input/mouse4)
[    40.032] (II) No input driver specified, ignoring this device.
[    40.032] (II) This device may have been added with another device file.
```

I guess I have to update ```libwacom.stylus```, right? There is no entry matching 0x2d1f (11551).

Aster-the-Med-Stu, 2020-01-18:
And libwacom seems to be unable to recognize all Samsung devices' digitizer.

This is another log taken from [AskUbuntu](https://askubuntu.com/questions/943041/unable-to-calibrate-samsung-galaxy-book-touchscreen-with-xinput?rq=1):

```
[    10.756] (**) WCOM0028:00 2D1F:000C: Applying InputClass "libinput pointer catchall"
[    10.756] (**) WCOM0028:00 2D1F:000C: Applying InputClass "libinput tablet catchall"
[    10.756] (II) Using input driver 'libinput' for 'WCOM0028:00 2D1F:000C'
[    10.756] (**) WCOM0028:00 2D1F:000C: always reports core events
```

whot, 2020-01-19:
please provide the whole log, otherwise one or two crucial lines may be missing, thanks

jigpu, 2020-01-24:
Can you please follow the instructions at the bottom of https://github.com/linuxwacom/wacom-hid-descriptors/ to contribute your system information. This is usually sufficient for us to create a libwacom tablet definition and will also contain a copy of your Xorg log and xorg.conf.d configuration which we can analyze to figure out why the driver isn't working as expected.

@whot regarding 0x2d1f, the intent was for devices with that VID to use the generic drivers rather than Wacom-specific ones. I was hesitant to add the VID to our 70-wacom.conf, but it looks like the "WCOM*" match PnP match that's been around for ages should already be matching them anyway... I'll skip adding the VID for now, though I suppose a USB device with that PID (if one exists) would still want it...

whot, 2020-01-28:
@jigpu: the generic drivers in the kernel, right? Or in userspace as well? because there's a reasonable expectation IMO that all wacom devices look the same in userspace, including the various special features the xf86-wacom driver provides. Obviously that won't matter once we're all on wayland+libinput anyway.

Aster-the-Med-Stu, 2020-01-28:
Sorry that I don't have any media with Linux distribution with me yet. I would provide additional infomation after Spring Festival holiday.

jigpu, 2020-02-07:
Thanks for providing the sysinfo. I don't see anything wrong with the device or driver which leads me to suspect that the problem is with `70-wacom.conf`. I would expect the section you highlighted above to match the device, but it pretty clearly isn't and instead falling through to libwacom. For the moment, you can try creating a file named `/etc/X11/xorg.conf.d/90-wacom-2d1f.conf` with the following contents and then rebooting:
~~~
Section "InputClass"
    Identifier "Wacom 2D1F tablet class"
    MatchUSBID "2d1f:*"
    MatchDevicePath "/dev/input/event*"
    MatchIsTablet "true"
    Driver "wacom"
EndSection
~~~

jigpu, 2020-02-07:
@whot that's hard to answer... I would personally like to see the distinction between 2D1F and 056A disappear (or, more accurately, just stop producing 2D1F hardware). There really is no practical difference between them at the moment. I suspect, however, that Wacom will at some point want the two VIDs to be treated differently by our first-party kernel and userspace drivers. Enabling the 056A experience for 2D1F would be a footgun waiting to go off.

Aster-the-Med-Stu, 2020-02-08:
I could confirm the additional config works, but there still is one peculiar problem. Unlike any other "normal" Wacom tablets, S Pen's eraser button is "locked down". I mean even after releasing the eraser button, it still works as eraser unless I move it out of electromagnetic field then move back. Is this done deliberately by Samsung?

whot, 2020-02-10:
@jigpu 

>  would be a footgun waiting to go off.

worse than right now where some of the devices aren't working correctly and we still have to fix them anyway? :) 

@Aster-the-Med-Stu 

>  I mean even after releasing the eraser button, it still works as eraser unless I move it out of electromagnetic field then move back. I

run `libinput record` against the event device node and reproduce the issue, then attach the resulting file here please. The [libinput documentation](https://wayland.freedesktop.org/libinput/doc/latest/tools.html) has some info on how to run this.

Aster-the-Med-Stu, 2020-02-11:
OK. The point mentioning this here is that I've noticed email address in commit message, and there is no response from Wacom Support in China too...

I'll open a separate issue for this.

closed 2020-02-11 by Aster-the-Med-Stu