linuxwacom/xf86-input-wacom #102: CTL-4100WL pen tip registered as both tip and eraser after firmware update

rongcuid, 2020-02-18:
I am using a CTL-4100WL on Pop OS 19.10. After updating the firmware of the tablet, pointer moves erratically over screen. Checking gnome tablet settings and xsetwacom shows the following:

```
$ xsetwacom --list
Wacom Co.,Ltd. Intuos BT S stylus	id: 15	type: STYLUS    
Wacom Co.,Ltd. Intuos BT S eraser	id: 16	type: ERASER
```

Here, the eraser should not be present as the stylus has only a tip. This causes conflicting input rendering the tablet unusable. I also notice the white LED light on tablet to be dimmer than usual.

However, Bluetooth connection still works as intended:

```
$ xsetwacom --list
Wacom Intuos BT S Pen stylus    	id: 15	type: STYLUS    
Wacom Intuos BT S Pad pad       	id: 16	type: PAD
```

Bluetooth works only if the tablet was not connected previously. If the tablet was connected via USB, disconnected, then connected through bluetooth, then Xorg crashes up on usage. Unsure if useful, but here are some kernel messages:

```
[  439.935685] usb 1-6: SerialNumber: 8DH00S1002016
[  439.941942] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:0376.000C/input/input52
[  439.942426] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:0376.000C/input/input54
[  439.942957] wacom 0003:056A:0376.000C: hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-6/input0
[  440.130637] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:0376.000D/input/input55
[  440.190824] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:056A:0376.000D/input/input57
[  440.191282] wacom 0003:056A:0376.000D: hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-6/input0
[  440.751732] usb 1-6: USB disconnect, device number 9
[  441.154579] usb 1-6: new full-speed USB device number 10 using xhci_hcd
[  441.304808] usb 1-6: New USB device found, idVendor=2d1f, idProduct=0376, bcdDevice= 1.11
[  441.304814] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  441.304818] usb 1-6: Product: Intuos BT S
[  441.304821] usb 1-6: Manufacturer: Wacom Co.,Ltd.
[  441.304824] usb 1-6: SerialNumber: 8DH00S1002016
[  441.307652] input: Wacom Co.,Ltd. Intuos BT S as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:2D1F:0376.000E/input/input58
[  441.308222] hid-generic 0003:2D1F:0376.000E: input,hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-6/input0
[  465.844139] usb 1-6: USB disconnect, device number 10
[  505.928927] wacom 0005:056A:0377.000F: unknown main item tag 0x0
[  505.929254] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.000F/input/input59
[  505.929673] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.000F/input/input61
[  505.930220] wacom 0005:056A:0377.000F: hidraw2: BLUETOOTH HID v0.00 Device [Intuos BT S] on 38:de:ad:5e:fe:a1
[  627.535886] wacom 0005:056A:0377.0010: unknown main item tag 0x0
[  627.536202] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.0010/input/input62
[  627.536735] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-11/1-11:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.0010/input/input64
[  627.537321] wacom 0005:056A:0377.0010: hidraw2: BLUETOOTH HID v0.00 Device [Intuos BT S] on 38:de:ad:5e:fe:a1
[  651.110917] show_signal_msg: 24 callbacks suppressed
[  651.110919] krita[6044]: segfault at 4 ip 00007fe5265536dd sp 00007ffcd8408ce8 error 4 in libQt5Core.so.5.12.4[7fe526495000+2df000]
[  651.110923] Code: 04 44 8b 40 04 48 03 76 10 48 01 c1 e9 6c fc ff ff 66 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 f0 48 8b 07 49 8b 10 <48> 63 70 04 48 63 4a 04 39 ce 75 47 48 83 ec 08 48 8b 3d 1c f4 3f
```

jigpu, 2020-02-19:
This is an issue with your tablet automatically switching into an incompatible "Android mode". Please see https://github.com/linuxwacom/xf86-input-wacom/wiki/Known-Issues#android-misdetect for details and a workaround.

closed 2020-02-19 by jigpu

rongcuid, 2020-02-20:
Workaround is effective. Though it is slightly inconvenient to switch mode manually each time the tablet is plugged in.